FreeCAD/FreeCAD #10134: Removing "constraint" from the names of most FEM workbench features

FEA-eng, 2023-08-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.0.33668 +7 (Git)
Build type: Release
Branch: (HEAD detached at 0.21)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Polish/Poland (pl_PL)
[/code]
```

### Subproject(s) affected?

FEM

### Problem description

I'd like to suggest a change in nomenclature. Currently, most tools in the FEM workbench are named "Constraint ...". For example, we have "Constraint force", "Constraint displacement", "Constraint temperature" and many more. What's wrong with that? It's confusing for people with some experience in FEA. For us, the "constraint" name is reserved for just a small subset of analysis features. That's also the case in CalculiX (and in Abaqus on which CalculiX is based). We have "Tie constraint" (available in FreeCAD and named "Constraint tie") and a few other constraints unsupported in the FEM workbench but that's it. If we were to follow the theory behind FEM, there are single-point constraints (SPCs) applied to a single degree of freedom (boundary conditions) and multi-point constraints (MPCs) applied to multiple degrees of freedom (equations). But it's really confusing when we call everything "constraint". For example, loads shouldn't be called this way. Even worse when we have something like "Constraint sectionprint", This one is ridiculous because it has literally nothing to do with constraints, it just prints the results from a selected face of the model. No way it could affect DOFs like a constraint would.

Since FEA is my job, I can suggest less confusing names for all the existing "constraints" if you agree with me that we should get rid of this unnecessary prefix.

Marking @kaktusus because we discussed the plans to improve the nomenclature and translations in the FEM workbench.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

kaktusus, 2023-08-09:
I believe that the theses presented are correct and it's worth using the knowledge and experience of FEA-eng who is a professional in this field ....
😉 
FEA-eng
My guess is that it would be helpful for developers to have Crowdin links to the strings that would be affected by the change.

FEA-eng, 2023-08-09:
Strings to change and suggested new names part 1:
https://crowdin.com/translate/freecad/559/en-pl#6593175 Heat flux
https://crowdin.com/translate/freecad/559/en-pl#6593993 Heat flux
https://crowdin.com/translate/freecad/559/en-pl#6593155 Creates a contact constraint between faces
https://crowdin.com/translate/freecad/559/en-pl#6593159 Creates a displacement boundary condition for a geometric entity
https://crowdin.com/translate/freecad/559/en-pl#6593163 Creates a fixed boundary condition for a geometric entity
https://crowdin.com/translate/freecad/559/en-pl#6593169 Creates a force load applied to a geometric entity
https://crowdin.com/translate/freecad/559/en-pl#6593171 Gear constraint
https://crowdin.com/translate/freecad/559/en-pl#6593173 Creates a gear constraint
https://crowdin.com/translate/freecad/559/en-pl#6593177 Creates a heat flux acting on a face
https://crowdin.com/translate/freecad/559/en-pl#6593181 Creates an initial temperature acting on a body
https://crowdin.com/translate/freecad/559/en-pl#6593185 Creates a plane multi-point constraint for a face
https://crowdin.com/translate/freecad/559/en-pl#6593193 Creates a pulley constraint
https://crowdin.com/translate/freecad/559/en-pl#6593199 Creates a spring acting on a face
https://crowdin.com/translate/freecad/559/en-pl#6593205 Creates a temperature/concentrated heat flux acting on a face
https://crowdin.com/translate/freecad/559/en-pl#6593297 Make bearing constraint
https://crowdin.com/translate/freecad/559/en-pl#6593299 Make contact constraint on a face
https://crowdin.com/translate/freecad/559/en-pl#6593301 Make displacement boundary condition on face
https://crowdin.com/translate/freecad/559/en-pl#6593303 Make fixed boundary condition for geometry
https://crowdin.com/translate/freecad/559/en-pl#6593307 Make force load on geometry
https://crowdin.com/translate/freecad/559/en-pl#6593309 Make gear constraint
https://crowdin.com/translate/freecad/559/en-pl#6593315 Make plane multi-point constraint on face
https://crowdin.com/translate/freecad/559/en-pl#6593319 Make spring on face
https://crowdin.com/translate/freecad/559/en-pl#6593321 Make pulley constraint
https://crowdin.com/translate/freecad/559/en-pl#6593325 Make local coordinate system on face
https://crowdin.com/translate/freecad/559/en-pl#6593449 Mesh groups are exported too. Every analysis feature and, if there are different materials, material consists of two mesh groups, faces and nodes where the constraint or material is applied.
https://crowdin.com/translate/freecad/559/en-pl#6612399 All analysis features are hidden in the model view when the results dialog is opened
https://crowdin.com/translate/freecad/559/en-pl#6497006 Analysis feature parameters
https://crowdin.com/translate/freecad/559/en-pl#6593593, https://crowdin.com/translate/freecad/559/en-pl#6593599 and https://crowdin.com/translate/freecad/559/en-pl#6593599 Only one type of selection (vertex,face or edge) per analysis feature allowed!
https://crowdin.com/translate/freecad/559/en-pl#6593667 Only one face can be selected for a plane multi-point constraint!
https://crowdin.com/translate/freecad/559/en-pl#6593697 Analysis feature update error
https://crowdin.com/translate/freecad/559/en-pl#6593705 Only one face for rectangular local coordinate system!
https://crowdin.com/translate/freecad/559/en-pl#6593709 Only one face for local coordinate system!
https://crowdin.com/translate/freecad/559/en-pl#6593715 Only transformable faces can be selected! Apply displacement boundary condition to surface first then apply local coordinate system to surface
https://crowdin.com/translate/freecad/559/en-pl#6597314 Whether the boundary condition defines a constant potential
https://crowdin.com/translate/freecad/559/en-pl#6597334 Whether the boundary condition defines a farfield potential
https://crowdin.com/translate/freecad/559/en-pl#6597356 Whether the boundary condition is for the electric force
https://crowdin.com/translate/freecad/559/en-pl#6593991 Fluid boundary condition
https://crowdin.com/translate/freecad/559/en-pl#6593179 Initial temperature
https://crowdin.com/translate/freecad/559/en-pl#6593997 Plane multi-point constraint
https://crowdin.com/translate/freecad/559/en-pl#6593191 Pulley constraint
https://crowdin.com/translate/freecad/559/en-pl#6593197 Spring
https://crowdin.com/translate/freecad/559/en-pl#6594167 TaskFemHeatflux
https://crowdin.com/translate/freecad/559/en-pl#6619751 Electromagnetic boundary conditions

FEA-eng, 2023-08-09:
Part 2:
https://crowdin.com/translate/freecad/559/en-pl#6619753 Electromagnetic boundary conditions
https://crowdin.com/translate/freecad/559/en-pl#6594399 Geometrical analysis features
https://crowdin.com/translate/freecad/559/en-pl#6594407 Thermal boundary conditions and loads
https://crowdin.com/translate/freecad/559/en-pl#6606405 Body heat source
https://crowdin.com/translate/freecad/559/en-pl#6606411 Creates a body heat source
https://crowdin.com/translate/freecad/559/en-pl#6606423 Centrifigual load
https://crowdin.com/translate/freecad/559/en-pl#6606433 Creates a centrifugal load
https://crowdin.com/translate/freecad/559/en-pl#6606447 Creates an electrostatic potential boundary condition
https://crowdin.com/translate/freecad/559/en-pl#6606463 Creates a flow velocity boundary condition
https://crowdin.com/translate/freecad/559/en-pl#6606497 Creates an initial pressure condition
https://crowdin.com/translate/freecad/559/en-pl#6606537 Tie constraint
https://crowdin.com/translate/freecad/559/en-pl#6606547 Creates a tie constraint
https://crowdin.com/translate/freecad/559/en-pl#6611299 Creates a magnetization boundary condition
https://crowdin.com/translate/freecad/559/en-pl#6593149 Bearing constraint
https://crowdin.com/translate/freecad/559/en-pl#6593151 Creates a bearing constraint
https://crowdin.com/translate/freecad/559/en-pl#6593153 Contact constraint
https://crowdin.com/translate/freecad/559/en-pl#6593183 Plane multi-point constraint
https://crowdin.com/translate/freecad/559/en-pl#6593183 Creates a pressure load acting on a face
https://crowdin.com/translate/freecad/559/en-pl#6593211 Create a local coordinate system on a face
https://crowdin.com/translate/freecad/559/en-pl#6593311 Make heat flux load on face
https://crowdin.com/translate/freecad/559/en-pl#6593313 Make initial temperature condition on body
https://crowdin.com/translate/freecad/559/en-pl#6593317 Make pressure load on face
https://crowdin.com/translate/freecad/559/en-pl#6593323 Make temperature boundary condition on face
https://crowdin.com/translate/freecad/559/en-pl#6593495 Hide analysis features when open result dialog
https://crowdin.com/translate/freecad/559/en-pl#6593889 Analysis feature properties
https://crowdin.com/translate/freecad/559/en-pl#6497043 Edit analysis feature
https://crowdin.com/translate/freecad/559/en-pl#6498312 Force load
https://crowdin.com/translate/freecad/559/en-pl#6593985 Contact constraint
https://crowdin.com/translate/freecad/559/en-pl#6593157 Displacement boundary condition
https://crowdin.com/translate/freecad/559/en-pl#6593161 Fixed boundary condition
https://crowdin.com/translate/freecad/559/en-pl#6593187 Pressure load
https://crowdin.com/translate/freecad/559/en-pl#6593203 Temperature boundary condition
https://crowdin.com/translate/freecad/559/en-pl#6593209 Local coordinate system
https://crowdin.com/translate/freecad/559/en-pl#6497066 Analysis features
https://crowdin.com/translate/freecad/559/en-pl#6594391 Electrostatic boundary conditions
https://crowdin.com/translate/freecad/559/en-pl#6594393 Electrostatic boundary conditions
https://crowdin.com/translate/freecad/559/en-pl#6594395 Fluid boundary conditions
https://crowdin.com/translate/freecad/559/en-pl#6594397 Fluid boundary conditions
https://crowdin.com/translate/freecad/559/en-pl#6594401 Geometrical analysis features
https://crowdin.com/translate/freecad/559/en-pl#6594403 Mechanical boundary conditions and loads
https://crowdin.com/translate/freecad/559/en-pl#6594405 Mechanical boundary conditions and loads
https://crowdin.com/translate/freecad/559/en-pl#6594409 Thermal boundary conditions and loads
https://crowdin.com/translate/freecad/559/en-pl#6594411 Analysis features without solver
https://crowdin.com/translate/freecad/559/en-pl#6594413 Analysis features without solver
https://crowdin.com/translate/freecad/559/en-pl#6606439 Electrostatic potential boundary condition
https://crowdin.com/translate/freecad/559/en-pl#6606455 Flow velocity boundary condition
https://crowdin.com/translate/freecad/559/en-pl#6606473 Initial flow velocity condition
https://crowdin.com/translate/freecad/559/en-pl#6606483 Creates initial flow velocity condition
https://crowdin.com/translate/freecad/559/en-pl#6606491 Initial pressure condition
https://crowdin.com/translate/freecad/559/en-pl#6606505 Section print feature
https://crowdin.com/translate/freecad/559/en-pl#6606509 Creates a section print feature
https://crowdin.com/translate/freecad/559/en-pl#6606519 Gravity load
https://crowdin.com/translate/freecad/559/en-pl#6606527 Creates a gravity load
https://crowdin.com/translate/freecad/559/en-pl#6610735 Current density boundary condition
https://crowdin.com/translate/freecad/559/en-pl#6610737 Creates a current density boundary condition
https://crowdin.com/translate/freecad/559/en-pl#6610749 Electromagnetic boundary conditions
https://crowdin.com/translate/freecad/559/en-pl#6611297 Magnetization boundary condition

FEA-eng, 2023-08-09:
Note that I not only removed "constraint" from most of them (apart from those that are actually constraints) but also removed "FEM" (IMO unnecessary) and rephrased them in general so that they make more sense for both professionals and amateurs.

fandaL, 2023-08-09:
Basically, I agree to remove excessive use of “constraint” and mostly follow a nomenclature which is used by solvers.

This naming has grown with FreeCAD since times when it had only few “constraints”. I do not orientate smoothly in the FreeCAD code, but maybe this naming follows how the code is structured. If we rename the gui tools (objects), should we rename the objects inside the code as well and check if it makes sense on the code level? To make the code and gui consistent, but e.g. breaking old macros?
Just a quick search:
https://github.com/FreeCAD/FreeCAD/tree/master/src/Mod/Fem/femobjects
https://github.com/FreeCAD/FreeCAD/blob/master/src/Mod/Fem/femcommands/commands.py 

It would be good if @berndhahnebach (or somebody who worked with FreeCAD FEM code structure) can comment on this.