FreeCAD/FreeCAD #10135: Removing "Constraints without solver" from the FEM workbench

FEA-eng, 2023-08-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.0.33668 +7 (Git)
Build type: Release
Branch: (HEAD detached at 0.21)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Polish/Poland (pl_PL)
[/code]
```

### Subproject(s) affected?

FEM

### Problem description

Hi,

there's a whole group of "Constraints without solver". They have been unusable for quite some time and there are no perspectives for this to change in the near future. We don't have to remove their source code but I think that we should at least remove them from the GUI. After all, there's no point in having non-functional tools there.

Marking @kaktusus since it's part of our discussion about the improvements in the FEM workbench ultimately leading to better translations.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2023-08-09:
Forum discussion: https://forum.freecad.org/viewtopic.php?t=80299

luzpaz, 2023-08-23:
@oliveroxtoby does this change impacts cfdof and would you mind weighing in ?

oliveroxtoby, 2023-08-23:
> @oliveroxtoby does this change impacts cfdof and would you mind weighing in ?

No, CfdOF is almost entirely independent of FEM now. But it seems like a good idea!

luzpaz, 2023-08-23:
Who wants to pick up the mantle on this ? (@FEA-eng have you made a PR before? Would you like to learn if not?)

kaktusus, 2023-08-23:
I think I can handle these edits 

assign me to this task
But first I have to finish another started subject

kaktusus, 2023-08-30:
@FEA-eng can you prepare a list of expected changes as in this thread: https://github.com/FreeCAD/FreeCAD/discussions/10217#discussioncomment-6715428

FEA-eng, 2023-08-30:
@kaktusus 

Here are the tools that should be removed (either completely from the source code or at least from the GUI):
- Fluid boundary condition
- Constraint bearing
- Constraint gear
- Constraint pulley

This whole category should be removed:

[image]

closed 2023-09-01 by FEA-eng