FreeCAD/FreeCAD #10136: [Problem] Regressions with the new theme handling

wwmayer, 2023-08-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
OS: Ubuntu 22.04.3 LTS (XFCE/xubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.33756 (Git)
Build type: debug
Branch: master
Hash: d150fa7164e07651b503e2fd16e6eb55a9849a77
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.5.1
Locale: German/Germany (de_DE)
```
```

### Subproject(s) affected?

Core

### Problem description

There are several issues with the new theme handling:
* Illogical user interface: In the General tab there is  a setting to choose between Classic, Dark and Light themes. and in the Display section there is a separate Theme tab. For the user it's not obvious how they are related -- especially as they are can be combined.
* Whenever switching between Classic, Dark or Light theme the report view will become invisible
* The Classic, Dark or Light theme modifies the background color of the 3d view. If after switching to another theme and adjusting the background colors under Display > Colors then the colors won't be accepted and rejected to the default colors of the theme. The only way to get the custom colors accepted is to close the the preferences dialog after changing the theme and reopening it again. This is extremely annoying behaviour.
* Whenever choosing a style sheet in the Theme tab a warning "Hello" is written to the report view

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

chennes, 2023-08-09:
@PaddleStroke

PaddleStroke, 2023-08-10:
- ui : The Theme page has a text to explain it's purpose 'This page let you customize your current theme. ...' I think the situation is now confusing because we still have the stylesheet combobox in there and this stylesheet combobox has been acting as 'kind-of-theme' for a long time. Besides the color pickers does nothing currently so that doesn't help.

So the general idea is that general page will have the Theme combobox to make theme selection proiminent. And the theme page will give the user theme-customization settings. Maybe the theme page can also have the theme combobox to make things clearer?

- report view hidding : I think this is because hidding the report view is included in the themes (they are preference packs). @MisterMakerNL can you confirm this?

- Indeed this is annoying. It's because once you changed the theme once, themeChanged bool is true so every-time you click on apply again, it will apply the theme again overriding the background color change. I will make a fix for this

- This is a coding leftover I'll remove it.

PaddleStroke, 2023-08-10:
Additional note: the end-goal is to remove the stylesheet combobox altogether. All the stylesheet should be transfered to themes. User should have to worry about what is a stylesheet and what's the difference with theme and so on.

luzpaz, 2023-08-13:
Is this ticket ready to be closed ?

wwmayer, 2023-08-13:
No. The point below still must be addressed:

> Whenever switching between Classic, Dark or Light theme the report view will become invisible

MisterMakerNL, 2023-08-13:
> No. The point below still must be addressed:
> 
> > Whenever switching between Classic, Dark or Light theme the report view will become invisible

I think I missed that in the cfg file.

PaddleStroke, 2023-08-16:
I can't find any related settings in the cfg file. Any idea where this is defined by the pref pack?

MisterMakerNL, 2023-08-16:
As I remeber there was a "on setting" and I removed it because I didn't want to change the screen layout. So I thought I maybe missed the off setting but it seems like a bug now.

chennes, 2023-10-01:
@PaddleStroke did this get resolved, or are you still looking for that setting?

PaddleStroke, 2023-10-02:
> Whenever switching between Classic, Dark or Light theme the report view will become invisible

This problem is still there. Apparantly it is a problem with the pref packs. I had a look at them but couldn't locate the setting hidding the report view.

luzpaz, 2023-10-13:
Since we have a lot of stylesheet related focus...lets keep an eye on this ticket as well

chennes, 2023-11-01:
Sure.

luzpaz, 2023-11-14:
> This problem is still there. Apparantly it is a problem with the pref packs. I had a look at them but couldn't locate the setting hidding the report view.

Need diagnosis here.

bgbsww, 2023-12-09:
Hmmm.  Unable to duplicate.  Fixed maybe?  I can't make the report view disappear, at least from the edit -> preferences -> general screen settings.

```
OS: Ubuntu 22.04.3 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35335 (Git)
Build type: Debug
Branch: bgbsww-patch-22
Hash: 0e5d2c46f0d2661d9f8b6692ae0bfad6c5504762
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.5.2
Locale: English/United States (en_US)
Installed mods: 
  * FreeCAD_BearingsWB 0.0.2
  * Assembly4 0.50.4
  * fcgear 1.0.0
  * Plot 2022.4.17
  * DynamicData 2.54.0
  * BIM 2021.12.0
  * parts_library
```

chennes, 2024-07-19:
I also cannot reproduce anymore, using
```
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38239 (Git)
Build type: Release
Branch: openglAddMissingIncludeToPCH
Hash: 076d770efb19aa7da8bb23d3305ce0d04209e95b
Python 3.12.4, Qt 6.7.2, Coin 4.0.2, Vtk 9.3.0, OCC 7.8.0
Locale: English/United States (en_US)
```
I'm going to close this one as completed: if there are further issues to address let's open individual Issues for them.

closed 2024-07-19 by chennes