FreeCAD/FreeCAD #10144: [Problem] FreeCAD freezes when clicking inside spreadsheet

peacememories, 2023-08-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
[code]
OS: macOS 13.3
Word size of FreeCAD: 64-bit
Version: 0.21.0.33675 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.0)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
[/code]
```

### Subproject(s) affected?

Spreadsheet

### Problem description

I've run into a problem where FreeCAD completely freezes up when clicking on a cell in a spreadsheet. Interestingly, this *only* seems to happen if the start page is closed.

While FreeCAD is frozen the CPU utilization is 100%, while the memory consumption stays normal, around 400mb.

## Reproducing
* Open FreeCAD
* Create a new document
* Switch to the spreadsheet toolbar
* Press the "Create spreadsheet" button
* You can click around the spreadsheet or edit, but this is not necessary for reproducing the bug
* Press the "x" on the "Start page" document, the selected cell in the spreadsheet should now turn grey
* Try to click inside the spreadsheet
* The program should now be locked up

It does not appear to make a difference if you close the start page before creating the spreadsheet, or if you switch to the 3d view before closing the start page. If you try to click inside a spreadsheet while the start page is not open the whole program freezes.

## Similar issues

there is https://github.com/FreeCAD/FreeCAD/issues/8061, but this seems somewhat different since the program does not start consuming more memory and does not recover even after many minutes.

### Anything else?

I am using the Apple Silicon build of FreeCAD on a 2023 m2 pro machine with 16gb ram.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

chennes, 2023-08-09:
I can't reproduce this issue, on nearly the exact same hardware setup. I'll check to see if any of the other Maintainers have access to a similar system, or can reproduce on different hardware.

peacememories, 2023-08-09:
Fwiw I don't have the same problem on the link-stage 3 branch (though that also runs via Rosetta).
I tried running the Intel version of the main release for comparison but did not get it to start for some reason.
I am running a very clean setup, the laptop is two weeks old and not even running homebrew, so I'm a bit confused why it would be uniquely broken 😅

funkyfourier, 2023-08-10:
I am having the exact same issue on a Macbook Pro with M1 Max running macOS 13.3.1 and the latest 0.21 build of FreeCAD. For me it happens regardless of the start page being open or not. It is enough to just open a new spreadsheet and clicking a few cells here and there, and the spinning beach ball of death will appear and the application is locked up.

luzpaz, 2023-08-13:
Please everybody always post your full [About](https://wiki.freecad.org/About) info. Trust me, we solve things a lot faster that way.

JohnOCFII, 2023-08-13:
I was not able to recreate the problem.  MacBook Pro 16" with M1 Pro.

```
OS: macOS 13.5
Word size of FreeCAD: 64-bit
Version: 0.21.0.33675 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.0)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
```

FilipSzkandera, 2023-08-22:
Same problem happening here

`[code]
OS: macOS 13.5.1
Word size of FreeCAD: 64-bit
Version: 0.21.0.33675 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.0)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * Behave-Dark-Colors 0.1.1 (Disabled)
  * freecad.gears 1.0.0
  * A2plus 0.4.60n
  * Manipulator 1.5.0
  * kicadStepUpMod 10.19.4
  * fasteners 0.4.60
  * ExtremeProDark 2.9.0 (Disabled)
  * Assembly4 0.50.2 (Disabled)
[/code]
`

It happens randomly (though very often). I'm just clicking into empty fields with a mouse cursor and the app will become unresponsive. This makes the spreadsheets unusable for me (I wanted to write just 3 aliases and it froze 4 times).

syntaxfoundry, 2023-08-29:
For me it happens almost all the time when I try to add or edit some cells in the spreadsheet. The beachball shows and freezes the app. The Start Page is still open in my case. CPU usage jumps to 100% but memory usage stays the same. 
I am using an MacBook Pro M1.

[code]
OS: macOS 13.4.1
Word size of FreeCAD: 64-bit
Version: 0.21.0.33675 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.0)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
[/code]

chennes, 2023-08-30:
I wonder if it's any relative of the problem we saw in #8061.

justin-cotarla, 2023-08-31:
I'm having the same problem, albeit with some variations:
- The app crashes once I interact with the spreadsheet, whether the start page is open or not.
- I cannot close the start page. Pressing the little x does nothing, and the UI in the page remains interactive.

The same behavior occurs on the unsigned x86 build.

I'm running Freecad on an M1 Mac Mini.

```
[code]
OS: macOS 13.4
Word size of FreeCAD: 64-bit
Version: 0.21.0.33675 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.0)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * fcgear 1.0.0
[/code]

I've also tried this on an M1 Max Macbook Pro with the same result.

```
[code]
OS: macOS 13.5.1
Word size of FreeCAD: 64-bit
Version: 0.21.0.33675 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.0)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
[/code]
```

```

syntaxfoundry, 2023-09-02:
I found a workaround:
I noticed that the crash only happens when clicking a cell in the spreadsheet. However, clicking to select a whole column or row seems to work fine. Doing this also allows me then to select a cell by using the the arrow keys. Editing any cell using the keyboard works fine as long as you do not click a cell.

Bill-Howell, 2023-09-04:
[code]
OS: macOS 13.5.1
Word size of FreeCAD: 64-bit
Version: 0.21.0.33675 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.0)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United States (en_US) [ OS: C/Default (C) ]
[/code]

I bet I have the same problem... 
The program is really not parametric with this function not working.   I have a short (about 100 seconds) video showing the problem but it is too big to post here.

I am new to this program and to this process.  Any help will me much appreciated.

berkakinci, 2023-09-05:
I have this issue as well.  Not exactly as in original reproduction instructions, but a hang on clicking around the spreadsheet.
Details at force quit are attached.

```
OS: macOS 13.4.1
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * kicadStepUpMod 10.19.4
```

[Hang of FreeCAD-0.21.1-33694.txt](https://github.com/FreeCAD/FreeCAD/files/12518185/Hang.of.FreeCAD-0.21.1-33694.txt)

chennes, 2023-09-05:
It's very interesting that the code seems to be locking up at something so similar to the problem we were having on Windows last year with the accessibility:
```
11  CopyElementAtPosition + 152 (AppKit + 3703800) [0x1a08d13f8] 1-11
11  -[NSApplication(NSApplicationAccessibility) accessibilityHitTest:] + 264 (AppKit + 3930952) [0x1a0908b48] 1-11
```
If just clicking around, why a call to CopyElementAtPosition? On Windows, the Accessibility code was the culprit, it ended up scanning the entire declared size of the spreadsheet.

carlosbaraza, 2023-09-13:
Same problem here with M2:

About
```
OS: macOS 13.2
Word size of FreeCAD: 64-bit
Version: 0.21.0.33675 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.0)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * Freecad-Built-in-themes-beta 1.2.2
  * Behave-Dark-Colors 0.1.1
  * Dracula 0.0.5
  * ExtremeProDark 2.9.0
  * ExtremeProLight 1.6.0
```

Backtrace when force closing the app:
```
Process:          freecad [99974] [unique pid 5370197]
UUID:             E11106B6-DDFA-3AE8-9E77-E7AB0CB3A362
Path:             /Applications/FreeCAD.app/Contents/Resources/bin/freecad
Codesigning ID:   freecad
Team ID:          394HN54UJM
Shared Cache:     3366B98C-6B8A-3546-8233-DC167320439F slid base address 0x184bc0000, slide 0x4bc0000 (System Primary)
Architecture:     arm64
Footprint:        413.54 MB
Time Since Fork:  375s
Num samples:      11 (1-11)
CPU Time:         1.006s (3.4G cycles, 12.7G instructions, 0.27c/i)
Note:             5 idle work queue threads omitted

  Thread 0x22c9ee6    DispatchQueue "com.apple.main-thread"(1)    Thread name "CrBrowserMain"    11 samples (1-11)    priority 47 (base 47)    cpu time 1.000s (3.4G cycles, 12.6G instructions, 0.27c/i)
  <process frontmost, thread QoS user interactive (requested user interactive), process unclamped, process received importance donation from WindowServer [368], IO tier 0>
  11  start + 2544 (dyld + 24144) [0x184c63e50] 1-11
    11  main + 5064 (freecad + 17768) [0x102428568] 1-11
      11  ??? (libqcocoa.dylib + 182884) [0x107aeca64] 1-11
        11  -[NSApplication run] + 464 (AppKit + 183800) [0x1882e4df8] 1-11
          11  -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 728 (AppKit + 231880) [0x1882f09c8] 1-11
            11  _DPSNextEvent + 632 (AppKit + 235576) [0x1882f1838] 1-11
              11  _BlockUntilNextEventMatchingListInModeWithFilter + 72 (HIToolbox + 203564) [0x18e74bb2c] 1-11
                11  ReceiveNextEventCommon + 672 (HIToolbox + 204260) [0x18e74bde4] 1-11
                  11  RunCurrentEventLoopInMode + 292 (HIToolbox + 204704) [0x18e74bfa0] 1-11
                    11  CFRunLoopRunSpecific + 612 (CoreFoundation + 522360) [0x18506b878] 1-11
                      11  __CFRunLoopRun + 2264 (CoreFoundation + 526500) [0x18506c8a4] 1-11
                        11  __CFRunLoopDoSource1 + 520 (CoreFoundation + 532152) [0x18506deb8] 1-11
                          11  __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE1_PERFORM_FUNCTION__ + 60 (CoreFoundation + 532376) [0x18506df98] 1-11
                            11  mshMIGPerform + 204 (HIServices + 48040) [0x18a7f3ba8] 1-11
                              11  _XCopyElementAtPosition + 356 (HIServices + 340212) [0x18a83b0f4] 1-11
                                11  _AXXMIGCopyElementAtPosition + 444 (HIServices + 188992) [0x18a816240] 1-11
                                  11  CopyElementAtPosition + 152 (AppKit + 3716144) [0x188643430] 1-11
                                    11  -[NSApplication(NSApplicationAccessibility) accessibilityHitTest:] + 256 (AppKit + 3943196) [0x18867ab1c] 1-11
                                      1   ??? (libQt5Gui.5.15.8.dylib + 44164) [0x102ec2c84] (running) 1
                                      1   ??? (libQt5Gui.5.15.8.dylib + 44180) [0x102ec2c94] (running) 2
                                      1   SpreadsheetGui::SheetTableViewAccessibleInterface::childAt(int, int) const + 0 (SpreadsheetGui.so + 267904) [0x2b4009680] (running) 3
                                      2   ??? (libQt5Gui.5.15.8.dylib + 44164) [0x102ec2c84] (running) 4-5
                                      1   ??? (libQt5Widgets.5.15.8.dylib + 1528876) [0x10544142c] (running) 6
                                      1   QAccessibleObject::object() const + 20 (libQt5Gui.5.15.8.dylib + 42668) [0x102ec26ac] (running) 7
                                      1   QAccessibleObject::object() const + 4 (libQt5Gui.5.15.8.dylib + 42652) [0x102ec269c] (running) 8
                                      1   ??? (libQt5Gui.5.15.8.dylib + 44180) [0x102ec2c94] (running) 9
                                      1   QAccessibleObject::object() const + 20 (libQt5Gui.5.15.8.dylib + 42668) [0x102ec26ac] (running) 10
                                      1   DYLD-STUB$$QAccessibleObject::isValid() const + 4 (libQt5Widgets.5.15.8.dylib + 2385652) [0x1055126f4] (running) 11
```

carlosbaraza, 2023-09-13:
I found a way to reproduce it in my machine.

- Windows > Tile mode
- Open 3D model in one tile and Spreadsheet in another tile
- Click on 3D model and then click back in the spreadsheet

Another way to reproduce it in my machine is to select all the the cells (cmd + A)

Workaround for me is not to use the Tile Window, and click back and forth between speadsheet and part from the combo tree view.

Bill-Howell, 2023-09-14:
Thanks for the workaround tip!Bill Howell(479) 236-3630Sent from my iPad

carlosbaraza, 2023-09-15:
Actually, my workaround doesn't work, it still crashes. The best workaround is to click the column and use the arrow keys like @syntaxfoundry said.

pw-64, 2023-09-20:
I am having the same problem. If a cell in the spreadsheet is highlighted / active / selected and you click on another cell using the mouse, the program immediately freezes up.

[code]
OS: macOS 13.5
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
[/code]

Scatchell, 2023-09-21:
I'm having the same problem. For me it freezes whenever the windows are tiled (whether the start page is open or not) immediately when I click any cell in the spreadsheet. The only work around that's working for me is switching back and forth between full windows (spreadsheet, then visual of the part)

[code]
OS: macOS 13.5.2
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * A2plus 0.4.60n
  * Curves 0.6.12
[/code]

pw-64, 2023-09-21:
Have you tried restarting?

Think71, 2023-09-25:
Same problem here. 
After opening a Spreadsheet, created with a former FreeCAD version, the application freezes immediate after clicking into a cell.

I installed the older FreeCAD versions from 0.21.1 to 0.21.0 and 0.20.2, to reproduce the behavior. FreeCAD 0.20.2 does not freeze.

First Release on my Computer wich freeze:
[code]
OS: macOS 12.6.9
Word size of FreeCAD: 64-bit
Version: 0.21.0.33675 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.0)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * sheetmetal 0.2.59
[/code]

ugr, 2023-09-30:
Same problem here. Everythink works fine except the spreadsheet. I'm using  the current release, downloaded and installed today:

[code]
OS: macOS 14.0
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * Nodes 0.1.36 (Disabled)
  * parts_library
  * fcgear 1.0.0
  * Render 2023.8.18
  * BOLTSFC 2022.11.5
  * Defeaturing 1.2.0
  * Glass (Disabled)
  * kicadStepUpMod 10.19.5
  * fasteners 0.4.66
  * ExplodedAssembly
  * Curves 0.6.13
  * Assembly4 0.50.2
  * Assembly3 0.12.0
[/code]

The workaround by clicking on the column and the use the keyboard works. It seems that the mouse selection triggers some events that are dispatched recursivly.

dancesWithMachines, 2023-10-01:
Same problem in my case.

Scenario:
1. Open spreadsheet.
2. Click on any cell until program freezes (often reproduces after second try).

Hardware: 2020 M1 MacBook Air

[code]
OS: macOS 14.0
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
[/code]

I tried updating MacOS from 13.3.1 to 14.0 and the problem still occurs.

luzpaz, 2023-10-11:
bumping for progress

JohnOCFII, 2023-10-11:
First step would be to try to focus on what is in common by people having the problem or not having the problem.  I continue to use FreeCAD every few days on my M1 MacBook Pro without issue.  I always use the built-in trackpad. Starting to wonder if this problem only occurs to people using mice? (Wacky, I know -- but haven't see a good way to narrow this down yet.)

```
OS: macOS 13.6
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34291 (Git)
Build type: Release
Branch: master
Hash: 71f1e6b09baa45812026832dfd74e59b5d4d1d35
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
```

ugr, 2023-10-11:
> First step would be to try to focus on what is in common by people having the problem or not having the problem. I continue to use FreeCAD every few days on my M1 MacBook Pro without issue. I always [...]

This could be possible. I'm using FreeCAD on a MacMini M1, with a Logitech MX Master 3 mouse.

As long as i don't click inside a spreadsheet cell and only use the keyboard to select and edit cells, everything works fine.

But this bug can also happen if I open a spreadsheet tab, go back to the 3D model tab and click somewhere there. Usually everything works, but sometimes the mouse click also triggers this freeze. But only if the spreadsheet is open in another tab. With spreadsheet closed, I never had a freeze.

So it obviously the bug is something with the mouse click event dispatching and the spreadsheet.

berkakinci, 2023-10-12:
Argh...  I went to reproduce this yet again. (Checking if mouse vs trackpad is a difference.)
Now, I can't reproduce w/ trackpad or mouse.
Also a change between trackpad and mouse for me: external display is attached to dongle along with the mouse.  So neither of those seem to have an effect.
I used to experience this every single time I tried using FreeCAD.  All my designs use spreadsheets.

One difference: an OS update since last try.

```
OS: macOS 13.6
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * kicadStepUpMod 10.19.4
```

Also a side note: I see a lack of info in FreeCAD about: There is no mention of arch.   I'm on arm64.  I wondered if those without issue could have been using x86, but I couldn't tell.

berkakinci, 2023-10-13:
> In macOS terminal, you can enter
> 
> `file /Applications/FreeCAD.app/Contents/Resources/bin/freecad`
> 
> to get informations about the executable.

Well, I know I'm using the arm64 binaries.  I just would like to know others' binary versions too.  It appears that's not the difference we're looking for.  I even had the same binary go from always hanging to seemingly working.  The only difference I can imagine is a reboot and a MacOS (minor) update.

On the MacOS version, I see a mix, so that doesn't seem like the common thing.

CaptainSmileyXL, 2023-10-13:
FYI...  Similar thing happens on Win11 x64 too.  It hangs anytime make changes to the spreadsheet (even non-calculation cells).  It eventually returns (avg about 2 min).  I'm using 0.21.1

berkakinci, 2023-10-13:
> file /Applications/FreeCAD.app/Contents/Resources/bin/freecad

Oh, another thing I found weird: This application is arm64, but I found one "bash" instance running on intel arch.  It has "/Applications/3D CAD/FreeCAD 0.21.app/Contents/MacOS/FreeCAD" open, and it comes and goes with launching the app from launcher.

MikDac, 2023-10-17:
Just reporting the same issue with freezing spreadsheet here.

[code]
OS: macOS 13.5.2
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * Freecad-Built-in-themes-beta 1.2.2
  * A2plus 0.4.60n
  * sheetmetal 0.3.0
[/code]

It works for some time but most often it ends up freezing with a restart needed.

I use the magic touchpad, not the mouse.

ryanmeador, 2023-10-19:
Same problem for me.  This is a regression from 0.20.

I have a 2019 16" MacBook Pro (i.e. Intel) and Logitech MX Master 3 mouse.

I've noticed some really strange things that might help:

1. I use BetterTouchTool to do some mouse stuff (tested both versions 4.272 and 4.286).  I couldn't reproduce the issue when I closed BTT, but almost immediately reproduced it upon reopening BTT (though see strange behavior 3).
1. I was not able to reproduce the problem using the MacBook's built-in trackpad, even when BTT was running.  However, I _was_ able to repro it while using the mouse to move the cursor and doing the clicking with the trackpad.
1. Even using the mouse + BTT, I can apparently click cells forever without triggering the crash if I keep my selections in the first 9 rows.  If I click into row 10 or greater, within a few clicks it will crash (not always the first click).  I was keeping my clicks in columns A and B; I do not know if a similar effect manifests for columns further to the right.

```
OS: macOS 13.4
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * Assembly3 0.12.0
```

ugr, 2023-10-20:
I use BetterSnap-Tool, and can confirm the behavior: if BetterSnap is running, FreeCAD hangs if I click on any cell below the 10th row (all above is ok). If I terminate BetterSnap, I don't get any hangs or crashes.

This is really weird, but obviously this is not directly a FreeCAD bug but a combination of FreeCAD and BetterSnap/BetterTouch. BetterTouch/Snap install global mouse and keyboard handlers so that every mouse event is routed through this software.

I think this is more basic problem of Qt5 and BetterTouch/Snap.

ugr, 2023-10-20:
> I use BetterSnap-Tool, and can confirm the behavior: if BetterSnap is running, FreeCAD hangs if I click on any cell below the 10th row (all above is ok). If I terminate BetterSnap, I don't get any hangs or crashes.

I found a simple work-around: just open FreeCAD and select in BetterSnap/Touch "Disable BetterSnapTool for current application" - then it works.

MikDac, 2023-10-20:
> I found a simple work-around: just open FreeCAD and select in BetterSnap/Touch "Disable BetterSnapTool for current application" - then it works.

Just tested (briefly admittedly), and seems to work! Super great, really appreciate your efforts!

freespace, 2023-10-21:
> I found a simple work-around: just open FreeCAD and select in BetterSnap/Touch "Disable BetterSnapTool for current application" - then it works.

While I don't use BetterSnap/Touch, I do use Rectangle, and on a hunch I quite Rectangle and the freeze problems stopped. Before that I could barely edit more than 5-6 cells before FreeCAD would freeze.

Hypothesis: some interaction between tools like Rectangle and BetterSnap/Touch and however FreeCAD/QT is capturing/handling keyboard or mouse inputs.

Scatchell, 2023-10-23:
Wow, this was some great collaborative investigation work 👏👏👏

I've tested this and closing the rectangle app fixes it for me to. As soon as I open it, in 1 or 2 clicks I get the crash.

I guess there is still some issue with the event handling somewhere, but at least there is an easy workaround for now!

syntaxfoundry, 2023-10-23:
I can confirm that for me Rectangle is triggering the issue. Quitting Rectangle allows me to click around in the spreadsheet everywhere without issue. As soon as Rectangle is running, it locks up immediately.

For me, it does not help telling Rectangle to ignore FreeCad. When I do this, I can click a few cells without issues, but eventually clicking a cell too many or too fast after each other still locks it up.

shandyba, 2023-10-25:
> I found a simple work-around: just open FreeCAD and select in BetterSnap/Touch "Disable BetterSnapTool for current application" - then it works.

Same story with the "Magnet" tool: if it's running then FreeCAD's spreadsheet would hang in case of clicking cells below 10th line or so. "Disable FreeCAD" option for Magnet fixes it!

JohnOCFII, 2023-10-25:
Ah yes!  Magnet.  I sometimes use that, but not regularly.  I launched it, and confirmed the issue when Magnet is running.

First part of the crash dump:

```
Date/Time:        2023-10-25 08:12:20.512 -0500
End time:         2023-10-25 08:16:20.539 -0500
OS Version:       macOS 13.6 (Build 22G120)
Architecture:     arm64e
Report Version:   40
Incident Identifier: 22227CA4-120F-47E9-8CD8-4BF65DAFA4EE
Share With Devs:  Yes

Data Source:      Stackshots
Shared Cache:     C26BE8CD-619E-3513-8673-3FF826317005 slid base address 0x19734c000, slide 0x1734c000 (System Primary)
Shared Cache:     94BC1025-393F-3940-A142-6BF004C71C39 slid base address 0x19ec10000, slide 0x1ec10000 (DriverKit)
Shared Cache:     75DD4664-3CE0-3CEE-8A6F-327DB863E105 slid base address 0x7ff80a6e4000, slide 0xa6e4000 (Rosetta)

Command:          freecad
Path:             /Applications/Other Apps/FreeCAD.app/Contents/Resources/bin/freecad
Codesigning ID:   freecad
Team ID:          394HN54UJM
Architecture:     arm64
Parent:           bash [15041]
PID:              15047
Time Since Fork:  270s

Event:            hang
Duration:         240.03s
Duration Sampled: 1.10s (process was unresponsive for 239 seconds before sampling)
Steps:            11 (100ms sampling interval)

Hardware model:   MacBookPro18,1
Active cpus:      10
HW page size:     16384
VM page size:     16384

Time Since Boot:  322916s
Time Awake Since Boot: 154926s
Time Since Wake:  799s

Fan speed:        0 rpm
Total CPU Time:   1.731s (4.5G cycles, 15.9G instructions, 0.28c/i)
Advisory levels:  Battery -> 3, User -> 2, ThermalPressure -> 0, Combined -> 2
Free disk space:  585.71 GB/926.35 GB, low space threshold 3072 MB
Vnodes Available: 84.44% (222209/263168)

Preferred User Language: en-US
Country Code:     US
OS Cryptex File Extents: 167

--------------------------------------------------
Timeline format: stacks are sorted chronologically
Use -i and -heavy to re-report with count sorting
--------------------------------------------------

Heaviest stack for the main thread of the target process:
  11  start + 2236 (dyld + 24360) [0x1973f3f28]
  11  main + 5064 (freecad + 17768) [0x104d04568]
  11  ??? (libqcocoa.dylib + 182884) [0x10a48ca64]
  11  -[NSApplication run] + 464 (AppKit + 184188) [0x19aa42f7c]
  11  -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 716 (AppKit + 232216) [0x19aa4eb18]
  11  _DPSNextEvent + 636 (AppKit + 235900) [0x19aa4f97c]
  11  _BlockUntilNextEventMatchingListInModeWithFilter + 76 (HIToolbox + 199044) [0x1a107a984]
  11  ReceiveNextEventCommon + 648 (HIToolbox + 199724) [0x1a107ac2c]
  11  RunCurrentEventLoopInMode + 292 (HIToolbox + 200176) [0x1a107adf0]
  11  CFRunLoopRunSpecific + 612 (CoreFoundation + 513096) [0x197828448]
  11  __CFRunLoopRun + 2240 (CoreFoundation + 517212) [0x19782945c]
  11  __CFRunLoopDoSource1 + 520 (CoreFoundation + 522876) [0x19782aa7c]
  11  __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE1_PERFORM_FUNCTION__ + 60 (CoreFoundation + 523100) [0x19782ab5c]
[... 30 lines cut ...]
            11  _DPSNextEvent + 636 (AppKit + 235900) [0x19aa4f97c] 1-11
              11  _BlockUntilNextEventMatchingListInModeWithFilter + 76 (HIToolbox + 199044) [0x1a107a984] 1-11
                11  ReceiveNextEventCommon + 648 (HIToolbox + 199724) [0x1a107ac2c] 1-11
                  11  RunCurrentEventLoopInMode + 292 (HIToolbox + 200176) [0x1a107adf0] 1-11
                    11  CFRunLoopRunSpecific + 612 (CoreFoundation + 513096) [0x197828448] 1-11
                      11  __CFRunLoopRun + 2240 (CoreFoundation + 517212) [0x19782945c] 1-11
                        11  __CFRunLoopDoSource1 + 520 (CoreFoundation + 522876) [0x19782aa7c] 1-11
                          11  __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE1_PERFORM_FUNCTION__ + 60 (CoreFoundation + 523100) [0x19782ab5c] 1-11
                            11  mshMIGPerform + 204 (HIServices + 46380) [0x19cfa652c] 1-11
                              11  _XCopyElementAtPosition + 356 (HIServices + 339400) [0x19cfeddc8] 1-11
                                11  _AXXMIGCopyElementAtPosition + 384 (HIServices + 186392) [0x19cfc8818] 1-11
                                  11  CopyElementAtPosition + 152 (AppKit + 3707004) [0x19ad9f07c] 1-11
                                    11  -[NSApplication(NSApplicationAccessibility) accessibilityHitTest:] + 264 (AppKit + 3934248) [0x19add6828] 1-11
                                      1   QAccessibleObject::object() const + 20 (libQt5Gui.5.15.8.dylib + 42668) [0x1057626ac] (running) 1
                                      1   ??? (libqcocoa.dylib + 280548) [0x10a4a47e4] (running) 2
                                      1   QAccessibleObject::object() const + 20 (libQt5Gui.5.15.8.dylib + 42668) [0x1057626ac] (running) 3
                                      1   ??? (libqcocoa.dylib + 280224) [0x10a4a46a0] (running) 4
                                      1   DYLD-STUB$$QAccessibleObject::isValid() const + 4 (libQt5Widgets.5.15.8.dylib + 2385652) [0x107db26f4] (running) 5
                                      1   ??? (libqcocoa.dylib + 280548) [0x10a4a47e4] (running) 6
                                      1   QAccessibleObject::object() const + 4 (libQt5Gui.5.15.8.dylib + 42652) [0x10576269c] (running) 7
                                      1   QAccessibleWidget::isValid() const + 20 (libQt5Widgets.5.15.8.dylib + 1524688) [0x107ce03d0] (running) 8
                                      1   QAccessibleObject::isValid() const + 16 (libQt5Gui.5.15.8.dylib + 42696) [0x1057626c8] (running) 9
                                      1   QAccessibleObject::object() const + 4 (libQt5Gui.5.15.8.dylib + 42652) [0x10576269c] (running) 10
                                      1   ??? (libqcocoa.dylib + 280224) [0x10a4a46a0] (running) 11
```

gregersn, 2023-11-25:
I am experience this issue, and I am also seeing that I can click around in the cells in row 1 through 9, but as I hit row 10, it freezes. Moving around with keyboard works. 

[code]
OS: macOS 12.6.7
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * fcgear 1.0.0
  * fasteners 0.4.1
[/code]

I also use the app "Rectangle", and quitting it avoids the hanging.

luzpaz, 2023-11-26:
@JohnOCFII can you retest on 0.22 build and report back ?

berkakinci, 2023-11-26:
> @JohnOCFII can you retest on 0.22 build and report back ?

Do try reproduce, but I caution this could be a misleading result.  Is there a specific fix that's in 0.22 (not in 0.21) you're trying to test?

Since my original issues, this has mysteriously stopped reproducing for me.  I can't get it to happen on the same version for FreeCad even though I try.  That's with Rectangle App running.
In the meantime, I've updated MacOS and Rectangle.  Another accessibility-enabled app according to my settings is VNC Viewer.  It also doesn't reproduce with that.

```
OS: macOS 13.6.1
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * kicadStepUpMod 10.19.4

```
```
Darwin **** 22.6.0 Darwin Kernel Version 22.6.0: Wed Oct  4 21:26:55 PDT 2023; root:xnu-8796.141.3.701.17~4/RELEASE_ARM64_T6020 arm64
```

```
Rectangle
Version 0.74 (80)
```

berkakinci, 2023-11-26:
> I can't get it to happen on the same version for FreeCad even though I try.

Well, literal seconds after my post, it hung again as I clicked back into FreeCad.  But, yet again, I can't make it reproduce.  BTW, I've been trying various orders of launching Rectangle, VNC and FreeCad.  I've used Rectangle features, right from within FreeCad etc...  So frustrating with intermittent issues.

JohnOCFII, 2023-11-27:
> @JohnOCFII can you retest on 0.22 build and report back ?

Resolved (testing with Magnet App).

```
OS: macOS 14.1.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35081 (Git)
Build type: Release
Branch: main
Hash: 32db044caa92a2adc448850179c524b793a03c3e
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
```

luzpaz, 2023-11-27:
@JohnOCFII What is Magnet app ?

JohnOCFII, 2023-11-27:
> @JohnOCFII What is Magnet app ?

https://magnet.crowdcafe.com

Earlier in this thread it became apparent this spreadsheet issue only appeared when users were using certain low-level apps on the Mac - seemingly apps that could intercept window management type activities.  Magnet is one of those types of apps.  It was the reason that I and others were not seeing this issue, since I don't often launch that app.  Since I owned the App, I launched it, and was able to experience the issue myself.  Now with the latest fixes, the issue is resolved, even when running the Magnet App (and, from what others have reported -- seemingly with the other similar apps as well).

closed 2023-11-27 by luzpaz

luzpaz, 2023-11-27:
Nice! @adrianinsaval are we going to bump stable on the website? (still shows 0.21.1 as latest stable)

adrianinsaval, 2023-11-27:
@chennes was having trouble getting the mac bundles signed so we haven't made an official announcement yet despite the version being tagged and assets being available: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/0.21.2

maybe we should just release anyways and update only the mac bundles when/if we get them properly signed

chennes, 2023-11-27:
If anyone is available to help me debug, here's the bundle with the signature that's not working right:
https://plsok-my.sharepoint.com/:u:/g/personal/chennes_pioneerlibrarysystem_org/Ea_hYMOkuMlHi0XjWngWZzIBdyAurmJeyOtBE7WTUN3Dzw?e=tqq31k

It should be signed, but the OS is reporting "unknown developer".

JohnOCFII, 2023-11-27:
> It should be signed, but the OS is reporting "unknown developer".

Hmm... `codesign` returned this reasonable reply:

```
(base)osh@OshMBPro Downloads % codesign -dvv FreeCAD.app 
Executable=/Users/osh/Downloads/FreeCAD.app/Contents/MacOS/FreeCAD
Identifier=org.freecadweb.FreeCAD
Format=app bundle with generic
CodeDirectory v=20200 size=278 flags=0x10000(runtime) hashes=1+5 location=embedded
Signature size=9093
Authority=Developer ID Application: The FreeCAD project association (289DJRF23X)
Authority=Developer ID Certification Authority
Authority=Apple Root CA
Timestamp=Nov 20, 2023 at 4:17:39 PM
Info.plist entries=18
TeamIdentifier=289DJRF23X
Sealed Resources version=2 rules=13 files=31878
Internal requirements count=2 size=236
(base)osh@OshMBPro Downloads % 
```

I had another Open Source vendor's app have the same issue, and got some help from a friend.  In that case, while it appeared to be a code signing issue, it was really a file attribute issue.

In that case, he suggested I run: `xattr -d com.apple.quarantine -r FreeCAD.app` 

I did that, and now that version of FreeCAD runs fine.  I _think_ it means that the signing signature is somehow "broken" or that the same signature isn't being applied correctly against both the app bundle AND the actual executable file.

You _might_ be able to run that `xattr` command before you place the file into the DMG.  Worth a shot, anyway.

```
OS: macOS 14.1.1
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
```

martinsp, 2023-12-01:
@chennes I was playing around `xcrun stapler` with validate & staple arguments and I noticed that v0.21.0 and v0.21.1 reports `teamId = 394HN54UJM;`, but for v0.21.2 it reports `teamId = 289DJRF23X;`

Was this change intentional?

martinsp, 2023-12-01:
I did a bit more looking at the logs in Console.app and noticed the following reported by `XprotectService`

> File /Applications/FreeCAD.app/Contents/Resources/sbin/../lib//libedit.0.dylib failed on rPathCmd /private/var/folders/76/zy5ktkns50v6gt5g8r0sf6sc0000gn/T/cirrus-ci-build/conda/osx/APP/FreeCAD.app/Contents/Resources/lib/libncurses.6.dylib (rpath resolved to: (path not found), bundleURL: /Applications/FreeCAD.app)

Here are some information https://apple.stackexchange.com/questions/339061/gatekeeper-refuses-validly-signed-application how the `rpath` related issue can be resolved

martinsp, 2023-12-01:
> is that the only wrong path? can it be changed to a relative path?

I'm not sure at the moment. Digging through the v0.21.1 version with `otool` there are paths that start with '/Users/adrianinsaval/repos/FreeCAD-Bundle/conda/...' and everything seems to be working fine.

One more thing I noticed, in the `FreeCAD.app/Contents/_CodeSignature` directory in v0.21.2 there is extra file `CodeEntitlements` which sets `com.apple.security.cs.disable-library-validation` to `true`

Apple documentation https://developer.apple.com/documentation/bundleresources/entitlements/com_apple_security_cs_disable-library-validation states that:

> Because library validation is such an important security-hardening feature, Gatekeeper runs extra security checks on programs that have it disabled. If your program is blocked by Gatekeeper, check whether you’ve unnecessarily disabled library validation.

Maybe the extra Gatekeeper checks now prohibits FreeCAD to run in v0.21.2? Is it possible to create FreeCAD build without the `com.apple.security.cs.disable-library-validation` set to `true` ?

martinsp, 2023-12-02:
@adrianinsaval I created a script that runs through all files in FreeCAD.app bundle (from https://github.com/FreeCAD/FreeCAD/issues/10144#issuecomment-1828345315), and found that there are 5 `LC_RPATH` paths in libraries which have an absolute path in them. I think it's safe to delete them

```bash
install_name_tool -delete_rpath "/private/var/folders/76/zy5ktkns50v6gt5g8r0sf6sc0000gn/T/cirrus-ci-build/conda/osx/APP/FreeCAD.app/Contents/Resources/lib" FreeCAD.app/Contents/Resources/lib/libffi.8.dylib
install_name_tool -delete_rpath "/opt/homebrew/opt/libomp/lib" FreeCAD.app/Contents/Resources/lib/python3.10/site-packages/opencamlib/ocl.so
install_name_tool -delete_rpath "/usr/local/opt/libomp/lib" FreeCAD.app/Contents/Resources/lib/python3.10/site-packages/opencamlib/ocl.so
install_name_tool -delete_rpath "/private/var/folders/76/zy5ktkns50v6gt5g8r0sf6sc0000gn/T/cirrus-ci-build/conda/osx/APP/FreeCAD.app/Contents/Resources/lib" FreeCAD.app/Contents/Resources/lib/libedit.0.dylib
install_name_tool -delete_rpath "/private/var/folders/76/zy5ktkns50v6gt5g8r0sf6sc0000gn/T/cirrus-ci-build/conda/osx/APP/FreeCAD.app/Contents/Resources/lib" FreeCAD.app/Contents/Resources/lib/libev.4.dylib

```

After those `rpath`s are removed you could check if re-signing the bundle works and if the app is allowed to run

adrianinsaval, 2023-12-02:
thanks a lot for the help martinsp! @chennes could you try that out?

this one looks potentially problematic:
```
install_name_tool -delete_rpath "/opt/homebrew/opt/libomp/lib" FreeCAD.app/Contents/Resources/lib/python3.10/site-packages/opencamlib/ocl.so
install_name_tool -delete_rpath "/usr/local/opt/libomp/lib" FreeCAD.app/Contents/Resources/lib/python3.10/site-packages/opencamlib/ocl.so
```

@looooo it seems the opencamlib package is pulling some dependencies from the system, what do we do here? should we attempt to add libomp to conda-forge?

adrianinsaval, 2023-12-02:
> @adrianinsaval I created a script that runs through all files in FreeCAD.app bundle (from [#10144 (comment)](https://github.com/FreeCAD/FreeCAD/issues/10144#issuecomment-1828345315)), and found that there are 5 `LC_RPATH` paths in libraries which have an absolute path in them. I think it's safe to delete them

could you share that script?

martinsp, 2023-12-02:
Here is a gist of a ruby script I used: https://gist.github.com/martinsp/a4e9f6d4ea2b9924d098a215c97fa5d3

The logic is quite simple (and inefficient), but it seems to work:
1. run `otool` for every file in the directory
2. in the output of otool look for "cmd LC_RPATH"
3. search for "path .... (offset ..)" after each "cmd LC_RPATH" found
4. write the results in result.json with pathname and all rpaths's found for each file

looooo, 2023-12-03:
> @looooo it seems the opencamlib package is pulling some dependencies from the system, what do we do here? ~should we attempt to add libomp to conda-forge?~ ->it's available as [llvm-openmp](https://anaconda.org/conda-forge/llvm-openmp) and included in the bundle

Running stool -l ocl.so in the Conda-env also shows these paths:
```
Load command 17
          cmd LC_RPATH
      cmdsize 48
         path /opt/homebrew/opt/libomp/lib (offset 12)
Load command 18
          cmd LC_RPATH
      cmdsize 40
         path /usr/local/opt/libomp/lib (offset 12)
Load command 19
          cmd LC_RPATH
      cmdsize 32
         path @loader_path (offset 12)
Load command 20
          cmd LC_RPATH
      cmdsize 40
         path @loader_path/../../../ (offset 12)
```

Maybe we can try to delete these libraries before creating the opencamlib package.

martinsp, 2023-12-04:
I checked the last weekly build from https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds and it looks like there are no more rpaths with absolute paths

[5 more replies]