FreeCAD/FreeCAD #10165: [Feature] Start page new file templates

qewer33, 2023-08-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
[code]
OS: Fedora Linux 38 (KDE Plasma) (KDE/plasmax11)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33675 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.0)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.50.2
  * Curves 0.6.12
  * DynamicData 2.46.0
  * fasteners 0.4.58
  * freecad.gears 1.0.0
  * Glass
  * lattice2 1.0.0
  * kicadStepUpMod 10.19.4
  * Manipulator 1.5.0
  * Behave-Dark-Colors 0.1.1
[/code]
```

### Subproject(s) affected?

Other (specify in description)

### Problem description

One of the problems of FreeCAD for newcomers is not knowing what to do when they create a new file for the first time, staring into the endless 3D void inside the Start workbench with icons that make no sense for any kind of 3D work. Of course this can be discarded with comments like "New users should just watch tutorials, they should look at the wiki before creating a new file" etc. But many newcomers (especially if they're from another CAD software) have the confidence to just dive in and start using the program. As an example, I myself came to FreeCAD from Solidworks (thus I knew how the feature and sketch based CAD workflow worked) and I remember having to google "FreeCAD Part Design vs Part" to actually determine which workbench I should use to create a simple part.

This idea proposes a simple "fix" for that problem. Adding new file templates to the start workbench will help these new users see what workbench they should use for which kind of project. It will also reduce clicks for advanced users (for example instead of creating a new file, switching to the PD workbench and pressing "Create Body" you can just click the Part Design file template).

Current template ideas:
- Parametric Part: Spawns the user inside the Part Design workbench with a pre created Part Design Body
- CSG Part: Spawns the user inside the Part workbench (potentially with a pre-created geometry such as a Cube or a Sketch)
- Assembly: This is for the future when FreeCAD gets a default assembly workbench
- 2D Drawing: Spawns the user inside the Draft workbench
- Architecture: Spawns the user inside the Arch workbench

If this idea is agreed upon. I am willing to work on this myself and create a PR that implements it.

### Anything else?

Here is a mockup:

[image]

Forum discussion (some discussion about this also happened in the Discord server on the hackathon channel): https://forum.freecad.org/viewtopic.php?p=698862

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2023-08-10:
IMO great idea, that’s what is really missing from FreeCAD in terms of making it clear for new users what to do. When I started FreeCAD for the first time (having experience with other CAD software), I was confused if I should use Part or Part Design workbench and which of the several workbenches are actually important for me. Of course, one can just read about this in the wiki documentation but it’s not what user-friendliness is about. It’s much better if the software can help users get started. And the start page is a perfect place for that. No need for additional pop-up windows and so on. We just add a few not so big buttons and the software suddenly becomes not only much more intuitive for new users but also more professional.

closed 2023-08-16 by chennes

kaktusus, 2023-08-31:
@qewer33 
The issue of translating the new GUI element labels remains unresolved.
:crying_cat_face:

qewer33, 2023-08-31:
> @qewer33 The issue of translating the new GUI element labels remains unresolved. 😿

I wasn't aware of this (but I was worried it could have been a problem). Can you provide some details on why they are untranslatable, I'll try to fix it