FreeCAD/FreeCAD #10173: [Problem] switching attachment support on FreeCAD for M1 ARM chip not work

psicman, 2023-08-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
[code]
OS: macOS 13.5
Word size of FreeCAD: 64-bit
Version: 0.21.0.33675 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.0)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * freecad.gears 1.0.0
[/code]
```

### Subproject(s) affected?

Sketcher

### Problem description

i'm a complete newbie learning FreeCAD by following a youtube tutorial, (using M1 macbook air with FreeCAD for ARM version)
but when i try to recreate the exact action that was conducted in this following tutorial link and time
(https://youtu.be/WZZK4vmXK9Q?list=PLWuyJLVUNtc0UszswD0oD5q4VeWTrK7JC&t=962) it is not working at all. 

and when i try to uninstall the ARM version and install the X86 one to run it with Rosetta 2, it won't even open at all.

i posted this on the freeCAD forum but it was not helpful.

not working video: [https://www.dropbox.com/scl/fi/byrmjo8m ... ljaft&dl=0](https://www.dropbox.com/scl/fi/byrmjo8mgxh8d331zjh8f/not-work-unlike-tutorial.mov?rlkey=tbk8uoufp1pebmqc01c9ljaft&dl=0)

3D file for trial: [https://www.dropbox.com/scl/fi/m428mvy2 ... 577l9&dl=0](https://www.dropbox.com/scl/fi/m428mvy20njjjfaxkr0qh/Tutorial-5.2.FCStd?rlkey=0sjz97kzyu5hcsmi8bgo577l9&dl=0)

i'm attaching a dropbox link because the file is too big.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2023-08-12:
Forum topic:
https://forum.freecad.org/viewtopic.php?t=80353

psicman, 2023-08-12:
> Forum topic:
> https://forum.freecad.org/viewtopic.php?t=80353

Yes i am the one who ask this on the forum, although it’s still not work despite i tried what people suggested on the list so far.

psicman, 2023-08-13:
it works with the "map sketch to face" button.

adrianinsaval, 2023-10-05:
I can only assume that you where working with a non attached sketch. "not working" video no longer available. Issue is not actionable -> closing.

closed 2023-10-05 by adrianinsaval