FreeCAD/FreeCAD #10174: [Problem] Materials Functionality is inconsistent and inflexible

sliptonic, 2023-08-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
any
```

### Subproject(s) affected?

None

### Problem description

this is a placeholder issue.    

Materials is a broad requirement in many different workbenches.   A more robust and integrated capability is needed that supports existing tools and allows us to develop more advanced features in the future.

Context: 

https://forum.freecad.org/viewtopic.php?t=16714&sid=3ee17d7b50064496fa66255f0a46672c

https://ondsel.com/blog/freecad-needs-a-better-materials-system

https://forum.freecad.org/viewtopic.php?style=4&p=685834#p685834

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-08-12:
CC @davesrocketshop (just linking you to ticket :wink: )

davesrocketshop, 2023-08-26:
1st PR submitted. From the pull request:

NOTE: This PR introduces a new dependency on yaml-cpp (https://github.com/jbeder/yaml-cpp). This is available through
apt on Debian based distributions, but the LibPack image will need to be updated for Windows builds

First PR for Issue https://github.com/FreeCAD/FreeCAD/issues/10174 - the Materials system update.

    The editor is minimally functional. You can save changes but may need to restart for them to take effect
    The API supports new and old methods of access. See the unit tests in src/Mod/Material/TestMaterialApp.py for examples
    Uses an updated .FCMat format. The old editor won't read the new files, but the new editor will still read the old files.
    Defaults to the old editor. This can be changed in the Material Preferences pane
    System materials are in the new format. You can still use the old format files by copying them to a separate folder and adding that as a custom folder in the Material Preferences pan
    Moves the preferences from FEM to a separate Materials section. This was submitted as a PR prior to 0.21 but that PR has been closed to avoid conflicts

sliptonic, 2023-10-03:
Dave did a good breakdown of the work that needs to be done.  I'm reproducing it here.  A lot of these topics are already individual issues or should become issues at the right time but let's not swamp the tracker just yet.

Materials Core
     - diff feature
     - export inherited models
     - internal APIs
     - unit tests

Material libraries API
     - support external libraries
     - e.g. Web Services/DB/git
     - API for add on workbenches

Material arrays
     - Validation (no duplicate entries, for eg)
     - Interpolation: in-between values

Materials Editor
     - edit/save/load material cards

Materials diff viewer
     - visualization of core diff function
     - Possibly as part of editor and/or standalone

Embedded Materials
     - Save materials with documents (optional)
     - Create external material card from embedded material
     - In-Document material card with no external file

Physical Materials
     - add new property for physicl materials
     - every Shape object would have an associated materials object
     - embed in FC document where desired
     - multi-materials

Appearance Materials
     - Associate Material property with cards
     - remove hard coded materials in favour of cards
     - add texture(image) support

Apperance Preview
     - Create more realistic preview for Materials UI
     - related to Appearance/UI tickets

Arch migration
     - Migrate Arch workbench to use new Materials
     - thread on forum
FEM migration
     - Migrate Arch workbench to use new Materials
     - thread on forum

davesrocketshop, 2023-10-22:
New PR #11138. Primarily editor enhancements to support arrays, but also improvements to the material save workflow. Numerous minor improvements

closed 2024-07-01 by maxwxyz