FreeCAD/FreeCAD #10176: [Problem] HiDPI problem after changing any settings in preferences

Serince, 2023-08-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
[code]
OS: Linux Mint 21.2 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33682 (Git)
Build type: Release
Branch: master
Hash: 9dce2525f8efe0873787f4388f10e11e168fe16b
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: Turkish/Turkey (tr_TR)
Installed mods: 
  * CfdOF 1.17.6
  * FEMbyGEN 1.2.0
[/code]
```

### Subproject(s) affected?

None

### Problem description

When I open freecad 0.21 on my hidpi screen it works fine, but when I open preferences and just press apply all the texts in report view, python console and macro editor shrinks. To fix it, I delete user.cfg in config folder and the problem is fixed.

### Anything else?

no user.cfg file
[image]

After pressing apply button in preferences
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2023-08-12:
I can confirm this in a more general sense. The preferences (*.ui files) are also a repository for default values. These values sometimes do not match the defaults used in other parts of the code. Pressing "Apply" tells the program to use the values from the preferences (I think only from the current tab) even if the user has not made any changes.

Roy-043, 2023-08-16:
I have added all workbench labels since this may effect them all.

There are two potential solutions:
1. Make sure defaults in the code **always** match the defaults stored in the ui files of the preferences.
2. Read the ui files at runtime and extract default values from them. Forum topic with an attempt to do this for the Draft WB: https://forum.freecad.org/viewtopic.php?p=607927.

chennes, 2023-08-18:
Oops, sorry @Roy-043, I see now you added those tags on purpose... I thought it must have been an accident.

benj5378, 2023-08-21:
@luzpaz This looks like an undefined variable path. Seems like a new issue, so we could check if topology.py in FEM workbench have been changed lately.

benj5378, 2023-08-22:
@luzpaz Some preferences rely on global code for saving a value. For instance, for saving the value of a checkbox, we have a global function (across all checkboxes and workbenches). If there is an error in this global function, it would break across all workbenches.

Though even it would break across all workbenches, we might only see an error of the first checkbox trying to be saved, as execution might then be halted. Let's say the first Sketcher workbench checkbox is the first to have the save function called, then an error would be thrown related to the Sketcher workbench and execution would halt - we wouldn't see any errors from the other workbenches because the code here is never run, even though the error affects and would throw an error across all workbenches if run. Only might; this is dependent on implementation: code might have been implemented such a way that it keep going (no halt) even though an error is thrown.

But not all preference settings use global save functions. Maybe we want to save a preference in a specific data format or use some UI rarely used. Then a specific save function is implemented, for the single preference setting in the specific workbench for instance. If that save function throws an error, we would only see the error when saving the preference setting for the specific workbench, because it's the only location it's been implemented/used.

luzpaz, 2023-08-23:
Thanks for the thorough explanation. Any ideas on what the way forward looks like? 

I wonder if this issue might be behind how config files mysteriously become corrupted? The solution given when this happens is to just nuke said config files and restart FreeCAD.

Serince, 2023-09-05:
I don't know how, but I can't reproduce the bug. It works fine.

luzpaz, 2023-09-07:
How is it a font error?! \<*scratches head*\>

benj5378, 2023-09-07:
> How is it a font error?! \<*scratches head*\>

I don't get this either. Why would a specific font get it's size set every time applying preferences

bgbsww, 2023-12-05:
Are we sure the font issue occured on Ubuntu, and not Mint as shown in the version info above?  I can't directly replicate on my Ubuntu machine, which seems to scale.  But it also does a good font substitution:

```
C940:~/tests$ xdpyinfo | grep dimensions
  dimensions:    3840x4560 pixels (1016x1206 millimeters)
C940:~/tests$ fc-match Arial
LiberationSans-Regular.ttf: "Liberation Sans" "Regular"
C940:~/tests$ cat /etc/issue
Ubuntu 22.04.3 LTS \n \l
```

That's a bit of a side point, looking into the general preferences defaults is clearly the core issue here, I was just looking for an easy example to test with.

bgbsww, 2023-12-06:
> 2.  Read the ui files at runtime and extract default values from them. Forum topic with an attempt to do this for the Draft WB: https://forum.freecad.org/viewtopic.php?p=607927.

After a little poking, I don't think this path is viable.  The core ui files are gone at runtime, as they are used to generate c++ code that is compiled into the application.  There are a few module based ui files, but for example, the preferences dialog is c++.

It is conceivably possible to generate something at compile time that collects all the default values declared in ui files.  So maybe this is better viewed as a build time test that flags potential places where there may be a default in two places.

bgbsww, 2023-12-06:
Here's a python program that tries to find every default set in a .ui file, then match then to .cpp files that reference the resulting ui class, locate every time a property value is set, and compare it to a default.

 [propertyDefaults.zip](https://github.com/FreeCAD/FreeCAD/files/13589291/propertyDefaults.zip)

Also attached is a result file that shows the ~373 places where defaults are set that don't match the code.  Most of these are false positives, and there are likely still bugs in the python script.

bgbsww, 2023-12-06:
For the specific font size bug, it might be https://github.com/FreeCAD/FreeCAD/blob/09f67f2f6bff1c57ac8019a5ba2f4a2c41025ee6/src/Gui/PreferencePages/DlgSettingsEditor.cpp#L294C1-L295 which is going to force a size of 10 if there isn't a default.

Doesn't solve the general case, and I don't have definite proof since I can't replicate.

maxwxyz, 2024-07-01:
@Serince @Roy-043 is this still an issue?

Roy-043, 2024-07-02:
Yes this is still an issue.
But let's close this one in favor of #14919.

closed 2024-07-02 by Roy-043