FreeCAD/FreeCAD #10177: [Problem] FreeCAD may move toolbars from user-defined positions

AjinkyaDahale, 2023-08-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
[code]
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.33789 (Git) Snap 734
Build type: Release
Branch: master
Hash: 22b173d0e27cbb641661d6844512e02020a9890a
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: English/India (en_IN)
[/code]
```

### Subproject(s) affected?

None

### Problem description

First came up in the FreeCAD Day "complaint" session in Feb 2023: https://forum.freecad.org/viewtopic.php?p=663445#p663445.

When a user moves toolbars of a workbench to custom positions, these changes are sometimes not respected when changing workbenches or on restarting FreeCAD.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2023-08-12:
do we step by step instructions to reproduce?

AjinkyaDahale, 2023-08-12:
> do we have step by step instructions to reproduce?

We are trying to find a reliable way to do so. This is why I added "may" and "sometimes". See, for example, https://discord.com/channels/870877411049357352/1132355125835485204/1139694497165484082.

That said it definitely happens erratically.

closed 2025-02-06 by github-actions[bot]