FreeCAD/FreeCAD #10180: [Problem] Sketcher colors are hardcoded in the Freecad

MisterMakerNL, 2023-08-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.0.33656 +4 (Git)
Build type: Release
Branch: pr/10015
Hash: 5368b0f8a290b683ef4816664d7a0c1330dcdba5
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Dutch/Netherlands (nl_NL)
Installed mods: 
  * Assembly3 0.12.0
  * Behave-Dark-Colors 0.1.1
  * BIM 2021.12.0
  * Dracula 0.0.4
  * ExtremeProLight 1.6.0
  * FeedsAndSpeeds 0.5.0
  * FEM_FrontISTR 0.1.0
  * Freecad-Built-in-themes-beta 1.2.2
  * IconThemes
  * lattice2 1.0.0
  * Nodes 0.1.36
  * PieMenu 1.2.7
  * Render 2023.2.6
  * scanObjects
  * SelectorToolbar
  * sheetmetal 0.2.59
  * SP_Startup
  * ToolbarStyle
[/code]
```

### Subproject(s) affected?

None

### Problem description

So the issue is that a new sketch is created by default with white lines.
The white color is assigned to a sketch.
If another user opens the file the colors stay white.
If the other user uses a white background.
He cannot see the sketch lines.
There is currently no way to change the sketch colors other than manually change each sketch.
@PaddleStroke proposed a good solution to this in the forum thread:
https://forum.freecad.org/viewtopic.php?t=79486&e=1&view=unread#unread

[image]
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2023-08-12:
As already explained in the forum topic: sketch colors are NOT hardcoded. There are preferences for these colors that set the defaults for new sketches.

MisterMakerNL, 2023-08-12:
> As already explained in the forum topic: sketch colors are not hardcoded. There are preferences for these colors that set the defaults for new sketches.

You are right! Title is better like this, thanks for adjusting!

MisterMakerNL, 2023-08-13:
topic suggest changing the color of all sketches although I would suggest having a default color that is used by default and comes from the settings instead of the sketch properties.
Urer could overwrite this setting per sketch by choosing a custom color.

luzpaz, 2024-10-18:
Related:  
* https://github.com/FreeCAD/FreeCAD/issues/10100
* https://github.com/FreeCAD/FreeCAD/issues/13318

maxwxyz, 2025-06-16:
I think this is resolved with the auto sketch color https://github.com/FreeCAD/FreeCAD/pull/14322

closed 2025-06-16 by maxwxyz