FreeCAD/FreeCAD #10183: 0.21 version - preferences menu missed in macOS

riparz, 2023-08-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
[code]
OS: macOS 13.5
Word size of FreeCAD: 64-bit
Version: 0.21.0.33675 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.0)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * A2plus.backup1665072169.758465
  * fcgear 1.0.0
  * A2plus 0.4.60h
  * A2plus.backup1665072549.143478
  * Assembly4 0.12.6
[/code]
```

### Subproject(s) affected?

Other (specify in description)

### Problem description

With the new version 0.21 the Preferences menu is no longer selectable. Instead, there are 4 sub-menus (Default, Transform, Crop, Color).
I can therefore no longer change the preferences as in version 0.20
Am I doing something wrong or is it a bug.
Thanks

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

chennes, 2023-08-13:
I can confirm - in Italian this menu is getting taken over by the Edit Mode menu.

luzpaz, 2023-08-16:
@chennes Is this a translation string that is causing the bug in Italian locale ?

luzpaz, 2023-08-16:
BTW, is the workaround ⌘+`,` (apple button + comma) to invoke the Preferences menu ?

chennes, 2023-08-16:
I cannot figure this one out -- I can't see any logical reason for the User Edit Menu to have anything to do with anything that's referred to by a name, and certainly not only on Mac OS. This needs some study.

riparz, 2023-08-17:
> BTW, is the workaround ⌘+`,` (apple button + comma) to invoke the Preferences menu ? (cc @riparz)

No, it's the first workaround wich I tried, but it doesn't work.

riparz, 2023-08-31:
Hi guys, do you have time on issue resolution? Maybe ... is there some Python commands to invoke preferences form from console command line? Thanks!

chennes, 2023-08-31:
No timeline on resolution (because I simply have no idea what the problem is!) -- but the Python command you need is `Gui.runCommand('Std_DlgPreferences',0)`.

riparz, 2023-09-30:
Hi guys, any news on the issue? Thanks