FreeCAD/FreeCAD #10190: [Problem] Changing a sketch constraint can cause sketch elements to flip

srwalter, 2023-08-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
[code]
OS: Ubuntu Core 20 (GNOME/gnome-xorg)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33668 +7 (Git) Snap 727
Build type: Release
Branch: tag: 0.21.0
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.8.10, Qt 5.15.7, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

None

### Problem description

On the sketch "Screw Mounts," if I change the 30mm constraint to 10mm, it flips/inverts the rectangle on the right.  Instead of both constraints being against the farther side of the rectangle, both constraints are now against the nearer side of the rectangle.  This is not equivalent to the constraints I originally entered.

[bug.zip](https://github.com/FreeCAD/FreeCAD/files/12328213/bug.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-08-12:
Did you check for duplicates ? 
#5763 
#5555 
#9683  

@AjinkyaDahale do you mind triaging this ticket ? :pray:

srwalter, 2023-08-13:
I did look around before posting, but apparently not well enough.  Sorry about that!  I agree that #5763 seems to be the same issue, and the work-around mentioned there (using the LevenbergMarquardt solver) seems to help

closed 2023-08-13 by luzpaz