FreeCAD/FreeCAD #10192: I don't find how to do a fluid simulation using the FEM workbench

Alberto4343, 2023-08-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.0.33668 +7 (Git)
Build type: Release
Branch: (HEAD detached at 0.21)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Spanish/Peru (es_PE)
Installed mods: 
  * Help 1.0.3
[/code]
```

### Subproject(s) affected?

None

### Problem description

I can't find in the wiki how to archieve this. In addition i can´t find it in the web.

There is a heavy lack of official guides and manuals. So much of the documentation doesn't have enough deep.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-08-13:
@Alberto4343 we're currently revamping our documentation. Thanks for the feedback. Can you mention something to the https://github.com/jaheyns/CfdOF folks ? They can also get you to where you want to go.

closed 2023-08-13 by luzpaz

Alberto4343, 2023-08-13:
I don't want to use CfdOF, i want to use the FEM workbench.

luzpaz, 2023-08-13:
@Alberto4343 please talk to the FEM folks in https://forum.freecad.org/viewforum.php?f=18

Alberto4343, 2023-08-13:
Can't subscribe to the forum. It loads the previous page when i select my language.