FreeCAD/FreeCAD #10207: [Feature request] Better line separation in depency graphs

Geobot19, 2023-08-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
[code]
OS: Fedora Linux 38 (Workstation Edition) (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.20.2.Unknown
Build type: Unknown
Python 3.11.1, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Alternate_OpenSCAD 1.0.0
  * fasteners 0.4.56
  * Pyramids-and-Polyhedrons
[/code]
```

### Subproject(s) affected?

Other (specify in description)

### Problem description

Subproject: dependency graph generator
when a dependency graph gets messy, the lines on the graph can get way to close together, making it difficult to follow any one line. Here is a close up of an example:
[image]

It would be nice for the sake of readability if these lines were spaced further apart.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2023-08-14:
The line(s) generation is handled by graphviz so I'm not sure how much the FreeCAD code can be tweaked to improve this. Firstly can you confirm which version of graphviz you are using. On Linux the command `/usr/bin/dot -V` gives the full version, on Windows I'm guessing the user has to right-click the graphviz.exe and select Properties but I'm guessing at that. At present I've just upgraded to version 8.1 and I've probably only got one model complicated enough to produce this scenario.

Syres916, 2023-08-14:
I used the same model and saved the dependency graph using version 2.38 and 8.1, see attached. I would suggest 8.1 does produces (slightly) clearer graphs. If this is still not clear enough then all I can suggest is using Inkscape to move the nodes in order to give more clarity.

Version 2.38

[image]

Version 8.1

[image]

luzpaz, 2023-08-15:
This is the current snapshot of graphviz in the package ecosystem. At the time of this post there are still many repos that aren't using an up to date stable of graphviz.

[image]

luzpaz, 2023-08-16:
@Geobot19 can you install v0.21 (latest stable) and see if it improves ?

Geobot19, 2023-08-16:
the original example was generated using 0.20.2 on fedora 38, I will try with 0.21.0 on arch soon

Geobot19, 2023-08-16:
[image]
this screenshot is from arch today, problem persists

Geobot19, 2023-08-16:
graphviz was installed fresh (it wasn't installed on this pc)

Geobot19, 2023-08-16:
Arch installation details
[code]
OS: Arch Linux (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33675 (Git)
Build type: Release
Branch: makepkg
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.11.3, Qt 5.15.10, Coin 4.0.1, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * fasteners 0.4.56
  * Pyramids-and-Polyhedrons
  * Alternate_OpenSCAD 1.0.0
[/code]

luzpaz, 2023-08-16:
Looks like https://archlinux.org/packages/extra/x86_64/freecad/ is running graphviz 8.10 (latest stable). I'm assuming this is the package you have installed @Geobot19  ? If so, that's weird. Can you share a test file ?

Geobot19, 2023-08-16:
[96 percent simple rev 2.zip](https://github.com/FreeCAD/FreeCAD/files/12364240/96.percent.simple.rev.2.zip)
this is the file I've been using, this file. sorry that it's a complex file, the are in the screenshots is around the spreadsheet

Geobot19, 2023-08-16:
> I'm assuming this is the package you have installed
yes

hasecilu, 2023-10-12:
Hi, I have seen his problem from a while ago, I tested this [Digital caliper](https://github.com/FreeCAD/FreeCAD-library/tree/master/Industrial%20Design/Hand%20tools/Measuring%20tools) file on FC 0.20.2, 0.21.1 and 0.22.0dev (34591) with same result: edge overlapping.

I read some docs and it seems that there is an algorithm to avoid overlapping of nodes (boxes) but not for edges (lines) because those are made individually.
The used engine is [dot](https://www.graphviz.org/docs/layouts/dot/)

In the `0.22.0dev` version (need to compile it) is posible to save a `.gv` file, a text file that contains info of the graph, using the `dot` package you can generate a new image (`.ps` file)

This is the generated in FreeCAD
[image]

And this the generated with: `dot -Gordering=out -Granksep=2 -Gnodesep=1 -Grankdir=TB -Gsplines=ortho -Nshape=box -Tps Mesa_de_centro.gv -o Mesa_de_centro_opti.ps`
Note that the image is bigger and the space is not really optimized, for instance the `Origins` take a lot of space.
But the main thing is that is possible to change some parameters to enhance the graph, tha main change here is to change the style of the splines.
[image]

I still need to figure it out what changes are needed in the source code, in: 
https://github.com/FreeCAD/FreeCAD/blame/aa5bea4e2e8089bf00f61075a0860dacda42dc8b/src/App/Graphviz.cpp#L41

the method seems unused but looks like it have a way to change the options, also `void Document::exportGraphviz(std::ostream& out) const` is using defaults?

AustinPackard, 2024-09-09:
Having a similar issue with the lines between nodes being impossible to trace. Are there any alternative methods for listing dependents of a named constraint?

luzpaz, 2024-09-09:
@hasecilu do you want to submit a PR that could get a fix rolling for this ?

luzpaz, 2024-09-10:
> Hi, do you think the solution above is good enough?, I think is not optimal but somewhat better because edges don't overlap.

Yes, if it's better solution then we have... sure make a PR. We'll be trending in the right direction. TIA!

commit 16897e2f (2024-11-11): DependencyGraph: Tweak `dot` flags to prevent overlaped edges

- `-Granksep=2` lets increase the separation between groups (nodes),
ideally should depend on graph size
- `-Goutputorder=edgesfirst` lets place the edges first to allocate more
space for them
- `-Gsplines=ortho` changes the edges from curves to straight,
orthogonal lines

Fix #10207
closed 2024-11-11 by yorikvanhavre