FreeCAD/FreeCAD #10210: [Problem]  Gradient Backgrounds Are Not Dithered

obelisk79, 2023-08-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
[code]
OS: EndeavourOS (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33675 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.0)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * persistenttoolbars
  * Defeaturing 1.2.0
  * Manipulator 1.5.0
  * A2plus 0.4.60n
  * TabBar (Disabled)
  * IconThemes
  * CubeMenu
  * Dracula 0.0.4
  * Render 2023.2.6
  * SelectorToolbar (Disabled)
  * 3DfindIT 1.2.0
  * lattice2 1.0.0
  * Curves 0.6.13
  * Glass (Disabled)
  * freecad.gears 1.0.0
  * fasteners 0.4.59
[/code]
```

### Subproject(s) affected?

None

### Problem description

When applying gradient backgrounds with colors close to each other on the color spectrum, visible banding is clearly seen. This give a poor appearance to the background in contrast to the phong shaders used to render objects in the main view area.
 Image provided as an example:
 
[image]

Simple gradient calculations are known to have this problem due to limited color depth on computers. Solutions to correct this generally involve introducing a dithered effect through the use of noise.

The current implementation is located here: https://github.com/xtemp09/FreeCAD/blob/81cfe340854152fe79b81c3a3fafc29f8c3521a1/src/Gui/SoFCBackgroundGradient.cpp

### Anything else?

Technical explanation of the phenomena and possible solutions (assuming opengl) are outlined here: 
https://shader-tutorial.dev/advanced/color-banding-dithering/
and here:
https://computergraphics.stackexchange.com/questions/3964/opengl-specular-shading-gradient-banding-issues

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2023-08-13:
I ought to ask ― if one adds dithering, will another user complain, demanding shaders? Is dithering worth adding at all?

You set two _slightly_ different colors and observe the lines invisible if you set just _different_ colors. I just don't think people do this.

How much of dithering are you thinking of?

obelisk79, 2023-08-13:
I am not sure if you read the links regarding the banding and dithering of gradients to create natural realistic color fade, but if you have not, it discusses how much dithering/noise  in order to smooth the bands.

Regarding the similarity of colors, I have been conducting extensive color research regarding accessibility for the color-impaired, and using colors that too greatly differ from each other is generally bad, particularly while sketching. https://forum.freecad.org/viewtopic.php?t=80355

Additionally, this is not just about accessibility and color contrast, but also about aesthetic. A well coordinated color palette can heavily impact if a user interface presents a cohesive or disjointed feel in user experience.

chennes, 2023-08-13:
GL_DITHER is enabled by default in OpenGL. I see two places in the codebase where it gets turned off, but they are pretty isolated. Is there another setting we need besides that one?

obelisk79, 2023-08-13:
Some further reading seems to indicate that a gradient generated in 16bit is less prone to banding. Not sure if that is easily done in this case.

xtemp09, 2025-09-15:
Occasionally, when I have spare time, I return to this issue. Through many trials, it seems like I have found an only solution: to prepare the picture with Qt.

The thing here is that since the picture is prepared with Qt, it could be anything (an SVG image or a PNG file of a picture of a kitty). Creation of a gradient, dithering or blurring, takes a couple of seconds for a FullHD resolution. For a HiDPI screen it requires more.

@obelisk79, does this solution satisfy your needs?

closed 2026-09-16 by maxwxyz