FreeCAD/FreeCAD #10212: [Problem] TechDraw complex section based on another complex section is incorrect

FEA-eng, 2023-08-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.0.33668 +7 (Git)
Build type: Release
Branch: (HEAD detached at 0.21)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

Techdraw

### Problem description

TechDraw Complex section based on another complex section is generated incorrectly because of the profile misalignment:

[image]

Forum discussion: https://forum.freecad.org/viewtopic.php?t=80362

File:
[Complex section on complex section.zip](https://github.com/FreeCAD/FreeCAD/files/12329894/Complex.section.on.complex.section.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

WandererFan, 2023-08-20:
Partially fixed by commit 69523b5ea4.

Needs more mods to have second section use original shape instead of result of first section.

WandererFan, 2023-09-13:
Commit 9c1bc2e0e9 allows original shape to be used.

closed 2023-09-13 by WandererFan