FreeCAD/FreeCAD #10213: [Feature Request] Cross Product Vector Function in Expressions

gbroques, 2023-08-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
OS: Ubuntu 20.04.6 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.20.29177 (Git)
Build type: Release
Branch: (HEAD detached at 0.20)
Hash: 68e337670e227889217652ddac593c93b5e8dc94
Python 3.10.5, Qt 5.12.9, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.3
Locale: English/United States (en_US)
Installed mods: 
  * fasteners 0.4.25
  * freecad-to-obj
```
```

### Subproject(s) affected?

Expressions

### Problem description

Can we add a function to take the cross product of two vectors in [Expressions](https://wiki.freecad.org/Expressions)?

Cross products are really useful for:

1. finding the normal vector to a plane by taking the cross product of two vectors on the plane.
2. finding the direction vector for the line of intersection between two planes by taking the cross product of the two plane's normal vectors.

Currently I calculate cross products manually myself.

For example, the cross product between two vectors [mono]v1[/mono] and [mono]v2[/mono] is:

```
=vector(.v1.y * .v2.z - .v1.z * .v2.y; .v1.z * .v2.x - .v1.x * .v2.z; .v1.x * .v2.y - ```

The following would be much more simple:
```
=cross(.v1; .v2)
```

Since the avaiable matrix functions are prefaced with "m", then we could preface this function with a "v" (i.e. `vcross`).

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-08-13:
Want to take a stab at it? (here are [some previous expression related issues](https://github.com/FreeCAD/FreeCAD/issues?q=label%3AExpressions+is%3Aclosed) you can reference

gbroques, 2023-08-13:
> Want to take a stab at it? (here are [some previous expression related issues](https://github.com/FreeCAD/FreeCAD/issues?q=label%3AExpressions+is%3Aclosed) you can reference)

I appreciate the encouragement.

To be honest, I don't have too much time to try this and it would be my first code contribution to FreeCAD which is intimidating.

If anyone else wants to try, then this seems like a good reference from the list luzpaz provided:
https://github.com/FreeCAD/FreeCAD/pull/8603

luzpaz, 2023-08-13:
CC @Daniel-Khodabakhsh interested ?

Daniel-Khodabakhsh, 2023-08-14:
Sure thing @luzpaz ! Let me try to get something working by this week.

gbroques, 2023-08-15:
If @Daniel-Khodabakhsh and the community are up for it, then we may want to expand this to add other vector functions like one to get the angle between two vectors, one to normalize a vector, distance to line, distance to line segment, etc.

Up to you all though! :)

Daniel-Khodabakhsh, 2023-08-15:
Thanks for the suggestion @gbroques, I went through the API and added all the functions that didn't already have an expression function. Not sure about the names though, please feel free to review.

luzpaz, 2023-08-16:
@Daniel-Khodabakhsh thanks for being so 'Johnny of the spot' :heart: