FreeCAD/FreeCAD #10220: [Problem] Using "master" as branch name

pieterhijma, 2023-08-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
OS: Arch Linux
Word size of FreeCAD: 64-bit
Version: 0.21.0.33675 (Git)
Build type: Release
Branch: makepkg
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.11.3, Qt 5.15.10, Coin 4.0.1, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * freecad.gears 1.0.0
  * Ondsel-Lens
  * BIM 2021.12.0
  * lattice2 1.0.0
  * LCInterlocking
  * parts_library
  * fasteners 0.4.56
  * Assembly4 0.50.2
```

### Subproject(s) affected?

None

### Problem description

Many communities are moving away from terms such as "master". This is also true in Git, for example discussed by the [Software Freedom Conservancy](https://sfconservancy.org/news/2020/jun/23/gitbranchname/) and [GitHub](https://github.com/github/renaming). I think it would be good if the FreeCAD project makes the same change by renaming `master` to `main`, which appears to be the most popular replacement.

It's probably best to set a date, announce it widely and provide some instructions for those who have clones on their local machines and in their GitHub account.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Pesc0, 2023-08-14:
#6807

luzpaz, 2023-08-14:
This will be a huge headache and will break a lot of links. This was discussed in the past and the enormity of it was daunting along with the repercussions (downstream packagers, old bugs, documentation, online links, etc..etc..). I understand the sentiment but yikes...