FreeCAD/FreeCAD #10221: /tmp/FreeCADStartThumbnails⍰⍰⍰⍰⍰⍰⍰  directories are not removed

berberic2, 2023-08-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
[code]
OS: Debian GNU/Linux 12 (bookworm) (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.33802 +2 (Git)
Build type: Release
Branch: Meins
Hash: c70ded0ad5c6dadc86d3731aa0233f291b2839fa
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * Curves 0.6.13
  * Assembly4 0.50.2
  * Pyramids-and-Polyhedrons
  * sheetmetal 0.2.63
  * Silk 0.1.3
  * CurvedShapes 1.0.4
  * DynamicData 2.46.0
  * FreeCAD_assembly3 0.11.3
  * fcgear 1.0.0
  * ExplodedAssembly
  * A2plus 0.4.60n
  * fasteners 0.4.59
  * Render 2023.2.6
[/code]
```

### Subproject(s) affected?

None

### Problem description

Whenever FreeCAD starts the `Start` workbench creates a folder in `/tmp`, named `/tmp/FreeCADStartThumbnails⍰⍰⍰⍰⍰⍰⍰`. These never get removed so over time a lot of theses folders pile up in `/tmp`.

The folder should be removed if FreeCAD terminates.

see: https://forum.freecad.org/viewtopic.php?p=697560&sid=c7efd2a32a269cdf1eceb2de29017068

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-09-29:
@yorikvanhavre this is Start WB related

xtemp09, 2023-09-29:
The `/tmp` directory is managed by OS and is automatically cleaned upon system reboot or on schedule. Might I ask why this should be removed when FreeCAD terminates? Is there a rule?

yorikvanhavre, 2023-09-29:
The fact that these files are not deleted by FreeCAD is useful as they can then be reused by a later session. But then indeed as discussed in #10414 one should put reusable files not in a temp directory but in a cache directory.

I think once things are clearly separated between disposable files (in temp) and cached files (in cache) then this very question stops being one, as we can let the system do garbage treatment in temp.

luzpaz, 2023-09-29:
> The fact that these files are not deleted by FreeCAD is useful as they can then be reused by a later session.

Per @berberic2 though '*These never get removed so over time a lot of theses folders pile up in /tmp.*'. I'm not sure FreeCAD is re-utilizing them at all.

yorikvanhavre, 2023-10-02:
> Per @berberic2 though 'These never get removed so over time a lot of theses folders pile up in /tmp.'. I'm not sure FreeCAD is re-utilizing them at all.

Some are and some aren't. The thumbnails are reused. But they should be moved out of /tmp.

yorikvanhavre, 2023-10-02:
Indeed that's not optimal at all, it will only work for current session...

chennes, 2023-10-02:
Switching those thumbnails to use cache instead of tmp seems like low-hanging fruit to me, right, @yorikvanhavre ? Just use  `getUserCachePath()` instead?

yorikvanhavre, 2023-10-03:
There are several actions needed, because ATM the thumbnails are created with a random filename, that can be used only within that FreeCAD session. We also need to change that to create thumbnails that further sessions of FreeCAD can match back to a given file

- use `getUserCachePath() instead of tmp in https://github.com/FreeCAD/FreeCAD/blob/master/src/Mod/Start/StartPage/StartPage.py#L362
- create a function that creates a unique filename from a file path
- replace all uses of mkstemp in https://github.com/FreeCAD/FreeCAD/blob/master/src/Mod/Start/StartPage/StartPage.py by that function
- search for existing thumbnails before creating a new one

Another easier option is to discard and delete all this after each run... It's not that it takes so much time to rebuild, or that rendering speed is of paramount importance here. Also caching the thumbnails has the problem that we should ideally check the timestamp to know when the file has been changed, and recreate the thumbnail

luzpaz, 2023-10-06:
Folks please help review #10951

commit 2bd89e57 (2023-10-09): Start: fixes #10221: /tmp/FreeCADStartThumbnails* directories are not removed when FreeCAD terminates

The commit adds some new functions:

* getThumbnailDir()
  Returns the path to the thumbnail directory which is a sub-directory of the Cache directory.
  There is no need to always create a unique directory after each restart because it doesn't harm
  if the thumbnail directoy contains deprecated files.

* createThumbnailDir()
  Creates the thumbnail directoy if it doesn't exist yet.

* getSha1Hash
  Helper function to compute a SHA-1 hash of a given path. If the same path is passed
  then th [... cut]
commit eb07bad7 (2023-11-10): Start: fixes #10221: /tmp/FreeCADStartThumbnails* directories are not removed when FreeCAD terminates

The commit adds some new functions:

* getThumbnailDir()
  Returns the path to the thumbnail directory which is a sub-directory of the Cache directory.
  There is no need to always create a unique directory after each restart because it doesn't harm
  if the thumbnail directoy contains deprecated files.

* createThumbnailDir()
  Creates the thumbnail directoy if it doesn't exist yet.

* getSha1Hash
  Helper function to compute a SHA-1 hash of a given path. If the same path is passed
  then th [... cut]
closed 2023-10-09 by yorikvanhavre