FreeCAD/FreeCAD #10235: [Problem]  Using arithmetic operators in parameter boxes is no longer honored

DavidGriffith, 2023-08-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
[code]
OS: Devuan GNU/Linux 5 (daedalus) (MATE/lightdm-xsession)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.33807 (Git)
Build type: Release
Branch: master
Hash: 6a7f47c9596d6d2d5940d9089435614ec1a1cb19
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Assembly3 0.12.0
  * Pyramids-and-Polyhedrons
  * sheetmetal 0.2.63
  * freecad.gears 1.0.0
  * Assembly4 0.50.2
  * A2plus 0.4.60n
  * fasteners 0.4.59
  * Manipulator 1.5.0
[/code]
```

### Subproject(s) affected?

None

### Problem description

I recently upgraded from 9dce2525f8efe0873787f4388f10e11e168fe16b (dated August 1) to 6a7f47c9596d6d2d5940d9089435614ec1a1cb19 (dated August 13) and found that I can no longer move something by typing in a position or angle parameter box an arithmetic operator followed by a number. For example, the X parameter is "34 mm". I could move it back or forth by 13mm by typing "- 13" or "+13" to the left or right of the "mm". Similarly, I could multiply and divide. Now when I do that, nothing changes in the boxes and there is no change in position of the thing I'm trying to move.

Today I posted to https://forum.freecad.org/viewtopic.php?t=80440 to see if I did something dumb and was told the problem may be related to 8cad3940a1daf411ac61954ae162fdf23206342e.  A casual perusal of issue reports from August 1 to now shows nothing similar.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2023-08-15:
Now that I've had a chance to test two different self compiled systems, I cannot replicate this issue therefore if the OP is self compiling and they have the last known good commit, they can carry out a _git bisect_ in the source folder using the full commit hash from August 1st as _good_ and the commit from August 13th as _bad_. See https://wiki.freecad.org/Source_code_management#Bisect

DavidGriffith, 2023-08-20:
I'm doing a bisect now.  It'll take a bit before it's complete.

DavidGriffith, 2023-08-21:
Bisecting is complete.  The trouble started with 8cad3940a1daf411ac61954ae162fdf23206342e.  At this point and for sometime afterwards, at least to 6a7f47c9596d6d2d5940d9089435614ec1a1cb19; no arithmetic was allowed at all.  

Past 6a7f47c9596d6d2d5940d9089435614ec1a1cb19 might if you add/subtract AND add the units.  Multiplication and division would work without adding the units.  I have not yet identified the point where this partial support came back.

DavidGriffith, 2023-08-22:
I apologize.  That absolutely no-arithmetic at all thing doesn't really exist.  I'm not sure why I thought it did.  It remains though that I feel that appending `+ 4` to the string `2 cm` should result in `6 cm` instead of nothing happening.  I'm guessing that the impetus for 8cad3940a1daf411ac61954ae162fdf23206342e was https://github.com/FreeCAD/FreeCAD/issues/9864.

Syres916, 2023-08-22:
> It remains though that I feel that appending `+ 4` to the string `2 cm` should result in `6 cm` instead of nothing happening. 

This works for me exactly as you describe (I'm using Building Euro units scheme for testing):

Placement->Position->x `2.00 cm + 4` and then _Tab_ key results in `6.00 cm`

Using:

```
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.33942 (Git)
Build type: Release
Branch: master
Hash: 5177007994d6673dff37d0243f46bfd99c2f7648
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * Assembly4 0.50.2
  * Silk 0.1.3
  * A2plus
  * Defeaturing 1.2.0
  * CurvedShapes 1.0.4
  * MakerWorkbench 1.0.1
  * freecad.gears 1.0.0
  * FreeCAD_Turning_Addon 0.2.0
  * Curves 0.6.13
  * Help 1.0.3
  * toSketch 1.0.1
  * AirPlaneDesign 0.4.0
  * fasteners 0.4.56
  * sheetmetal 0.2.63

```

DavidGriffith, 2023-08-23:
Excellent!  Evidently the behavior I was missing got put back in sometime after 6a7f47c9596d6d2d5940d9089435614ec1a1cb19.

closed 2023-08-23 by DavidGriffith