FreeCAD/FreeCAD #10248: [Feature Request] Making symmetry work on arc with constrained center again

FEA-eng, 2023-08-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.0.33668 +7 (Git)
Build type: Release
Branch: (HEAD detached at 0.21)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

Sketcher

### Problem description

In 0.21, if an arc from center has the center coincident with the sketch's origin or with an endpoint of a line, the symmetry constraint applied to the endpoints of the arc and vertical axis (or aforementioned line) fails due to partially redundant constraints. However, in 0.20 it used to work. I'd like to ask to allow such a behavior in the next version since it's more convenient for the user.

0.21:

[image]

0.20.2:

[image]

Perhaps more general FR should be created on this occassion (like "making the auto remove redundants option handle partially redundant constraints" or "making the auto remove redundants option work better with redundancies caused by the symmetry constraint") but for now I'm mainly concerned about this particular case. Anyway, it's all about making the constraint handling more user-friendly, especially since there are no such issues in other CAD software.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PLChris, 2023-08-30:
FEA-eng, you forgot that you have autoremove redundants enabled.
I wouldn't call the current behaviour a regression, but rather an improvement. "Partially redundant" means that there is a redundancy and this should be removed by appropriate other constraints.

FEA-eng, 2023-08-31:
I am aware that this option is enabled in my case (otherwise, I would have to struggle with basic constraints that work normally in all other CAD applications). But this change can be an improvement only in terms of agreeing with theoretical concepts and nomenclature. But for a user it's a regression since previously I was able to draw something that should work in any CAD software and now I have to use some workarounds to achieve the same effect. I don't really care how it's handled internally, it doesn't have to be resolved by "auto remove redundants", there can be an additional option like "auto remove partial redundants", "make symmetry constraint work as expected" or anything. As I said, there might be a need for more general discussion about making constraints (especially this one) more user-friendly but here I want to specifically point out the behavior that changed in a negative way for users (at least for the ones that don't want to count DOFs and think about potential redundancies/partial redundancies when sketching).

luzpaz, 2023-11-14:
What caused this regression? 

CC @PaddleStroke any idea ?

PaddleStroke, 2023-11-15:
Sorry I do not know this part of sketcher.

PaddleStroke, 2023-11-15:
I think that autoconstraint was writen by @abdullahtahiriyo. Or at least I think he'll know what's going on with partial redundant.

brettowe, 2024-01-25:
I stumbled across this bug today, I don't have auto remove redundant constraints on.
so after the arc points are marked to be symmetric to the axis line it will say its redundant.
but if you grab one of the arc end points and try to rotate the arc after release it will snap back to center and report its under constrained.
closing the sketch and re opening it will go back to redundant message
I did test in 21.2 and a weekly dev snapshot and it acts the same.

FEA-eng, 2025-01-16:
Still relevant

Syres916, 2025-01-29:
So the first thing to do is a Git Bisect of the 0.21.x dev cycle to find out which commit resulted in the regression, if no-one's going to do it, I'll sort it tomorrow but I'll have to use my old distro as there will be a significant number of errors if I use a Python 3.12/OCCT7.8.x based distro.

Syres916, 2025-01-30:
So Git Bisect determined the commit that changed the warning output of `The Sketch has partially redundant constraints!` to not creating the symmetry constraint was https://github.com/FreeCAD/FreeCAD/commit/b626caffdc53614a0f71bb578a987e9712608bac

Now there's no way I'm changing that code as it will definitely cause regressions elsewhere but I can look at whether the coincident constraint with the origin can be automatically set to be a constraint to axis instead.

Syres916, 2025-01-31:
There's something about the logic brought in by https://github.com/FreeCAD/FreeCAD/commit/b626caffdc53614a0f71bb578a987e9712608bac that doesn't feel correct. The following output is from the old working code and the new non-working code:

Working:

```
16:34:41  Chosen redundants: [2 ]
16:34:41  Sketcher::RedundantSolving-DogLeg-
16:34:41  Sketcher Redundant solving: 1 redundants
16:34:41  (Partially) Redundant, Group 0, index 2, Tag: 2
16:34:41  Sketcher::setUpSketch()-T:0
```

Not working:

```
16:37:20  Chosen redundants: [2 2 ]
16:37:20  Sketcher::RedundantSolving-DogLeg-
16:37:20  Sketcher Redundant solving: 2 redundants
16:37:20  (Partially) Redundant, Group 0, index 1, Tag: 2
16:37:20  Sketcher::setUpSketch()-T:0.001
```

The bit that I have an issue with is that one Partially redundant constraint gets added twice then suddenly making it a fully Redundant constraint hence it gets deleted with _Automatically remove redundants_ ticked.

PaddleStroke, 2025-08-25:
@gima is right here. This is actually no bug, the sketch is overconstrained because of the coincidence constraint that is on the center.

So this is the same issue (more or less) as https://github.com/FreeCAD/FreeCAD/issues/13645

FEA-eng, 2025-08-25:
@PaddleStroke I guess it's just an example showing that "Auto remove redundants" should be improved to handle more cases (e.g. partial redundancy). I assume it's not an easy thing to implement but this is one of the biggest inconveniences in the Sketcher - having to plan each constraint so much. There should be an option to handle it in a smarter way - help the user by removing unnecessary constraints and leaving only the ones that result in the (predicted) desired behavior. Sounds complicated but it mostly applies to symmetry constraint which is particularly prone to overconstraint.

PaddleStroke, 2025-08-25:
I agree that it sucks.
How other CAD packages manage these situations?

PaddleStroke, 2025-08-26:
> just resolves them internally without even a warning.

Yes this is what we want. The question is how :)

Looking at catia v6 I see that when you apply the symmetric, there is also a symmetric icon that appears on the center. So it looks like they are replacing the coincidence with something else.