FreeCAD/FreeCAD #10250: [Problem] Cannot open any Autocad DWG files

antonical, 2023-08-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
[code]
OS: Ubuntu 22.04.2 LTS (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33675 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.0)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
[/code]
```

### Subproject(s) affected?

None

### Problem description

Thought I would try this product to see if it was useful. Product appimage runs and displays startup window. elect file open and select any of the architects .dwg files and it appears to be doing something. Then just disappears. No warnings, just gone!

Tried multiple different files. 

Not sure what else I can provide. Platform is Ubuntu 22.04 system has 12 cores and 32 gb of ram and a Nvidia Quadro P400.

Cheers
Tony
 

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

d1nuc0m, 2023-08-16:
Currently FreeCAD supports different methods for DWG conversion (see wiki [FreeCAD and DWG Import](https://wiki.freecad.org/FreeCAD_and_DWG_Import)), the QCAD trial message is about the usage of [QCAD](https://qcad.org), an open 2D CAD which has a paid verion that supports DWG. 

I don't know if/how it works in the AppImage and how to buy a license, meanwhile you might try another conversion method:

- Follow the instructions in the wiki page to install LibreDWG or ODA Converter
- Go to Preferences/Umport-Export/DWG
- Select LibreDWG or ODA Converter in the dropdown menu and select the path to the file converter

luzpaz, 2023-08-16:
Welcome to closed format 'lock-in' and the problems it causes for users.

luzpaz, 2023-08-16:
There isn't much we can do ATM here on the bugtracker for you. Best to bring this to the community and see if upstream importers can be improved at all (you'll need to supply a file though).

closed 2023-08-16 by luzpaz

antonical, 2023-08-16:
I don't have qcad installed all I have done is download the latest appimage and run it!

d1nuc0m, 2023-08-16:
> I don't have qcad installed all I have done is download the latest appimage and run it!

Which OS are you using?