FreeCAD/FreeCAD #10251: [Problem] Can't paste into the model tree in a KDE Wayland session using Flatpak

Chrismettal, 2023-08-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
OS: KDE Flatpak runtime (KDE/plasmawayland)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33675 (Git)
Build type: Release
Branch: (HEAD detached at 41b058e)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.10.11, Qt 5.15.10, Coin 4.0.0, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods:
  * Dracula 0.0.4
  * FreeCAD-Desktop-Preference-Pack 0.0.1
  * kicadStepUpMod 10.19.4
```

### Subproject(s) affected?

None

### Problem description

Using the Flatpak version of FreeCAD in a Wayland session does not allow any pasting of any object, feature or body into the model tree. This includes pasting inside the same project, as well as pasting a body from another project.

In this example, I copied a simple body "Case" inside this project, and wanted to paste an independent copy into the Part above, but in a Wayland session I do not get to paste anywhere in the model tree:

[image]

Doing the same thing with the same project file on the same system in an X11 session allows me to paste just fine:

[image]

I have tested this behavior between an X11 and Wayland session on the same system.

Version exhibiting this issue:

```
OS: KDE Flatpak runtime (KDE/plasmawayland)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33675 (Git)
Build type: Release
Branch: (HEAD detached at 41b058e)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.10.11, Qt 5.15.10, Coin 4.0.0, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods:
  * Dracula 0.0.4
  * FreeCAD-Desktop-Preference-Pack 0.0.1
  * kicadStepUpMod 10.19.4

Operating System: Arch Linux
KDE Plasma Version: 5.27.7
KDE Frameworks Version: 5.108.0
Qt Version: 5.15.10
Kernel Version: 6.4.10-arch1-1 (64-bit)
Graphics Platform: Wayland
Processors: 32 × AMD Ryzen 9 7950X 16-Core Processor
Memory: 61.9 GiB of RAM
Graphics Processor: AMD Radeon RX 7900 XTX
Manufacturer: ASRock
Product Name: X670E Steel Legend
```

Version NOT exhibiting this issue:

```
OS: KDE Flatpak runtime (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33675 (Git)
Build type: Release
Branch: (HEAD detached at 41b058e)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.10.11, Qt 5.15.10, Coin 4.0.0, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods:
  * Dracula 0.0.4
  * FreeCAD-Desktop-Preference-Pack 0.0.1
  * kicadStepUpMod 10.19.4

Operating System: Arch Linux
KDE Plasma Version: 5.27.7
KDE Frameworks Version: 5.108.0
Qt Version: 5.15.10
Kernel Version: 6.4.10-arch1-1 (64-bit)
Graphics Platform: X11
Processors: 32 × AMD Ryzen 9 7950X 16-Core Processor
Memory: 61.9 GiB of RAM
Graphics Processor: AMD Radeon RX 7900 XTX
Manufacturer: ASRock
Product Name: X670E Steel Legend
```

### Anything else?

Project this was tested with:
[Flipper-Zero-Backpacks FreeCAD file](https://github.com/Chrismettal/flipper-zero-backpacks/blob/master/3d_printing/flipper-zero-backpacks.FCStd)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Chrismettal, 2023-08-16:
Potentially related to https://github.com/FreeCAD/FreeCAD/issues/9392

adrianinsaval, 2023-09-21:
this is weird because the flatpak version should be running in xwayland, where it should behave the same as in an x11 session, I don't know much about how the clipboard works on freecad. @wwmayer or @PaddleStroke could you give some insight? Where do I even begin to look into this?

PaddleStroke, 2023-09-21:
Maybe the discussion there will help : https://github.com/FreeCAD/FreeCAD/pull/5480

> Cut/Copy/Paste commands are defined in https://github.com/FreeCAD/FreeCAD/blob/master/src/Gui/CommandDoc.cpp

closed 2025-02-07 by github-actions[bot]