FreeCAD/FreeCAD #10253: [Problem] Legacy DXF importer gets import scale wrong for 'inch' files

Roy-043, 2023-08-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.21.0.33675 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.0)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: Dutch/Netherlands (nl_NL)
Installed mods:
```

### Subproject(s) affected?

Draft

### Problem description

A DXF with INSUNITS set to 1 (=Inches) is not scaled upon import with the legacy DXF importer.

Attached a ZIP with two files:
508mm_x_254mm_V2000.dxf
20in_x_10in_V2000.dxf

Both contain a rectangular polyline that should result in a shape with the same dimensions when imported in FreeCAD.

[DXF_import_test.zip](https://github.com/FreeCAD/FreeCAD/files/12360933/DXF_import_test.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

yorikvanhavre, 2023-09-01:
>  However it does appear that INSUNITS has become the de-facto indicator of units through the whole DXF file whether DesignCenter is involved or not.

Yes that's my understanding as well.

IIR INSUNITS is not supported by the legacy importer. It does not seem like the dxfReader lib can give that to us, so it's probably a matter of opening and parsing the actual file... Then the dxfScaling value should be overwritten if INSUNITS is present. Only remaining question, what to do if the user set a custom scaling value themselves? Which has the priority?

Maybe we only override the scaling value if it's zero or not set... Otherwise we use the user-provided scaling value, regardless of INSUNITS?

Roy-043, 2023-09-03:
There is also the MEASUREMENT that can provide information if INSUNITS is missing.
MEASUREMENT == 0 -> 'Imperial': inch.
MEASUREMENT == 1 -> 'Metric': mm.

> Maybe we only override the scaling value if it's zero or not set... Otherwise we use the user-provided scaling value, regardless of INSUNITS?

Yes, that is a good approach. So in order of priority:
1. Use dxfScaling if non-zero.
2. Use INSUNITS if available.
3. Use MEASUREMENT if available.
4. Do not scale.

It may be a good idea to inform the user about the detected units and the proposed scaling?

Different, but related, issue: Scaling can result in huge coordinates which FreeCAD does not handle very well.

yorikvanhavre, 2023-09-04:
That plan seems perfect.

> Different, but related, issue: Scaling can result in huge coordinates which FreeCAD does not handle very well.

Indeed, but there is I think not much we can do beforehand, because there is no way to know how large the model will be before importing. Maybe display a warning message after the import?

kpemartin, 2023-09-04:
What is the issue with "huge coordinates"? How "huge" is "huge" before there are issues?
I don't really think this (huge coordinates) is an importer problem at all, and one should certainly be able to work around it by using the scaling factor option for the import. In any case if this is an issue, it should be made into an Issue if there is not one already.

Treating the dxf import scaling option being 0 as meaning "infer from dxf file contents, defaulting to 1 unit = 1mm" and non-zero means "treat dxf units as this many mm" might want some re-wording of the label text for the options dialog, and clarification in any documentation (Q: Is there any documentation for this?).

This still leaves the user who has an old (pre-INSUNITS-etc) inch-scaled dxf file (like many of mine) having to enter that awkward constant which is 1/25.4.

The most complex but most generally usable method would be for the options to include a drop-down list of scaling choices (probably all the possible meanings of INSUNITS) which select how to interpret one dxf unit, with the additional choice: "Infer from DXF file contents" (i.e. use INSUNITS etc), which could either treat a dxf file with no such info as being in mm or being an error (this is an implementation choice, I somewhat prefer the latter). In addition the existing scaling factor would always be applied (if non-zero) and would default to 1 (i.e. no additional scaling). But I don't know how much trouble it is to change the options, even ignoring considerations of backwards compatibility with saved option sets, this is text that needs translation so who provides the translations?

Barring UI changes for options, I think the easiest-to-explain method is:
If INSUNITS or MEASUREMENT are in the file, use them to infer the size of the drawing units, otherwise assume 1 unit = 1mm.
Always apply (in addition) the specified scaling factor.

This does mean that someone importing a dxf file that contains these variables specifying a unit other than 1mm will get a different imported drawing that they would have with older FreeCAD. This includes the OP but the new behaviour is what they want.  I don't know if this would be considered a significant regression to other users: how often do users import the same file over and over again?

In any case, both importers should behave the same in these respects, and I haven't looked at what the new importer does.

It would certainly be useful for the importer (both of them) to issue a message explaining what scaling was used and how it was determined. There is however no good way to ask the user if this is OK before doing the actual import.

yorikvanhavre, 2023-09-06:
> This does mean that someone importing a dxf file that contains these variables specifying a unit other than 1mm will get a different imported drawing that they would have with older FreeCAD. This [...]

I think this is totally okay. INSUNITS need to be supported, and indeed if it changes the way older DXF files were imported before, I think everybody will consider it an improvement anyway, even if they have to adjust something.

Only case I'm not sure: I've seen a lot of people never ever touching AutoCAD's block insert units setting (which controls INSUNITS) and leaving it to the default (which is often inches) even when working in meters (they just have their blocks saved that way too). So it could be that these people suddenly get their model inserted at a wrong scale.

One way to solve this would be: If the dxf scaling preference setting is not zero or one,  then that setting overrides everything else. Otherwise, if INSUNITS is present, use it. Otherwise,1 dxf unit = 1 mm

Roy-043, 2023-09-06:
FWIW: another example with huge coords:
https://forum.freecad.org/viewtopic.php?t=80476

PR #11318 (2023-11-10): Draft: fix import scale for Legacy DXF importer
Fixes #10253.
commit 6ec2ecff (2023-11-10): Draft: fix import scale for Legacy DXF importer

Fixes #10253.
closed 2023-11-10 by yorikvanhavre