FreeCAD/FreeCAD #10254: [Problem] Legacy DXF importer cannot handle layers that are OFF

Roy-043, 2023-08-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.21.0.33675 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.0)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: Dutch/Netherlands (nl_NL)
Installed mods:
```

### Subproject(s) affected?

Draft

### Problem description

DXF Layers that are OFF have a negative ACI color index. The legacy importer doesn't handle this properly:
```
17:46:13  Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "D:\BKG_Tmp\FreeCAD_0.21\Mod\Draft\importDXF.py", line 2766, in insert
    processdxf(doc, filename)
  File "D:\BKG_Tmp\FreeCAD_0.21\Mod\Draft\importDXF.py", line 2157, in processdxf
    color = tuple(dxfColorMap.color_map[layer.color])
<class 'KeyError'>: (-1,)
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2023-08-16:
ZIP with test file:
[Layer_Off_V2000.zip](https://github.com/FreeCAD/FreeCAD/files/12361214/Layer_Off_V2000.zip)

luzpaz, 2023-08-16:
@kpemartin care to take a stab at this ?

kpemartin, 2023-08-17:
Yes, this is on my to-do list already. Making it do something (rather than failing) is easy; making it do something sensible (e.g. making the objects in that layer Hidden in FC) will require a little more analysis on my part to see how to do this.

And now I see I already have that wrong, I don't want the objects or layer Hidden, I want the layer Visibility toggled OFF.

For that matter Frozen layers should also be toggled OFF. I don't know if FC has any equivalent functionality to frozen layers, which are not visible and are immune to any attempts to modify them.

Roy-043, 2023-08-17:
FYI: there is a very particular behavior when inserts (= block references) are on a layer that is off. Their content (provided their layers are not off) will still be rendered in AutoCAD. This is not the case for layers that are frozen: inserts on frozen layers are not rendered at all in AutoCAD.

(Optionally) not importing geometry on frozen layers may be a good idea.

yorikvanhavre, 2023-09-01:
I think we could do this:
- the color is just a matter of taking `abs(color)`
- if a layer has a negative color, it should have its visibility off (`Layer.ViewObject.Visibility = False`)
- we don't care about the layer of individual block components anyway.. I think we don't need to do anything here for now
I wouldn't care about frozen layers for now, because it's a different matter, and there is more work (there should be a pref setting for it, etc) it could be done in another PR

Roy-043, 2023-09-03:
I am fine with that.

PR #11090 (2023-10-19): Draft: importDXF: fix off layer bug
Fixes #10254.
commit 7eda70c7 (2023-10-19): Draft: importDXF: fix off layer bug (#11090)

* Draft: importDXF fix off layer bug

Fixes #10254.

* Typo

* Typo 2
commit 5a4648e5 (2023-10-19): Draft: importDXF: fix off layer bug (#11090)

* Draft: importDXF fix off layer bug

Fixes #10254.

* Typo

* Typo 2
closed 2023-10-19 by yorikvanhavre