FreeCAD/FreeCAD #10255: [Problem] Crash on execution of transform

sfmissionmark, 2023-08-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
[code]
OS: macOS 13.5
Word size of FreeCAD: 64-bit
Version: 0.21.0.33575 (Git)
Build type: Release
Branch: master
Hash: 2559d7e7b63081116a2b3509c2230bcbef8ebf88
Python 3.11.0, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * ThreadProfile 1.87.0
  * Behave-Dark-Colors 0.1.1
  * ProDarkThemePreferencePack 1.0.0
  * A2plus 0.4.60n
  * fasteners 0.4.58
  * ExtremeProDark 2.9.0
[/code]
```

### Subproject(s) affected?

None

### Problem description

Transforming an object crashes on macOS when clicking ok button.

To Reproduce create a new document.
Go to Draft workbench create a cone.
control click on the cone and transform.
Rotate the object on any axis.
Click OK button.

App hangs and eventually crashes.

### Anything else?

Crash log shows this. 
Full crash log available.

-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               freecad [48461]
Path:                  /Applications/FreeCAD.app/Contents/Resources/bin/freecad
Identifier:            freecad
Version:               ???
Code Type:             X86-64 (Native)
Parent Process:        Exited process [48455]
User ID:               501

Date/Time:             2023-08-11 09:02:31.4360 -0700
OS Version:            macOS 13.4.1 (22F770820d)
Report Version:        12
Bridge OS Version:     7.5 (20P5060)
Anonymous UUID:        90EBA5F8-3D40-BCEE-1A25-6668B15CC6F6

Sleep/Wake UUID:       16693902-6210-4C45-863D-CA6815A072C8

Time Awake Since Boot: 1000000 seconds
Time Since Wake:       78806 seconds

System Integrity Protection: enabled

Crashed Thread:        0  CrBrowserMain  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000028
Exception Codes:       0x0000000000000001, 0x0000000000000028

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [48461]

VM Region Info: 0x28 is not in any region.  Bytes before following region: 140737488023512
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      shared memory            7ffffffaf000-7ffffffb0000 [    4K] r-x/r-x SM=SHM  

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2023-08-17:
The Draft WB does not have a cone object. Probably the Part WB is meant.

sfmissionmark, 2023-08-17:
> The Draft WB does not have a cone object. Probably the Part WB is meant.

Yes I am sorry. I meant part wb. It was an example transforming anything crashes.

Roy-043, 2023-08-17:
This issue may be a duplicate. See #9465.

chennes, 2023-08-17:
Please install the latest release of 0.21.0 for Mac OS -- the bug is a problem with the Coin library version in the Conda build.

sfmissionmark, 2023-08-25:
0.21.0 works fine. Is there any work arounds for TOT?

adrianinsaval, 2023-08-31:
closing as duplicate

closed 2023-08-31 by adrianinsaval