FreeCAD/FreeCAD #10261: [Feature Request] Automatic vertical and horizontal constrain

NXTler, 2023-08-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
[code]
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33458 (Git) AppImage
Build type: Release
Branch: master
Hash: 8b89dd29d2382c72d738c52e8974216130ffa9db
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * PieMenu 1.2.7
  * sheetmetal 0.2.63
  * toSketch 1.0.1
  * A2plus 0.4.60n
  * QuickMeasure 2022.10.28
  * Curves 0.6.12
  * CfdOF 1.23.1
  * TabBar
  * IconThemes
  * Glass
  * ProDarkThemePreferencePack 1.0.0
[/code]
```

### Subproject(s) affected?

Sketcher

### Problem description

Hello all,
due to the vertical and horizontal constrain being separate, you have to do a lot of switching back and forth while sketching, which makes the workflow feel chunky. Other CAD programs get around this, by having only one function, which applies the right constrain based on if the line is closer to vertical or horizontal.
In case of FreeCAD, I would leave the existing functions as they are and only add a function on top, which chooses as stated.

Many thanks in Advance

### Anything else?

Here is an example from another CAD programs:
[image]
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2023-08-17:
So pressing the `H` key or the `V` key is too clunky? I've been using FreeCAD for 6 years on a near daily basis for real work and have never wanted or even thought the functionality you are suggesting was necessary.

Roy-043, 2023-08-17:
There is also the "Auto constraints" feature. When switch on, new lines that are nearly horizontal or vertical automatically receive the corresponding constraint. See [Sketcher Dialog](https://wiki.freecad.org/Sketcher_Dialog).

NXTler, 2023-08-17:
I didn't even know that, the `h` `v` shortcut existed... That explains a lot, thanks!
I find the "Auto constraints" feature a bit too finicky from time to time, and I have a lot of cases where I set the constraints afterward.

closed 2023-08-17 by NXTler

FEA-eng, 2023-08-21:
There's another issue with separate horizontal/vertical constraints - when you work with a rotated view of the sketch, those constraints often have switched meanings (so you have to use vertical for lines that are viewed as horizontal and vice versa). Such automatic constraint would eliminate this issue.

NXTler, 2023-08-21:
Yeah, this is also a reason, why I wanted this feature. Though, I think I will open another Feature Request for this one, since one Solution might be to automatically change the meaning internally without extra function. You can reopen this one, if you feel like a better solution.

luzpaz, 2023-08-22:
@NXTler please open a new ticket and link to this one so we have continuity

PR #11330 (2023-11-20): [Sketcher] Auto horizontal/vertical
Adds a auto horizontal / vertical constraint and command group.
Refactor horizontal/vertical commands which had lot of duplicates.
Fixes #10320 
Fixes #10261

Also participate solving https://github.com/FreeCAD/FreeCAD/issues/11208