FreeCAD/FreeCAD #10265: [Problem] Dressup Tags can not be copied

ModuloFS, 2023-08-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
[code]
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.21.0.33668 +7 (Git)
Build type: Release
Branch: (HEAD detached at 0.21)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * A2plus 0.4.60k
  * btl 0.9.5
  * Curves 0.6.8
  * FeedsAndSpeeds 0.5.0
  * Lithophane
  * Manipulator 1.5.0
  * ProDarkThemePreferencePack 1.0.0
  * QuickMeasure 2022.10.28
[/code]
```

### Subproject(s) affected?

Path

### Problem description

As descriped in this forum thread: https://forum.freecad.org/viewtopic.php?t=80507 i am not able to copy the tags from a tag dressup to another dressup tag. 

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2025-01-17:
@ModuloFS can you test if this is still an issue in weekly or stable ?

ModuloFS, 2025-01-18:
Yes, still present in:

OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.1.0dev.39896 (Git) Conda
Build type: Release
Branch: main
Hash: a977fade2de741c94c9efa59b1c2c26df18eb631
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * fasteners 0.5.33
  * MakerWorkbench 1.0.1
  * QuickMeasure 2022.10.28
  * Road-main 2025.1.1

FEA-eng, 2025-09-14:
@sliptonic Still relevant?

tarman3, 2026-04-26:
To start work on this need actual project
File from forum link created in FreeCAD 0.21 and opened with errors in FreeCAD 1.1

Probably this PR can be closed now
It can be opened later if someone attach FCStd file, saved at least in FreeCAD 1.0