FreeCAD/FreeCAD #10273: OpenSCAD file export hangs indefinetly

fallingcats, 2023-08-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
[code]
OS: Arch Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33675 (Git)
Build type: Release
Branch: makepkg
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.11.3, Qt 5.15.10, Coin 4.0.1, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

None

### Problem description

Trying to export my model as either .scad or .csg seems to hang indefinitely (>30min), with the process using 100% of one core.

I don't know if it would work with other projects or other versions of freecad.

### Anything else?

[tested project, rename to .FCStd](https://github.com/FreeCAD/FreeCAD/files/12383606/activ.energy.XYZ562a.socket.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-08-18:
@KeithSloan may be AFK but JIC I'm pinging him  (CC @chennes)

grubertechnic, 2023-08-19:
The default values of the export preferences are not good, when adapting it (Triangulation settings), it works flawless.

```
OS: Debian GNU/Linux 12 (bookworm) (X-Cinnamon/lightdm-xsession)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.33788 (Git)
Build type: Release
Branch: master
Hash: 58ef6c244e64c8d4b766e53615cf62dff9af7a68
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.7.1
Locale: C/Default (C) [ OS: English/United States (en_US) ]
```

fallingcats, 2023-08-19:
> The default values of the export preferences are not good, when adapting it (Triangulation settings), it works flawless.

Maybe I'm "holding it wrong", but scad/csg does not seem to have an import/export page? Neither can I find triangulation settings in any other page on preferences. In any case exporting stl works just fine and that would also be a triangles, no?

grubertechnic, 2023-08-21:
> but scad/csg does not seem to have an import/export page?

It has it own tab, since scad have its won workbench.

[image]

Note, that workbench must be loaded before (because of the lazy loader).

[image]

luzpaz, 2023-08-21:
wait, is the openscad not getting pre-loaded (therefore not in memory) ?

grubertechnic, 2023-08-21:
> wait, is the openscad wb not getting pre-loaded

I do not know, does also not really matter, since it can still switched off per hand. It was just a hint, that the preferences are only showed, when it is loaded and have its own tab.

luzpaz, 2023-08-21:
@fallingcats can you first load the openscad workbench and then try to reproduce the bug ?

fallingcats, 2023-08-21:
The OpenSCAD workbench is not loaded per default for me. I have not yet changed the default settings for that workbench

Manually loading it does not seem to prevent exporting my PartBench body in .csg from hanging.

fallingcats, 2023-08-21:
It does not hang if I manually change Triangulation settings from 0 to 0.1. Maybe that should be default?

grubertechnic, 2023-08-21:
Yes 0.1 is pretty rough. With 0.01 i get a, for that case, good result. I never use OpenSCAD for myself, what is a overall good default value?

Suggestions anyone?

luzpaz, 2023-08-21:
@chennes (openscad related)

chennes, 2023-08-21:
If changing that default prevents the hang I have no objection to it. I can't remember what default we have in the Mesh WB, but if I recall it was recent change from 0.1 to something smaller. Maybe we can make them match?

grubertechnic, 2023-08-21:
> I can't remember what default we have in the Mesh WB, but if I recall it was recent change from 0.1 to something smaller.

To be honest, i do not know what triangulation means in OpenSCAD in that context, seems to be a kind of ratio depends on a brep size per face. But on the Mesh WB there are other methods, from classic tessellation (deviation in % and angular deflection) to meshes with edge length.

fallingcats, 2023-08-22:
> @fallingcats have you tried with the "Alternate OpenSCAD WB" (available via the Addon Manager)?

Just to be clear, what should I try with an alternative workbench? I've not used the other one either. I'm modeling in Part Design and exporting as .scad from there.

benj5378, 2023-08-23:
I believe triangulation means that the CAD model (with real circles etc.) are converted to triangles, that only looks like circles. Question is, if setting triangulation to true is desired for all cases. For big models, yes, else it will hang forever. For small models, maybe no, depends if we want to have real splines/circles or polygon derivatives.

This is based on a guess of what triangulation means, I actually don't know.

benj5378, 2023-08-23:
> wait, is the openscad wb not getting pre-loaded (therefore not in memory) ?

Only workbench by defualt pre-loaded is Start Workbench

KeithSloan, 2023-08-30:
Sorry guys but I am at a total loss on this.

1) I do not recognise the export options in the OpenSCAD preferences, it is not something that I wrote and I cannot match up with the code in exportCSG.py. Maybe somebody enhanced exportCSG and somehow the code got lost perhaps?
@chennes Any ideas.

2) Looking at the code in exportCSG it is just handling Objects created in the Part workbench, the reason for the infinite loop is that if the Object being processed is not recognised i.e. been created by Part it does not issue any warning or move to the next object.

The code in process_object ends around line 245 with
````
    elif ob.isDerivedFrom('Part::Feature'):
        print("Part::Feature")
        mm = check_multmatrix(csg,ob,0,0,0)
        csg.write('%s\n' % shape2polyhedron(ob.Shape))
        if mm == 1 : csg.write("}\n")
````
It should then have at least something along the lines of
````
    elif ob.isDerivedFrom('Part::Feature'):
        print("Part::Feature")
        mm = check_multmatrix(csg,ob,0,0,0)
        csg.write('%s\n' % shape2polyhedron(ob.Shape))
        if mm == 1 : csg.write("}\n")

    else:
       print(f"Object {ob.TypeId} not handled")
       return
       # or maybe exit(3)
````

Why the need to rename all the test cases?
Could the directory not just be zipped and the zipped file name have .txt added.

But unless I am miss understanding I do not see how the export preferences come into play given the code,
can only assume the enhancements got lost or when adding code for export of Part::Objects as csg/scad code the function got lost.

KeithSloan, 2023-08-30:
Looking at the history for exportCSG.py [https://github.com/FreeCAD/FreeCAD/commits/master/src/Mod/OpenSCAD/exportCSG.py](url)

Code has not changed for a long time and I cannot see anything that would bring the OpenSCAD export Preference into play.

Also don't understand the rational of getting the OpenSCAD workbench to mesh.

KeithSloan, 2023-08-30:
MY BAD - Okay I stand corrected 

The code reads

````
elif ob.isDerivedFrom('Part::Feature'):
        print("Part::Feature")
        mm = check_multmatrix(csg,ob,0,0,0)
        csg.write('%s\n' % shape2polyhedron(ob.Shape))
        if mm == 1 : csg.write("}\n")
````

And shape2polyhderon(ob.Shape) does the tessellation and refers to the tessellation options, so maybe change the to
 ````
if hasAttr(ob,'Shape'):
       print('Has Shape')
       mm = check_multmatrix(csg,ob,0,0,0)
       csg.write('%s\n' % shape2polyhedron(ob.Shape))
       if mm == 1 : csg.write("}\n")

    else:
       print(f"Object Has no Shape")
       return
       # or maybe exit(3)

KeithSloan, 2023-08-30:
Okay I have made the change to the Alternate Importer, if you could try testing the change.

KeithSloan, 2023-09-01:
> > @fallingcats have you tried with the "Alternate OpenSCAD WB" (available via the Addon Manager)?
> 
> Just to be clear, what should I try with an alternative workbench? I've not used the other one either. I'm modeling in Part Design and exporting as .scad from there.

If you install the "OpenSCAD Alternate Importer" via the Tools | Addon Manager.

Then when you select your Part Design body and use File | Export
and use scad or csg as the file type the file will be exported with the exporter in the Aternate Workbench.

( Note : You may be prompted for which. exporter to use )

Need to remove extra .txt from the following
[Prop glue jpg87.FCStd.txt](https://github.com/FreeCAD/FreeCAD/files/12492130/Prop.glue.jpg87.FCStd.txt)
[Prop glue jpg87-Body.csg.txt](https://github.com/FreeCAD/FreeCAD/files/12492133/Prop.glue.jpg87-Body.csg.txt)
[image]
[image]

luzpaz, 2023-09-28:
CC @fallingcats

fallingcats, 2023-10-13:
Sorry for being so unresponsive.
Having installed the suggested plugin with its dependency and restarting, I don't see any new workbench, not even in `Preferences > Workbenches`.

Exporting does not ask me which workbench to use and still hangs with the openscad triangulation settings set to 0.

KeithSloan, 2023-10-13:
My Bad I should explain things more clearly and accurately

You will NOT see any new Workbench, it is implemented as Workbench but acts like a Module. I would change to being a Module but the OpenSCAD workbench maybe dropped in FreeCAD v1, so if that happens it is less work to adapt.

To use the new exporter you need to select the New Exporter when selecting the file type.
Note: FC exporter is a few entries below.

[image]

fallingcats, 2023-10-13:
Does the new exporter respect the old triang. settings? Not having changed them from 0, the new exporter seems to hang just like the old one.

Does it work for you? The file I'm testing with is provided the project file in the initial report. I'm happy to assist, but I don't want to cause additional friction by having you wait on my testing results if I can help reproduce it on your end.

KeithSloan, 2023-10-14:
Thanks for persevering

When you say the project file you are testing is as per the initial report, if I download the link I get directory activ with lots of file type document and another directory with thumbnails with file extension brp. I assume the file types for the documents need to be changed to FCStd. Which file are you testing? 

I recreated your problem with file I posted and used that for testing.

The test I ran before I did not change any OpenSCAD workbench options
[image]

When I changer the triangulation option in report view I got a barf
```"Empty widget item in QHBoxLayout 'horizontalLayout_21'."
08:30:35  "Empty widget item in QHBoxLayout 'horizontalLayout_21'."
```

And the options now look like
[image]

Still ran successfully with a larger mesh

[image]

My FreeCAD information is
[code]
OS: macOS 12.6.8
Word size of FreeCAD: 64-bit
Version: 0.21.0.33675 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.0)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * ImportNURBS 1.1.0
  * ImportExport_3DM 0.1.0
  * DynamicData 2.53.0
  * Pyramids-and-Polyhedrons
  * occi-freecad-plugin 0.1.0
  * GDML 2.0.0
  * Design456 0.0.1
  * Alternate_OpenSCAD 1.0.0
  * toSketch 1.0.1
  * Curves 0.6.13
[/code]

KeithSloan, 2023-10-14:
The code being exercised for an Object that drops through to having just a Shape is
```
def shape2polyhedron(shape):
    import MeshPart
    return mesh2polyhedron(MeshPart.meshFromShape(Shape=shape,\
        Deflection = params.GetFloat('meshdeflection', 0.0)))
```
So maybe your file is showing up a bug in mesh2polyhedron or MeshPart the Mesh workbench.

I assume you have tried creating an STL with your file, so that would point the suspicion at mesh2polyhedron.

Please could you test with the file I posted on your system, also let me know which file in your directory you are testing.

KeithSloan, 2023-10-14:
mesh2polyhedron is python code and for large meshes can be slow.

What if I look into wrapping it in a time out, which would then give you the option of making the mesh courser?

But it would be good to establish that what you are seeing as a hang is just a long running loop.

fallingcats, 2023-10-15:
> When you say the project file you are testing is as per the initial report, if I download the link I get directory activ with lots of file type document and another directory with thumbnails with file extension brp. I assume the file types for the documents need to be changed to FCStd

Thats exactly right, github doesn't allow uploading `.FCStd`, so I renamed it to `.zip`.

Trying with the prop.glue file, exporting works pretty much instantly with the New Exporter for me.
Likewise, exporting mine as `.stl` works fine as well, but exporting with New Exporter doesn't work for me with the settings shown

KeithSloan, 2023-10-15:
I must be missing something. I adding FCStd extension to a couple of files and FreeCAD fails to load them
```
12:47:43  <Exception> Invalid project file: /Users/keithsloan/Downloads/activ-4/LineColorArray20.FCStd
12:47:43  Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'OSError'>: Invalid project file: /Users/keithsloan/Downloads/activ-4/LineColorArray20.FCStd
12:48:11  <Exception> Invalid project file: /Users/keithsloan/Downloads/activ-4/PointColorArray20.FCStd
12:48:11  Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'OSError'>: Invalid project file: /Users/keithsloan/Downloads/activ-4/PointColorArray20.FCStd
```

fallingcats, 2023-10-15:
> I must be missing something. I adding FCStd extension to a couple of files and FreeCAD fails to load them
>
> ```
> 12:47:43  <Exception> Invalid project file: [...]

Please don't unzip the file I uploaded. Freecad projects are zip files in disguise, just rename it.

KeithSloan, 2023-10-15:
Okay on a Mac it automatically unzips. I have it now.

KeithSloan, 2023-10-15:
Okay it works with the triangulation setting value set to 1 ( very quickly ) but loops with the default 0.

I separated the statements to confirm that the problem is with

```
msh = MeshPart.meshFromShape(Shape=shape, Deflection = params.GetFloat('meshdeflection', 0.0))
```
Maybe 0 is probably not the best choice for a default? 

But maybe a bug in Mesh workbench and function meshFromShape, would be interested in Chris's thoughts?

I will look into wrapping the code in a timeout but obviously that does not fix if just a bug.

 @chennes 
 @luzpaz

KeithSloan, 2023-11-04:
> @KeithSloan any results from your examinations ?

I did have a quick look at it seems adding a timeout would require using two processes so not a quick fix
see https://medium.com/@chaoren/how-to-timeout-in-python-726002bf2291

Maybe somebody could like @wmayer could add some functionality to FC to make using a timeout easier. 

I have implemented a time out in the Alternate OpenSCAD workbench but that was where it calls OpenSCAD command line which is a separate process, so was just a case of adding some options.

luzpaz, 2023-11-05:
It's @wwmayer on github ;)

KeithSloan, 2023-11-05:
Good, think I have seen some other FC posts where people could usefully use a FC provided timeout but as usual I can not easily find them again.

luzpaz, 2024-08-20:
softest of bumps

luzpaz, 2025-06-17:
@KeithSloan i re-opened this issue. Is it worth keeping open ?

KeithSloan, 2025-06-17:
> Where is the discussion to drop openscad happening? It makes sense to make it a 3rd party addon instead.

If I remember correctly the discussion was at the first FreeCAD meeting in Brussels but I think there was some Forum posts about that time.