FreeCAD/FreeCAD #10274: [Problem] v21.0 Addon Manager preview causes QtWebEngine crash on macOS Ventura 13 (regression?)

systemresearch, 2023-08-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
[code]
OS: macOS 13.5.1 (mini x86_64)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33675 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.0)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * Assembly3 0.12.0
[/code]

[code]
OS: macOS 13.4.1 (mini M1 Arm)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33675 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.0)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * Assembly3 0.12.0
[/code]
```

### Subproject(s) affected?

Addon Manager

### Problem description

FreeCAD v0.21.0 Addon Manager preview for any selected addon fails with a "Could not load README from URL …" message, and causes a QtWebEngine crash report to be generated on macOS Ventura 13. Maybe this is a regression since previews worked OK with v0.20? Or, maybe something is amiss with the library dependency code signing relative to QtWebEngine?

I tested and found this issue on both macOS M1 Arm and macOS x86_64 machines. I did not find this issue on Ubuntu 18.04 or Windows 10 test machines.

Steps:

1. Launch FreeCAD v0.21.0 on a macOS Ventura system.
2. Select `Tools` > `Addon manager` menu.
3. Click on one of the addons. (Assembly3 was the first one I tried. Other had the same issue.)
4. Result: Addon manager displays "Could not load README from URL …"  message. 
    - Note: This message does not show in the FreeCAD "Report view" panel.
5. macOS Console.app > Crash Reports : "QtWebEngineProcess-2023-08-18-205443.ips" 

[image]

**Crash Report**

```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               QtWebEngineProcess [1682]
Path:                  /Applications/FreeCAD.v21_0.app/Contents/Resources/libexec/QtWebEngineProcess
Identifier:            QtWebEngineProcess
Version:               ???
Code Type:             ARM-64 (Native)
Parent Process:        freecad [1663]
Responsible:           freecad [1663]
User ID:               501

Date/Time:             2023-08-18 20:54:42.0145 -0700
OS Version:            macOS 13.4.1 (22F82)
Report Version:        12
Anonymous UUID:        BBE053F7-BF6B-31D9-3C81-30CE931F7337

Time Awake Since Boot: 22000 seconds

System Integrity Protection: enabled

Crashed Thread:        0  CrRendererMain  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGKILL (Code Signature Invalid))
Exception Codes:       UNKNOWN_0x32 at 0x0000007100088040
Exception Codes:       0x0000000000000032, 0x0000007100088040

Termination Reason:    Namespace CODESIGNING, Code 2 Invalid Page
```

Crash report with txt extension added and then zipped: [QtWebEngineProcess-2023-08-18-205443.ips.txt.zip](https://github.com/FreeCAD/FreeCAD/files/12385022/QtWebEngineProcess-2023-08-18-205443.ips.txt.zip)

### Anything else?

No forum discussion was found on this topic.

Note: The recent daily builds *did*  *not* have this issue. The following builds tested as being OK relative to the issue reported above:

- FreeCAD_weekly-builds-33848-2023-08-17-conda-macOS-arm64-py310.dmg
- FreeCAD_weekly-builds-33848-2023-08-17-conda-macOS-x86_64-py310.dmg

I could not find mention of this issue in recent commits to the `master` branch. _So, it's not clear to me if the difference between recent daily builds and the v0.21.0 stable builds is a code concern or a build/release signing (or app permissions) concern._

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

chennes, 2023-08-19:
Can you test with the unsigned version of the 0.21 release?

systemresearch, 2023-08-19:
> Can you test with the unsigned version of the 0.21 release?

yes. 

Now tested `FreeCAD-0.21.0-mac-intel_x86-unsigned.dmg` from [FreeCAD/releases/tag/0.21.0](https://github.com/FreeCAD/FreeCAD/releases/tag/0.21.0). 

The _`Addon manager` preview rendered OK_ for the x86_64 unsigned 0.21 release version on the following machines:

- x86_64 mini: macOS Ventura 13.5.1 (22G90) 
- M1 arm mini: macOS Ventura 13.4.1 (22F82) … using Rosetta2

Note: I first removed FreeCAD related files in `~/Library/Application Support/…`, `~/Library/Preferences/…` and `~/Library/Caches/…` for this unsigned version to be a clean install & launch test.

NSsweens, 2023-08-23:
Hi, this is exactly what I was searching for.  Having the same problem with Ventura 13.4 on M1 Macbook air.  Not getting the Read me files and if I try to install, I get the following message

Failed to install A2plus
Git returned a non-zero exit status: 1
Called with: /usr/bin/git clone --recurse-submodules https://github.com/kbwbe/A2plus /Users/dana_mba/Library/Application Support/FreeCAD/Mod/A2plus

Returned stderr:
xcrun: error: invalid active developer path (/Library/Developer/CommandLineTools), missing xcrun at: /Library/Developer/CommandLineTools/usr/bin/xcrun

Thanks

chennes, 2023-08-23:
@NSsweens that's a slightly different error, with a different solution: you can either go to the Addon Manager preferences and disable git, or you can install Apple's XCode command-line tools. The way that the Addon Manager detects git does not work correctly on the Mac, which always reports the existence of git, even when it hasn't really been installed.

NSsweens, 2023-08-25:
Thank you for the help. I decided to install xcode to keep the installation automatic.  Add on manager always seemed to work fine for me before the 0.21 upgrade, but I just installed the workbench I wanted without issue.  It still won't load the readme files, but that's fine.

Thanks again

chenshaw13, 2024-01-10:
running on mac book.   Sonoma 14.2.1 and cant seem to get pas the issue.   im not following the way to get around it either.  ????

chennes, 2024-01-26:
This issue was resolved by the removal of QtWebEngine from the Addon Manager in #12097 .

closed 2024-01-26 by chennes