FreeCAD/FreeCAD #10276: addon manager: cannot get any package README.md from github

tabddorf, 2023-08-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
[code]
OS: macOS 13.4.1
Word size of FreeCAD: 64-bit
Version: 0.21.0.33675 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.0)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
[/code]
```

### Subproject(s) affected?

None

### Problem description

I just installed freecad the first time on a mac M1 and wanted to install the sheet-metal addon. But I get the same error message for all addon packages:
Could not load README data from URL https://github.com......
What am I misssing ?
Do I need to install git also ?
is there a workaround to load an addon manually ? ( the link on the sheet-metal web page ofm this topic is also dead )

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tabddorf, 2023-08-19:
[image]

systemresearch, 2023-08-19:
The _Addon manager "Could not load README data from URL …"_ is the key part of [[Problem] v21.0 Addon Manager preview causes QtWebEngine crash on macOS Ventura 13 (regression?) #10274](https://github.com/FreeCAD/FreeCAD/issues/10274).

The  `FreeCAD-0.21.0-mac-intel_x86-unsigned.dmg` from [FreeCAD/releases/tag/0.21.0](https://github.com/FreeCAD/FreeCAD/releases/tag/0.21.0) was found to not have the _Addon manager "Could not load README data from URL …"_ problem.

chennes, 2023-08-19:
The git problem has to do with the way Apple provides (or doesn't provide) git -- you can manually deactivate git in the user preferences of the Addon Manager, that should resolve the failure to install.

luzpaz, 2023-08-19:
@chennes does this warrant a way to let user know via the error ?

michael-reed-hampton, 2023-10-02:
This problem persists for me in spite of the work around.
I have disabled git in the preferences for Addon Manager. The message occurs for every addon I try to install.  I shut down and restarted FreeCAD to verify.
My OS: macOS, Ventura 134.5.2
My FreeCAD info: 
[code]
OS: macOS 13.5.2
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
[/code]

I am unsure of what the specific issue is with git on MAC, but there is an installation of git on my machine, and FreeCAD does point to it in the preferences > Addon Manager > "Path to git executable (optional)".

Per https://forum.freecad.org/viewtopic.php?t=30573, I also checked "General issues > If the AddonManager complains of no connection to Github" (even though it did not complain).  The response indicated no problem with github.

A specific plugin that fails is A2Plus.  The response is: "Could not load README data from URL https://github.com/kbwbe/A2plus/blob/master/README.md".  I verified that the file can be read by my browser, and via 

[code]
import urllib.request
r = urllib.request.urlopen("https://github.com/kbwbe/A2plus/blob/master/README.md")
r.read()
[/code]

From the FreeCAD python console.

chennes, 2023-10-02:
Are you running the signed or the unsigned executable? (Another workaround for you to just get things working is to manually install the addon: just use git clone to clone it into your `/Users/XXX/Library/Application Support/FreeCAD/Mod` directory)

michael-reed-hampton, 2023-10-02:
@chennes - thank you.  I am using freecad installed via the mac `brew` package installer.
This was MY fault.  I focused on the error message, not realizing that the 'install' button was still enabled.  I was able install, I just ignored the error message.

Again - thank you.

alpdigitech, 2023-11-18:
It's probably related to rendering markdown rather than connection/security setting. "boltsfc" addon shows the README.md but in text format (github redirected it).

chennes, 2024-01-26:
I expect this to have been resolved by #12097 -- @tabddorf (or others) can you confirm?

closed 2024-02-26 by chennes