FreeCAD/FreeCAD #10284: [Problem] Helix negative cone angle reverses the coil

Anthony-Gaudino, 2023-08-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: (HEAD detached at 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.8, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * MeshRemodel 1.8919.0
  * Launcher
  * InventorLoader 1.4.0
  * Render 2023.2.6
  * Silk 0.1.3
  * Defeaturing 1.2.0
  * A2plus 0.4.60n
  * Manipulator 1.5.0
  * fasteners 0.4.59
  * lattice2 1.0.0
  * sheetmetal 0.2.63
  * toSketch 1.0.1
  * Curves 0.6.13
  * Assembly4 0.50.2
  * Assembly3 0.12.0
```

### Subproject(s) affected?

PartDesign

### Problem description

When creating an additive coil in PartDesign, If I set the cone angle to a negative number it reverses the coil.
I expected the diameter of the coil to decrease, but instead it reverses the coil and increases it's diameter.

Coil with positive cone angle value:
[image]

Coil with negative cone angle value:
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

grubertechnic, 2023-08-21:
Confirmed in the newest version.

```
OS: Debian GNU/Linux 12 (bookworm) (X-Cinnamon/lightdm-xsession)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.33917 (Git)
Build type: Release
Branch: master
Hash: 0c0a4473cd032945912768c2e5821b668a132199
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * QuickMeasure 2022.10.28
```

luzpaz, 2023-08-21:
Whomever take a look at this, do you mind also checking out #9377 ? TIA

bgbsww, 2023-11-07:
I'd take a swing at this, but I don't understand it.  The two images above look correct to me.  Is the expectation that the second image should be automatically offset by the height, or should it look different somehow?

Anthony-Gaudino, 2023-11-09:
> I'd take a swing at this, but I don't understand it. The two images above look correct to me. Is the expectation that the second image should be automatically offset by the height, or should it look different somehow?

The behavior as is is not correct. When a negative value is entered the whole helix is flipped.

When using the Part workbench helix it also provides an angle property that seems to be correct. Try inputting 10deg or -10deg for example.

closed 2023-11-27 by chennes

luzpaz, 2023-11-27:
Thanks so much for the fix @bgbsww. You rock!

luzpaz, 2023-11-29:
@Anthony-Gaudino please confirm fix when you get a chance, TIA

PR #11399 (2023-11-27): Correct PartDesign Helix feature negative angles
Fix #10284.