FreeCAD/FreeCAD #10285: [Problem]

jmz1, 2023-08-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
[code]
OS: macOS 13.5.1
Word size of FreeCAD: 64-bit
Version: 0.21.0.33675 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.0)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
[/code]
```

### Subproject(s) affected?

FEM

### Problem description

Error on opening the FEM workbench, from an empty (or any) document:

```
14:22:48  (qt.qpa.fonts) Populating font family aliases took 75 ms. Replace uses of missing font family "Courier" with one that exists to avoid this cost. 
14:23:02  dlopen(/Applications/FreeCAD.app/Contents/Resources/lib/Fem.so, 0x0002): symbol not found in flat namespace '___kmpc_critical'
14:23:02  Traceback (most recent call last):
  File "<string>", line 71, in Initialize
  File "/Applications/FreeCAD.app/Contents/Resources/lib/python3.10/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
```

The FEM toolbars do not load. This is on my M1 Max 2021 MBP. I would like to be able to use the Elmer integration.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

jmz1, 2023-08-20:
Same error on 0.22 dev build:
```
OS: macOS 13.5.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.33909 (Git)
Build type: Release
Branch: master
Hash: afd06d09f796f0069971fb1d4a1bf60536595f2c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
```

luzpaz, 2023-08-21:
@oliveroxtoby could you reproduce this on macOS by chance ?

oliveroxtoby, 2023-08-21:
> @oliveroxtoby could you reproduce this on macOS by chance ?

Sorry, I don't have a Mac.

chennes, 2023-08-21:
I cannot replicate on MacOS 13.3.1, ARM, using the 0.21.0 ARM release.

JohnOCFII, 2023-08-21:
Interesting.  I **did** see the same failure with the 0.21 release on my Arm Mac, but **did not** see the failure with my locally built version from yesterday.  I see the version of Python differs between the release and the version I built.

Details below:

Failure:

```
13:52:42  (qt.qpa.fonts) Populating font family aliases took 47 ms. Replace uses of missing font family "Courier" with one that exists to avoid this cost. 
13:52:55  dlopen(/Applications/Other Apps/FreeCAD.app/Contents/Resources/lib/Fem.so, 0x0002): symbol not found in flat namespace '___kmpc_critical'
13:52:55  Traceback (most recent call last):
  File "<string>", line 71, in Initialize
  File "/Applications/Other Apps/FreeCAD.app/Contents/Resources/lib/python3.10/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
```

```
OS: macOS 13.5.1
Word size of FreeCAD: 64-bit
Version: 0.21.0.33675 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.0)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
```

Success:

```
OS: macOS 13.5.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.33917 (Git)
Build type: Release
Branch: master
Hash: 0c0a4473cd032945912768c2e5821b668a132199
Python 3.11.0, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United States (en_US)
```

jmz1, 2023-08-23:
I'd be interested in an Arm 0.22 dev binary compiled with Python 3.11, if one is available!

LilithHafner, 2023-08-24:
First-time user of FreeCAD here. I just downloaded 0.21.0 from https://www.freecad.org/downloads.php and hit this issue immediately. M2 CPU, MacOS 13.3.

chennes, 2023-08-24:
For those of you experiencing this, can you try with the _unsigned_ release on the GitHub release page? The code signing has given a bit of difficulty in other areas. 
https://github.com/FreeCAD/FreeCAD/releases/download/0.21.0/FreeCAD-0.21.0-mac-intel_x86-unsigned.dmg

LilithHafner, 2023-08-25:
That fixed the crash and also fixed the "the [Start Workbench](https://wiki.freecad.org/Start_Workbench) shows only a blank page" issue.

jmz1, 2023-08-28:
> For those of you experiencing this, can you try with the _unsigned_ release on the GitHub release page? The code signing has given a bit of difficulty in other areas.
> https://github.com/FreeCAD/FreeCAD/releases/download/0.21.0/FreeCAD-0.21.0-mac-intel_x86-unsigned.dmg

Likewise, fixed it for me (but now running x86 🙃)

adrianinsaval, 2023-08-31:
try the arm unsigned bundle from here instead please: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/0.21.0

LilithHafner, 2023-08-31:
The unsigned arm (FreeCAD_0.21.0-2023-08-01-conda-macOS-arm64-py310.dmg) gave me a popup with 
```
dlopen(/private/var/folders/hc/fn82kz1j5vl8w7lwd4l079y80000gn/T/AppTranslocation/D804BB07-95D6-484A-AA34-B3E9DD68B853/d/FreeCAD.app/Contents/Resources/lib/Fem.so, 0x0002): symbol not found in flat namespace '___kmpc_critical
```

[image]

luzpaz, 2023-11-08:
Marked missing diagnosis

001101111011101101, 2024-01-03:
[code]
OS: macOS 14.2.1
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * Help 1.0.3
[/code]

Hi and Happy New Year,

the unsigned version 0.21.2.33771 has similar issues:
`dlopen(/Applications/FreeCAD.app/Contents/Resources/lib/Fem.so, 0x0002): symbol not found in flat namespace '___kmpc_critical'`

I'm running a M3 Max and am willing to help, test and provide feedback

luzpaz, 2024-03-15:
> I'd be interested in an Arm 0.22 dev binary compiled with Python 3.11, if one is available!

Is there one ?

krl91, 2024-05-31:
> [code] OS: macOS 14.2.1 Word size of FreeCAD: 64-bit Version: 0.21.2.33771 (Git) Build type: Release Branch: (HEAD detached at 0.21.2) Hash: [b9bfa5c](https://github.com/FreeCAD/FreeCAD/commit/b9bfa5c5507506e4515816414cd27f4851d00489) Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3 Locale: C/Default (C) Installed mods:
> 
>     * Help 1.0.3
>       [/code]
> 
> 
> Hi and Happy New Year,
> 
> the unsigned version 0.21.2.33771 has similar issues: `dlopen(/Applications/FreeCAD.app/Contents/Resources/lib/Fem.so, 0x0002): symbol not found in flat namespace '___kmpc_critical'`
> 
> I'm running a M3 Max and am willing to help, test and provide feedback

Same issue on my side on MacOS 14.5 and M3 Pro.

nmadani, 2024-06-14:
Also running into 
dlopen(/Applications/FreeCAD.app/Contents/Resources/lib/Fem.so, 0x0002): symbol not found in flat namespace kmpc_critical' 
when trying to start FEM. Macbook Pro M2 chip, macOS 14.5.

[code]
OS: macOS 14.5
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * fasteners 0.5.22
  * CubeMenu
  * Help 1.0.3
[/code]

nmadani, 2025-01-17:
Looks like it is fixed in 1.0.0 revision 39109.  Thank you to all FreeCad developers!

closed 2025-01-17 by luzpaz