FreeCAD/FreeCAD #10292: [Problem] There is no core assembly workbench.

PaddleStroke, 2023-08-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
0.21
```

### Subproject(s) affected?

Other (specify in description)

### Problem description

There is currently no assembly workbench.

This is paving the way to integrate an assembly workbench in FreeCAD.

So to solve this first issue we should start by creating an empty workbench called Assembly. This would of course be only a first step and is paving the way to integrate an assembly workbench in FreeCAD.

I suggest using the A4 assembly icon because it's for me the icon that looks the best and is the most consistent with the rest of FC (part/body icons).

@sliptonic 

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

chennes, 2023-08-21:
I'm a bit hesitant about using the A4 icon -- advanced users who actually *have* A4 installed and want to use it might be confused by that.

luzpaz, 2023-08-21:
Should this be a discuss thread or a ticket ?

PaddleStroke, 2023-08-21:
Or we change the colors of the blocks.
Let's use A4 icon with colors from A3 then. This way everyone is confused ;)
Joke aside that would be different enough.

Though I think even if we use the exact same icon as A4 it will be a
short-lived issue no?

PaddleStroke, 2023-08-22:
[image]
Simple color change.

> How so?

Nevermind that you are correct.

sliptonic, 2023-08-22:
> > Though I think even if we use the exact same icon as A4 it will be a short-lived issue no?
> 
> How so?

Let's see if we can get one of the icon guys involved early.  A whole new icon set designed from the beginning would be good.

onekk, 2023-08-22:
Better to find an icon totally different, confusion is high enough on FreeCAD, increasing it will be a "bad service".

Side note, there is some discussions place where the details of the "Official Assembly WB" will be discussed?

Regards

PaddleStroke, 2023-08-24:
It appears that the current A4 logo is actually a core icon dedicated for assemblies (Geoassembly.svg). And this is also the icon that the Std_Part object is using when its type is set to assembly. And it seems that we are going to use this Std_Part as the basic container for assembly, then it only makes sense that the core assembly workbench use this icon rather than a new one.

PaddleStroke, 2023-08-25:
@Zolko-123 please weight in about the icon. I think it makes more sense that the core assembly wb use the core icon. Especially since you advocated that we use the App::Part object with Assembly type that use exactly this icon.
As this comes with a conflict with your Assembly4 wb I think its best to ask your opinion on the matter.

Both wb cannot have different icons while both having the same icon as the App::Part type assembly container.

For me it's not really a problem that the integrated and A4 share the same icon. A4 is an addon so people using it are installing it themselves so they will know what this is about so I don't think it will be that confusing.

PaddleStroke, 2023-08-25:
@onekk discussion can take place in the github issues (this one for now). 
We are currently working on a basic wb that will be almost empty. From there we will build up by making issues and inviting people to discuss.

I will soon push a draft PR so that people can already weight in this offered solution.

Zolko-123, 2023-08-28:
> **Problem description**
> There is currently no assembly workbench.

This is false : there are currently at least 3 assembly workbenches for FreeCAD.

PaddleStroke, 2023-08-28:
Thank you for your great contribution. I modified this sentence to :

> Problem description
> There is currently no assembly workbench in core.

Zolko-123, 2023-08-29:
> **Problem description**
> There is currently no assembly workbench in core.

this is still false: workbenches are a core integral feature of FreeCAD, and there are 3 assembly workbenches available with the core tool _AddonManager_ :

[image]

Zolko-123, 2023-08-29:
> In an ideal world, core FreeCAD would contain all that is needed to open and correctly display assembly files. Then, you could choose your favorite editor, ASM2, 3 or 4 (or 5), to edit that assembly...
> Would that be something thinkable?

yes, this is also what I'm proposing in Issue #9278 : to ADD a new command _Std_Assembly_ to the structure toolbar, which creates the standard FreeCAD assembly container. This way, all assembly workbenches become automagically compatible.

PaddleStroke, 2023-08-29:
> Zolko has a point there, that addons are an inherent and fully functional
> part of FreeCAD, and if a functionality is provided by an addon, it should
> be considered fulfilled as much as if it was in "core". That's actually
> something we want to encourage and develop more.
>
> This could be a good moment to reflect on what exactly it is that we want
> in core, and why?
>
> For me this boils down to one point: Be able to open files that have been
> produced with an addon, on a machine where this addon is not installed,
> without errors and with a correct display. Then, if needed, the user should
> be informed of what addon they need to install to *edit* the file
> contents.
>
> That's basically what I'm aiming at achieving with BIM and Arch: Reduce
> Arch to just allow to correctly open a BIM file in FreeCAD, but if you want
> to edit, then you need to install the BIM addon.
>
> In an ideal world, core FreeCAD would contain all that is needed to open
> and correctly display assembly files. Then, you could choose your favorite
> editor, ASM2, 3 or 4 (or 5), to *edit* that assembly...
>
> Would that be something thinkable? That a kind of "assembly engine" is
> added to core, together with a series of basic object types, that could be
> used by other ASM workbenches?

No he doesn't.
Sorry but add-ons are not part of core by definition. They do not belong on
Freecad repository, they belong to individuals who can do whatever they
want with them.

Tomorrow RT decides to takes down A3, no more A3. He decides to replace all
icons by bunnies icons? Done.

Having assembly in core present many many obvious advantages.
- not owned by one person.
- development would be rigorous and qualitative because of the contributing
process
- Development efforts wouldn't be squandered on X différents wbs.
- For users having no default assembly means they need to learn about the 3
existing add-ons and their limitations to find the one that will use.

Would it be better if we had 3 Path workbenches? Or 4 Arch wb maybe? Yes
why not making draft and arch add-ons so they could be competitive
solutions? Like Arch2, Arch2+, Arch3, Arch4. That would be really helpful.
Tell me Yorik do you think that would be a good solution? All the
contributions you get for Arch would be split between all those instead.

And most important, imagine the pain for the user if for every workbench he
has to choose from n wbs that he has to install manually.

yorikvanhavre, 2023-08-29:
> Tell me Yorik do you think that would be a good solution? All the contributions you get for Arch would be split between all those instead.

I actually think Arch at least, should get out of core. I'm thinking of shrinking it to the bare minimum, and putting all the UI stuff into BIM. And yes, it would be awesome to have BIM2, BIM3... That is, you could still open a BIM model with no BIM WB installed (the minimal Arch provides support for it), but you need to install a BIM addon if you want to edit the file.

About Draft, I'm not sure because it's now itself used by many other WBs (arch, openscad,...) so it has become technically complicated now. But in theory, why not?

Now I get your point about integrated, community-supported development. Almost all addons are mostly a one-person job. People want to support and contribute to FreeCAD, it's complicated to ask them to support and contribute to a series of addons. Anything that will get into FreeCAD core is likely to get a huge boost in attention and development from the community, while addons hardly get many contributors. That's indeed an important point and something we must consider.

Probably indeed there is a difference of scale here. Things that are meant to be massively used and that would benefit from more community development should become core. But we want people to be able to use alternative solutions (addons) and addons developers to be first-class FreeCAD developers too, so we should always facilitate and be careful about "core solutions" to potentially "kill" addons.

I think the key is in defining common grounds here. Things that should be core but also be useful for addon developers. Things that promote compatibility between workbenches.

sliptonic, 2023-08-29:
> Sorry but add-ons are not part of core by definition. They do not belong on Freecad repository, they belong to individuals who can do whatever they want with them.

Don't confuse the architecture with the system/organization that produces it.
_3rd party_ addons are owned by one person but there's no reason the FreeCAD organization cannot have its own set of addons that are built and maintained under the contributing policy and developed to the same standards.

That said, the research we did earlier this year, starting at Fosdem and continuing through our long blog series was unambiguous.  Users want an integrated assembly system.  It doesn't have to be the most powerful and address every use-case but they want to build basic assemblies with the out-of-box experience.  Not only do they want it but they view FreeCAD as deficient without it because every other advanced CAD has an assembly capability integrated into the core experience.

tobiasfalk, 2023-08-29:
@yorikvanhavre it is completely normal for a software to look at the features that the Add-on/Modding community created and implement the most used or highly usable, and that is exactly what is happening here. 
This is how for example Windows Task manager started, it was not a Program created by Microsoft for Microsoft, but something someone developed by its own and then Microsoft included it in to its own source code.
Another example is Minecraft, here multiple features started as a Mod and the Mojang saw it and implemented it in to the base game, many of the ocean animal for example.

As for:
>I'm thinking of shrinking it to the bare minimum, and putting all the UI stuff into BIM. And yes, it would be awesome to have BIM2, BIM3... 

This is something a "Poweruser" may want, but FreeCAD is not and should not be for "Powerusers", at least in my opinion. Most users of FreeCAD want to just use FreeCAD without any add-ons and do not want to downland all add-ons just so that they can open a file without a problem. Imaging trying to open .txt file only to get the message "This file can not be opened because you do not have XYC"

Without realizing that someone already started the conquest of an implemented workbench, I have written a post on this topic: [https://forum.freecad.org/viewtopic.php?p=702787#p702787](url)
There I have explained a bit what benefits a dedicated Assembly workbench would have.

sliptonic, 2023-08-31:
Since PR #10427 is merged, we now have an integrated workbench.  It just doesn't do anything yet.  Let's close this issue and start opening new issues for missing functionality.

closed 2023-08-31 by sliptonic

LibrarySeph, 2023-11-28:
Just spent the day testing out A2Plus, Assembly 3, and Assembly 4.  My test case is attaching 8 cubes together into a bigger cube.   Kind of a lot of work for something that really comes down to "make these two faces touch, and also these two vertices"