FreeCAD/FreeCAD #10300: [Problem] Reverse Engineering Workbench has poor discoverability

sliptonic, 2023-08-21:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Full version info

[code]
OS: Linux Mint 20 (i3/i3)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.33942 +2 (Git)
Build type: Unknown
Branch: (HEAD detached from upstream/pr/10033)
Hash: 0772da42e31fef257934a36c87f613d83501e0b1
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.5.2
Locale: English/United States (en_US)
Installed mods: 
  * ProDarkThemePreferencePack 1.0.0
  * sheetmetal 0.2.63
  * Ondsel-Lens 2023.8.8.02
  * Assembly3 0.12.0
[/code]

### Subproject(s) affected?

None

### Problem description

The reverse engineering workbench is built by default and thus appears to the user in the selector.

When the user open the workbench, there is only a single tool on the toolbar.
A menu is added  with several submenus.  However, it is unclear how the user is supposed to use these tools or what the next steps should be.

This issue is created to discuss how this workbench can be improved.

Does the WB mostly deal with fitting surfaces to point clouds?  
Can the tools be added to other more task-oriented workbenches?
Does this need further development to be useful to the user?

If so, should the workbench be disabled by default?

### Anything else?

This issue was mentioned in a PR and is part of a larger project to clean up the workbench selector
https://github.com/FreeCAD/FreeCAD/pull/10168
https://github.com/orgs/FreeCAD/projects/16

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

PaddleStroke, 2023-08-23:
I suggest disabling it by default until it reaches an usable state and that documentation is done.

sliptonic, 2023-08-23:
> I suggest disabling it by default until it reaches an usable state and that documentation is done.

Do you mean disabling it in the preferences or disabling the build in cmake?

PaddleStroke, 2023-08-23:
I meant in the preferences. If some people need these tools it's best not to remove it completely I guess

Roy-043, 2023-08-29:
IMO this issue is not well defined. The main problem is that there are in fact several issues being discussed.
1. State of development of the WB and how to improve it.
2. Lack of documentation.
3. Should the WB be presented to the user in its current state.

Does this conform to the new protocols? Or should this be closed and 3 new issues be created instead?

The proposed solution only addresses one of the issues raised. It certainly does not improve the 'discoverability' of the WB, in fact it achieves the opposite.

FlachyJoe, 2023-08-30:
> Can we make it clearer

In order of difficulty:

- [ ] Add an icon in the toolbar for each menu entry
- [ ] Write appropriate tooltips
- [ ] Write wiki page for each tool

Plus:
what about including Mesh tools as Arch includes Draft ones?

maxwxyz, 2024-03-02:
When looking at the commands the RE WB offers, these would also make sense in the Mesh WB. Would this be an option to merge the RE into the Mesh WB?

FlachyJoe, 2024-03-02:
@maxwxyz the mesh tools could be offer in a RE toolbar. RE users **always** use Mesh but the contrary is false.

luzpaz, 2024-03-18:
@wwmayer would you weigh-in regarding this issue ?

closed 2025-02-07 by github-actions[bot]