FreeCAD/FreeCAD #10306: [Problem] System::solve_DL(SubSystem* subsys, bool isRedundantsolving)  error

YggSky, 2023-08-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
all
```

### Subproject(s) affected?

GCS

### Problem description

file path   FreeCAD/src/Mod/Sketcher/App/planegcs /GCS.cpp

double beta = 0;
                Eigen::VectorXd b = h_gn - h_sd;
                double bb = (b.transpose()*b).norm();
                double gb = (h_sd.transpose()*b).norm();
                double c = (delta + h_sd.norm())*(delta - h_sd.norm());

                **if (gb > 0)
                    beta = c / (gb + sqrt(gb * gb + c * bb));
                else
                    beta = (sqrt(gb * gb + c * bb) - gb)/bb;**

                // and update h_dl and dL with beta
                h_dl = h_sd + beta*b;

beta  Calculate is error  .the dogleg formula was used wrong. Quadratic equation 
about beta   solving error.

### Anything else?

Numerical Optimization

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-08-25:
@AjinkyaDahale can you take a look here ?

AjinkyaDahale, 2023-08-25:
@YggSky please add some literature if you can.

YggSky, 2023-08-26:
> please add some literature if you can.
the book 《Numerical Optimization》 (Jorge Nocedal Stephen J. Wright)
chapter 4  -- Trust-Region Methods
about the dogleg method Detail explain

YggSky, 2023-09-14:
beta = (h_gn.transpose()* b + sqrt(gb * gb + c * bb)) / c;

luzpaz, 2023-09-15:
@AjinkyaDahale :point_up_2:

AjinkyaDahale, 2023-09-15:
@YggSky thanks for the formula and sorry for the delay.

@luzpaz thanks for the ping. I had planned to take a look at this this weekend but I suppose I should have said it here as well.

I'm currently caught up with some other stuff (the expression engine) so the time I can allocate to this is limited.

YggSky, 2024-05-07:
> @YggSky
> 
> Hi there!
> 
> I did not write the DogLeg implementation myself.
> 
> I do have the paper version of the book "Numerical optimization". I cannot find it, but I found an online version in pdf.
> 
> I am having problems mapping the code to the book and then understanding what you mean. You may need to help me out here.
> 
> Here we mean: h_gn = step of Gauss-Newton h_sd = step of steppest-descent
> 
> So here, we start calculating the dogleg step:
> 
> https://github.com/FreeCAD/FreeCAD/blob/826194e0bc71c6d1c2d775b2e708a285bb2aa22b/src/Mod/Sketcher/App/planegcs/GCS.cpp#L2472
> 
> First case is when Gauss-Newton step is within the trust region, then the dogleg step for the iteration is updated with the Gauss-Newton step.
> 
> Second case is when (the Gauss-Newton step is outside the trust region) AND the steepest descent step is also outside the trust region.
> 
> Third case is when (the Gauss-Newton step is outside the trust region) BUT the steepest descent step is inside the trust region. This is where the error may be located (the else):
> 
> https://github.com/FreeCAD/FreeCAD/blob/826194e0bc71c6d1c2d775b2e708a285bb2aa22b/src/Mod/Sketcher/App/planegcs/GCS.cpp#L2483
> 
> So we are trying to solve pg. 68, eq 4.1 reformulated as pg. 69, eq. 4.5, by applying eq. 4.15 (pg. 74). So basically, we approximate the optimal optimization curve of figure 4.4 via two lines (pu of eq. 4.14 and pb of eq. 4.15).
> 
> From your post, I understand that you mean that this should be the calculation of beta.
> 
> ```
> beta = (h_gn.transpose()* b + sqrt(gb * gb + c * bb)) / c;
> ```
> 
> instead of the one marked with double asterisk below:
> 
> ```
> if (gb > 0)
>                     beta = c / (gb + sqrt(gb * gb + c * bb));
>                 else
>                     beta = (sqrt(gb * gb + c * bb) - gb)/bb;**
> ```
> 
> **Is this what you mean?**
> 
> **Could you help me out bridge what I think I have figured out above and your solution? (and letting me know if I said something that is wrong)**

sorry，I am wrong。

YggSky, 2024-05-07:
@abdullahtahiriyo    beta = c / (gb + sqrt(gb * gb + c * bb));  This formula is a simplification of the molecules. The solution of quadratic equation,in order to reduce the error.   so formula is right , with higher precision compare to no simplification molecules.

YggSky, 2024-05-09:
@abdullahtahiriyo  by the way. i find h_gn stand formula is [https://wikimedia.org/api/rest_v1/media/math/render/svg/41515fb47a56a0575efa229ef07c41f8ff2ee21e](url) why code is that?

YggSky, 2024-05-09:
[](https://wikimedia.org/api/rest_v1/media/math/render/svg/41515fb47a56a0575efa229ef07c41f8ff2ee21e)

YggSky, 2024-05-09:
https://wikimedia.org/api/rest_v1/media/math/render/svg/41515fb47a56a0575efa229ef07c41f8ff2ee21e

AjinkyaDahale, 2024-06-03:
> @abdullahtahiriyo by the way.

Unfortunately, Abdullah has currently left FC development.

> h_gn stand formula is (https://wikimedia.org/api/rest_v1/media/math/render/svg/41515fb47a56a0575efa229ef07c41f8ff2ee21e) why code h_gn is that?...

> Neither feels right. but it is truely effctive

Could you elaborate on what you mean by "truly effective"?

closed 2025-06-18 by github-actions[bot]