FreeCAD/FreeCAD #10308: [Problem] Selection order priority

PaddleStroke, 2023-08-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
0.22 dev
```

### Subproject(s) affected?

None

### Problem description

Recently we have a new tool which appeared called 'selection filter'. This also exist in Assembly4 as a toolbar. (introduced in core following https://github.com/FreeCAD/FreeCAD/issues/10243)

The underlying problem is that it is hard/impossible to select some points or lines. And this problem is not fixed by the selection filter.

The technical reason for that (I think) is that the selection mechanism use pointOnRay function, but it uses the overload that returns only the 1st object found. So if you try to select a point but a face is closer to the camera, then the face is going to be selected.

The selection mechanism should not behave like this. Instead it should use the pointOnRay that return ALL the objects found. Then these objects should be sorted by priority by the following order : 
- Points
- Edges
- Faces
Then the selection should take the item with the higher priority.

With this proper behavior the 'selection filter' would no longer be needed.

@wwmayer Do you have any insight on the matter? Or can confirm if my analysis is correct?

@Syres916 @FEA-eng you might be interested to follow this.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2023-08-22:
I would definitely appreciate further enhancement in selection but I strongly believe that filters are needed anyway. They are available in pretty much all CAD software (and CAE too) for a reason. However, they could be improved to e.g. influence the selection via a bounding box so that only a particular type of geometric entity is picked (very useful for fillets, FEM and so on). Another improvement would be that once the filter is active and you click close to a different type of entity (e.g. vertex instead of edge), the closest entity of the correct type is picked instead of showing a wrong selection symbol. That could be very helpful and it works like this in other software.

Then there are more advanced possibilities like:
- selection by angle
- inverting selection
- selection of inside faces
- some form of selection order

It would be also great to make geometric entities easier to select (bigger radius, selection order suggested by you, maybe something else) but control over the type of entities that you want to select at the moment (filters) is also needed. After all, when applying fillets you may want to pick only faces or only edges, for instance.

PaddleStroke, 2023-08-22:
Got it. I edited the issue then to not say the tool will no longer be needed then

FEA-eng, 2023-08-22:
Thanks. I agree that the default selection should be enhanced. As I said on the forum, it might be mostly a matter of too small pick radius (you have to be ultra-precise to pick a vertex or an edge instead of a face) but your selection order priority idea may also help resolve this issue.

wwmayer, 2023-08-22:
> The selection mechanism should not behave like this. Instead it should use the pointOnRay that return ALL the objects found. Then these objects should be sorted by priority by the following order :

SoFCUnifiedSelection provides the method getPickedList that basically does what you want. Therefore in uses SoFCUnifiedSelection::getPriority that sets the priority: point, line, face

adrianinsaval, 2023-08-22:
> * Points
> * Edges
> * Faces

I would add datum/origin planes as lowest priority in this list

additionally it would be beneficial to have a way to cycle through the list of objects under the cursor to be able to select things bellow an object.

maxwxyz, 2023-10-25:
+1 for the priority 
and also maybe to long click, right click or Alt/Option click to get a drop-down to choose between all features in a defined radius under the cursor, maybe even when hidden by other geometry or on the backside (if feasible).

maxwxyz, 2023-10-27:
@adrianinsaval I've come across _Pick Geometry_ in RT link daily:

Uploading pick-geometry-RT.mp4…

Praveen-Rai, 2023-10-31:
Can we consider a spherical region, around the cursor point as the bounding box for selection ?

Getting precisely on a point, edge or curve is often a challenge. 

[image]

In above scenario, the expectation would be to list, two edges, vertex & face. Right now it's highlighting only the face even when cursor is fraction of mm away from the edges. Its very tiring to get to exact point over an wireframe entity.
One quick change ( assuming it's a small effort ) to help in precise selection would be to replace windows cursor with Cross-hair cursor in the graphics area. The leaning cursor creates problems in precise positioning.

Kudos to @Syres916 for brining filters. Lack of which was making it tiring to work with FreeCAD. It's something that everyone would want right now. Also, hoping these filters shall work with Rectangle Selection. 

A big thanks to you all for all your efforts :-)

maxwxyz, 2024-03-02:
Comments from the FreeCAD Day 2024 Complaint Session:
I find sometimes it's too hard to select the vertex or a line. The selection area is so tiny or like really aiming for it and then.
First of all, that's configurable. But second of all, yes. So the default selection radius is wrong. It is far too small.

wwmayer, 2024-03-02:
> So the default selection radius is wrong. It is far too small.

This hasn't much to do with the selection radius. Even if you change the settings to make it considerably bigger you won't see a big improvement in most cases.

The actual issue has been explained [here](https://github.com/FreeCAD/FreeCAD/issues/9205#issuecomment-1696961176) and [here](https://github.com/FreeCAD/FreeCAD/issues/9205#issuecomment-1723236170)

If you have an example where loosing the tolerance isn't enough then please share it.

SpeeQz1, 2024-12-13:
I saw this and wanted to +1 this suggestion, I've used it in some other programs and even CAD ones and it's a boon to have.