FreeCAD/FreeCAD #10319: [Problem] Crash opening a particular STL file

MethylBromide, 2023-08-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
[code]
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.21.0.33668 +7 (Git)
Build type: Release
Branch: (HEAD detached at 0.21)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Help 1.0.3
[/code]
```

### Subproject(s) affected?

None

### Problem description

When opening a particular STL file, FreeCAD abruptly exits. There are no messages.
I searched open issues for the keywords **crash stl** and no matches.
[Cool Duup.zip](https://github.com/FreeCAD/FreeCAD/files/12410470/Cool.Duup.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

MethylBromide, 2023-08-22:
The log file ends with these lines:

Err: Cannot mix incompatible Qt library (5.15.1) with this library (5.15.2)
Err: Abnormal program termination...

Roy-043, 2023-08-23:
No problem here (exact same version as in the initial report):
```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.21.0.33668 +7 (Git)
Build type: Release
Branch: (HEAD detached at 0.21)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Dutch/Netherlands (nl_NL)
Installed mods:
  * Help 1.0.3
```

GeneGH, 2023-08-23:
Qt 5.15.1 appears to come from the Addon Manager (only).

MethylBromide, 2023-08-23:
I uninstalled the program, reinstalled 0.21.0 from the official download page, and ran it. The first thing I did was try to open that file, Bam, gone. Apparently there's something different with my installation compared to yours. Perhaps because you got to where you are by installing upgrades as opposed to a fresh install from nothing. Perhaps you have different versions of some addons. Is there some underlying software that might be different on my system? I don't understand your programming environment well enough to know what to make of the log file message, but I do think whatever is going on, if it's going to fail it should at least fail in a less catastrophic way, perhaps with some sort of helpful error dialog that would explain how to fix the problem. At present, I don't know how to fix it manually so I can get my work done, Yes my work involves a rubber ducky -- roll with it.

Syres916, 2023-08-23:
@MethylBromide you don't mention anything about your GPU, do you have more than one and if so, have you tried changing the specific one used by FreeCAD to the other available as per https://pureinfotech.com/set-default-gpu-apps-windows-11/

Also you were politely asked to submit the FreeCAD.Log but you only quoted the last two lines, the reason we ask for the  whole log is there maybe a line that isn't an error but experienced users can immedateily highlight as pointer to what the problem maybe therefore could you please attach the whole file.

MethylBromide, 2023-08-24:
[FreeCAD.log](https://github.com/FreeCAD/FreeCAD/files/12424294/FreeCAD.log)
The "Graphics Preference" dialog when I add FreeCad as an application, doesn't list multiple GPUs, only "Intel(R) UHD Graphics". I have two monitors attached to a notebook PC via a USB hub, and I have been trying to use FreeCad on one of those monitors -- never on the built-in display.

Syres916, 2023-08-24:
I said the log would be useful, in your Windows Path is `C:\Program Files\Calibre2` which is a Qt based product and I'm fairly certain it uses _Qt5.15.1_ and this is being picked up before the _Qt5.15.2_ in the FreeCAD\bin folder hence the crash. What you need to determine is whether removing `C:\Program Files\Calibre2` from the path actually breaks Calibre, personally I doubt it but if it does then you'll need a batch file for one of the products to change the path before executing the app and then resetting the path back after the app closes.

MethylBromide, 2023-08-24:
[FreeCAD.log](https://github.com/FreeCAD/FreeCAD/files/12430747/FreeCAD.log)
Okay. I changed the path as you suggested and am still experiencing the same failure. Here's the new log file (with Calibre no longer in the path). Perhaps there is some other program causing a similar conflict, but it seems to me that requiring everything in the path to be using the same Qt version as this application requires, makes your application fragile. Is there not a way to either read it from the program's own directory rather than scanning the path, or failing that, at least specify that a particular version is required and please ignore any Qt of an earlier version?

luzpaz, 2023-08-25:
> Is there not a way to either read it from the program's own directory rather than scanning the path, or failing that, at least specify that a particular version is required and please ignore any Qt of an earlier version?

Is this a contingency FreeCAD should honor ?

MethylBromide, 2023-08-25:
> Is this a contingency FreeCAD should honor ?
You're asking whether it should be able to coexist with other software completely unrelated to 3D design, as opposed to what? Requiring a dedicated computer to run FreeCAD on, containing only programs from a small approved list? How is this even a question?

luzpaz, 2023-09-09:
Yes, as you frame it, FreeCAD should be able to coexist with other (read: different version) Qt run software on on ones machine.

grubertechnic, 2023-09-10:
Since it is a pure 3rd qt party issue, i would close it. Nothing to do here.

MethylBromide, 2023-09-10:
> Since it is a pure 3rd qt party issue, i would close it. Nothing to do here.

Can you describe in a little more detail the analysis that leads you to believe the FreeCAD team is helpless to improve on matters here? It seems to me that some code somewhere is selecting the wrong Qt library to load. Whose code is making that selection? Not the code in the Qt library, because that code isn't loaded yet. I thought it would be the code in FreeCAD locating the library.

Somehow other software seems to avoid this dilemma. Calibre clearly uses the same library, and since removing Calibre from the path doesn't solve the problem, so does something else on my computer. Those programs never interfered with each other, despite their developers probably not being in lock-step about which Qt versions they were using. So I think they are loading in the right way, and you are loading in the wrong way.

EVEN IF this is somehow impossible to solve, handling the error and putting up an explanatory message is surely the minimum requirement.

tummechatronics, 2023-09-14:
Just for confirmation, I could open your file and couldn't replicate the behavior.  (Fresh installation)

[code]
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.21.1.33668 +26 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * Assembly4 0.12.3
  * fcgear 1.0.0
[/code]

MethylBromide, 2023-09-15:
We were already aware this issue involves interaction with the system environment settings and other software installed on the computer, so it's not a surprise that a random fresh installation would not have this problem. I was able to resolve it for myself by adding the path "C:\Program Files\FreeCAD 0.21\Ext\PySide" into the PATH system environment variable before other entries already in there. I don't yet know whether this might cause problems with other software that depends on having a different version of Qt.

I still maintain that the program should just work and not require the user to manually adjust the system settings to prevent it from crashing. Failing that, it at least needs to exit gracefully and informatively so the end user will _know_ how to address the problem.

luzpaz, 2023-09-16:
> I still maintain that the program should just work and not require the user to manually adjust the system settings to prevent it from crashing. Failing that, it at least needs to exit gracefully and informatively so the end user will know how to address the problem.

This is a good point. Is there a fail-safe we can have to double check we're using the correct Qt libs ?

wwmayer, 2023-09-24:
> Err: Cannot mix incompatible Qt library (5.15.1) with this library (5.15.2)

What's weird is that Qt 5.15.1 and Qt 5.15.2 are incompatible because Qt always claims that if major and minor versions are equal then the binaries are compatible.

> I was able to resolve it for myself by adding the path "C:\Program Files\FreeCAD 0.21\Ext\PySide" into the PATH system environment variable before other entries already in there.

Does the PySide directory also contain Qt DLLs (like Qt5Core.dll, ...)? This is weird and it should be avoided by making sure that PySide uses the exact same Qt DLLs as FreeCAD.

> Can you describe in a little more detail the analysis that leads you to believe the FreeCAD team is helpless to improve on matters here?

FreeCAD perfectly co-exists with other SW that uses a different Qt version as long as this SW puts its DLLs into its own installation directory.

But too often I have seen that 3rd party SW puts its DLLs into C:\Windows or C:\Windows\system32 which is an absolute no-go. If this is the case then any other application that uses a different version has almost no chance to avoid a crash.

> Failing that, it at least needs to exit gracefully and informatively so the end user will know how to address the problem.

This is more or less impossible. If an application loads an incompatible DLL then it will be in an undefined state that usually causes a crash.

xtemp09, 2023-09-24:
> Err: Cannot mix incompatible Qt library (5.15.1) with this library (5.15.2)

The message comes from [this place](https://github.com/qt/qtbase/blob/36a169e31e86fadde41c61804ce437164977f60a/src/corelib/kernel/qobject_p.h#L231).

Incompatibility can come from different UCRT or MSVCRT runtime library; it can be built with different compilers, like, _MSVC_, _clang_ or _mingw-w64_; some plugins cannot be loaded or require certain Qt version.

Errors also can come from different libraries involved. I experienced that when mixed Qt libraries compiled with the same compiler (gcc 12.1), but with different settings. Where exactly the problem's origin lies, I don't know, but, I suppose it came from different 3<sup>rd</sup> party libraries, like _libiconv_, _ICU_ or _OpenSSL_.

----

> This is more or less impossible. If an application loads an incompatible DLL then it will be in an undefined state that usually causes a crash.

I would point out that [qt.conf](https://doc.qt.io/qt-6/qt-conf.html) may alleviate the problem; a little launching application or at least a cmd file like this can help

```cmd
set PATH=%~dp0\libraries;%PATH%
start "" "bin\FreeCAD.exe"
```

thus, cleaning _bin/_ directory from all the DLLs and exes files. I think, _bin/_ ought to contain only `freecad.exe` and `freecadcmd.exe`, while other files should be somewhere in _lib/_.

Or FreeCAD _/_ directory should contain just `freecad.exe` and `freecadcmd.exe` and the other stuff is in _bin/_.

Having a launcher is beneficial, besides checking the environment, it can also log, restart after crash.

wwmayer, 2024-04-27:
In the past the STL reader used Qt's concurrent framework (which for some reason then loads a wrong DLL into memory).
With 053d0fa047 I have replaced this now with C++ threads because the Qt way requires a running event loop. This is undesired behaviour when using FreeCAD in command line mode.

So, I guess this change should also fix the crash.

maxwxyz, 2024-07-01:
@chennes please test with Qt6 build

luzpaz, 2024-09-07:
@MethylBromide can you rerproduce on latest RC1 ?

closed 2025-06-21 by github-actions[bot]