FreeCAD/FreeCAD #10325: [Problem] Carbon Copy repeated multiple times on drag

regularfry, 2023-08-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.21.0.33668 +7 (Git)
Build type: Release
Branch: (HEAD detached at 0.21)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * 3DfindIT 1.2.0
  * Assembly4 0.12.5
  * Curves 0.6.6
  * freecad.gears 1.0.0
  * lattice2 1.0.0
  * Manipulator 1.5.0
  * Pyramids-and-Polyhedrons
  * sheetmetal 0.2.61
```

### Subproject(s) affected?

Sketcher

### Problem description

With the Carbon Copy tool selected, accidentally dragging the mouse while selecting another sketch in the Combo View results in the elements of that sketch being imported multiple times.

To replicate:

1. Select File -> New
2. Switch to Sketcher workbench
3. Create a sketch, on the XY plane, and add a rectangle (any size, doesn't matter). Close the sketch. This will be called `Sketch`.
4. Deselect `Sketch` if it's selected, and create another one on the XY plane. This will be called `Sketch001`.
5. Select `Sketch` -> Sketcher geometries -> Create carbon copy
6. Switch to the Model tab in the Combo View
7. Now: mouse down on the `Sketch` name in the Combo View, and without releasing the mouse button, move the pointer horizontally. It doesn't need to be by much, just enough to register a mouse move event. Release the mouse button.
8. Switch back to the Task tab in the Combo View, and have a look at how many lines have been created.

When I do that, I get some random multiple of 4 lines created in `Sketch001`.

This is, for some reason, very easy to trigger with my hardware.  Discussion (such as it is) [here](https://forum.freecad.org/viewtopic.php?t=80446&start=10).

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2023-08-23:
For me there's two issues, firstly if the user followed the instructions in the https://wiki.freecad.org/Sketcher_CarbonCopy then there wouldn't be a problem but more importantly the code should be able to handle the scenario.

It looks to me (complete guess) as there's been some copy/paste of code gone on in the past. The comments https://github.com/FreeCAD/FreeCAD/blob/master/src/Mod/Sketcher/Gui/DrawSketchHandlerCarbonCopy.h#L133 and https://github.com/FreeCAD/FreeCAD/blob/master/src/Mod/Sketcher/Gui/DrawSketchHandlerCarbonCopy.h#L163 mention `ContinuousCreationMode` but that parameter isn't being checked for (correctly IMHO) as _CarbonCopy_ isn't a command that I would relate to _Continuous Creation Mode_, it's a one-stop shop in my experience.

Maybe @PLChris could give a view on this.

Therefore I would use:

`        sketchgui->purgeHandler();`

in place of both of those comments.

Syres916, 2023-09-05:
@abdullahtahiriyo As you are now in the vicinity, welcome back by the way. I'd like your thoughts particularly around the above comment regarding ContinuousCreationMode and CarbonCopy. As a user they don't sit well with each other, even if it does work but the fix for the OP's report does look a bit too easy.

maxwxyz, 2024-07-06:
is this still an issue  in the latest weekly?

regularfry, 2024-07-08:
Yes, confirmed with the above instructions with this version:

```
OS: macOS 14.3.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38043 (Git)
Build type: Release
Branch: main
Hash: 561e521817034ea7660de987d8ab7183ad583595
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * freecad.gears 1.2.0
  * lattice2 1.0.0
  * Curves 0.6.21
```

FEA-eng, 2025-09-14:
Strange. Still relevant.

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/09 06:43:51
Build type: Release
Branch: main
Hash: c1dd116e1e3304511d9e5e634e711e549b2b84e7
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```

marbocub, 2026-04-05:
I have narrowed the possible solutions down to two approaches. Both are implemented and being tested.

## A: eventFilter approach (non-intrusive, cross-layer)

An event filter is installed on the "Model" TreeWidget to filter out MouseMove events. This prevents the TreeWidget from entering drag-selection mode.

**Pros**
- Non-intrusive: preserves existing behavior and user experience.
- Targets only the repetition caused by drag-selection.

**Cons**
- Introduces a cross-layer interaction (handler → GUI TreeWidget).

## B: copiedList approach (self-contained, behavior change)

A list is used to track sketches that have already been processed by CarbonCopy. If the same sketch is selected again, repeated execution is skipped.

**Pros**
- Fully self-contained within DrawSketchHandler.

**Cons**
- Changes existing behavior, with a small impact on user experience.

marbocub, 2026-10-17:
Still an issue — a fix is already up for review: #29104

closed 2026-11-03 by chennes