FreeCAD/FreeCAD #10326: [Problem] Addon Manager detection of git does not work on a Mac

chennes, 2023-08-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
[code]
Version: 0.21.0.33668 +7 (Git)
Build type: Release
Branch: (HEAD detached at 0.21)
[/code]
```

### Subproject(s) affected?

None

### Problem description

On Macs, an executable called "git" always exists, even if the developer tools have not been installed: this proto-git simply instructs the user to install XCode Tools. So a better way of detecting whether git is available is needed.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-09-05:
How are other projects dealing with this ?

chennes, 2023-10-12:
Good find, @luzpaz !! Anyone willing to put together a PR? Probably it should go someplace around here:
https://github.com/FreeCAD/FreeCAD/blob/4f1fcbd03b703fb76886c1bbfec172167ea48aef/src/Mod/AddonManager/addonmanager_git.py#L312

luzpaz, 2023-10-12:
Here's how spyder did it: https://github.com/spyder-ide/spyder/commit/ca3de2d7892e284a2dc31ef556949d6ef004072e

RohanMishra315, 2023-12-28:
Hey @chennes @luzpaz  Can you check the updated code for these , and provide suggestion.

    def _find_git(self):
        # Find git in preference order
        # A) The value of the GitExecutable user preference
        # B) The executable located in the same directory as FreeCAD and called "git"
        # C) The result of a shutil search for your system's "git" executable

        prefs = fci.ParamGet("User parameter:BaseApp/Preferences/Addons")
        git_exe = prefs.GetString("GitExecutable", "Not set")

        # A) Use the value of the GitExecutable user preference if set and valid
        if git_exe and os.path.exists(git_exe):
            self.git_exe = git_exe
            return

        # B) Look for the Git executable in the same directory as FreeCAD
        fc_dir = fci.DataPaths().home_dir
        git_exe = os.path.join(fc_dir, "bin", "git")
        if "Windows" in platform.system():
            git_exe += ".exe"

        if os.path.exists(git_exe):
            self.git_exe = git_exe
            prefs.SetString("GitExecutable", git_exe)
            return

        # C) Use shutil.which to search for the Git executable in the system's PATH
        git_exe = shutil.which("git")
        if git_exe and os.path.exists(git_exe):
            # Check if Xcode Command Line Tools are installed
            if self._is_xcode_command_line_tools_installed():
                self.git_exe = git_exe
                prefs.SetString("GitExecutable", git_exe)
                return

        # If Git executable is not found, you may want to handle this case accordingly.
        # For example, raise an exception or log a warning.
        print("Git executable not found or Xcode Command Line Tools not installed.")

    def _is_xcode_command_line_tools_installed(self):
        try:
            subprocess.check_output(["xcode-select", "-p"])
            return True
        except subprocess.CalledProcessError:
            return False

chennes, 2023-12-31:
Of course, that check should only be done on the Mac 😄

RohanMishra315, 2023-12-31:
Oh ! I forgot that , thanks for reminding .I'll update it😅

PR #12129 (2024-01-27): Addon Manager: Fix git detection on mac
Fixes #10326
closed 2024-01-27 by chennes