FreeCAD/FreeCAD #10334: [Problem] [PartDesign] [SubObject Binder] Binder edges are invisible in sketch, if overlapping with Axis

V5Studio, 2023-08-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
[code]
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.21.0.33675 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.0)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * 3DfindIT 1.2.0
  * freecad.gears 1.0.0
  * parts_library
[/code]
```

### Subproject(s) affected?

PartDesign

### Problem description

Created a simple 3 edge binder

[image]

Moving to Sketch, the Axis seems to overlap the edge making it impossible to see

[image]

### Anything else?

[Binder Issue.zip](https://github.com/FreeCAD/FreeCAD/files/12422097/Binder.Issue.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2023-08-23:
Yes, this is the rendering order, while a fix is done, there is a workaround by @0penBrain which removes the axes just in the current Sketcher Edit Mode (they are redrawn after leaving and re-entering Edit Mode), see https://forum.freecad.org/viewtopic.php?style=3&p=502114#p502114 to generate a macro which you can add to either the Sketcher or Global toolbar for example.

V5Studio, 2023-08-23:
Thanks @Syres916 for a quick solution. I am curious though, shouldn't the axes always be the first to be drawn in the render order?

Syres916, 2023-08-23:
>  I am curious though, shouldn't the axes always be the first to be drawn in the render order?

They are the first to be drawn in a new sketch but if you draw a Normal Geometric line on top of one and then draw a construction line on top of that then there needs to be am order top-down. At present objects outside the sketch have the lowest priority, then the axes and then the user can select the order or Normal, Construction and External Geometry.

V5Studio, 2023-08-23:
Thanks for enlightening me.. I guess the axes should be considered special case and should always be drawn first!

FEA-eng, 2025-09-14:
Still relevant

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/09 06:43:51
Build type: Release
Branch: main
Hash: c1dd116e1e3304511d9e5e634e711e549b2b84e7
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```

FEA-eng, 2026-10-25:
Still relevant

```
OS: Windows 11 build 26300
Architecture: x86_64
Version: 27.1.0dev.20261020 (Git shallow)
Build date: 2026/10/20 22:15:52
Build type: Release
Branch: (HEAD detached at 9946038aa)
Hash: 9946038aace97af288eb28f2727a169f9d1158c7
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 2.0.1, IfcOpenShell 0.8.5, OCC 8.0.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/113.268/1.25
OpenGL version/vendor/renderer: 4.6.0 - Build 27.20.100.9664/Intel/Intel(R) UHD Graphics 620
```