FreeCAD/FreeCAD #10336: Helix Extra Offset error in v0.21

Stevecosoft, 2023-08-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
OS: Ubuntu 22.04.3 LTS (Trinity/trinity)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33675 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.0)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * CurvedShapes
  * LCInterlocking
  * Help 1.0.3
  * Manipulator
```

### Subproject(s) affected?

Path

### Problem description

I'm getting an error when attempting to use the Extra Offset setting in a Helix operation (to generate a roughing toolpath).  An example error message: 'hole_radius - inner_radius = 7.920000000000001 is < tool diameter of 12.7'

To reproduce this, in the attached file, edit the Helix operation and change Extra Offset from 0 to of 0.01mm.  The hole diameter is 15.88mm and the tool diameter is 12.7mm.  (There is also a second operation, Pocket_Shape_ExtraOffset_1.0mm, which is me proving to myself that the tool will fit inside the desired hole with Extra Offset = 1.0mm.)

I did some experimentation, and found that the hole diameter needs to be much larger to avoid the error.  If the hole diameter = 26.0mm, Extra Offset of 0.01mm works.  Extra Offset of 0.15mm works, 0.16mm fails.

The full error:
```
Traceback (most recent call last):
  File "/tmp/.mount_FreeCASFOdr3/usr/Mod/Path/PathScripts/PathUtils.py", line 70, in new_function
    res = function(*args, **kwargs)
  File "/tmp/.mount_FreeCASFOdr3/usr/Mod/Path/Path/Op/Base.py", line 817, in execute
    result = self.opExecute(obj)
  File "/tmp/.mount_FreeCASFOdr3/usr/Mod/Path/Path/Op/CircularHoleBase.py", line 194, in opExecute
    self.circularHoleExecute(obj, holes)
  File "/tmp/.mount_FreeCASFOdr3/usr/Mod/Path/Path/Op/Helix.py", line 221, in circularHoleExecute
    results = helix.generate(**args)
  File "/tmp/.mount_FreeCASFOdr3/usr/Mod/Path/Path/Base/Generator/helix.py", line 87, in generate
    raise ValueError(
<class 'ValueError'>: hole_radius - inner_radius = 7.920000000000001 is < tool diameter of 12.7
Recompute failed!
``` 
[HelixOffsetTest_1.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/12425268/HelixOffsetTest_1.FCStd.zip)

### Anything else?

My forum post: [Helix Extra Offset error in v0.21](https://forum.freecad.org/viewtopic.php?t=80599)

A related post with a much larger set of changes: [trouble with helix path](https://forum.freecad.org/viewtopic.php?p=673606)

For this small issue of the Extra Offset error, it would be great to have a fix in v0.21.1 (assuming there is one).

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Stevecosoft, 2023-08-24:
I searched the existing issues but not well enough...  This is probably a duplicate of [https://github.com/FreeCAD/FreeCAD/issues/10157](url).

Roy-043, 2023-08-24:
Thanks. Closing this then.

closed 2023-08-24 by Roy-043