FreeCAD/FreeCAD #10338: [Problem] Std_Part is confusing.

PaddleStroke, 2023-08-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
0.22 dev
```

### Subproject(s) affected?

Core

### Problem description

This has started to be discussed in this issue : https://github.com/FreeCAD/FreeCAD/issues/9278

The problem is that the Std_Part is confusing for a lot of users. Many people are confused between this and part design body. Besides the name does not carry out the meaning of the object. If we look at the wiki https://wiki.freecad.org/Std_Part
 : 

> The Std Part element was developed to be the basic building block to create mechanical [assemblies](https://wiki.freecad.org/Assembly).

So how did we end up with the confusing name 'Part' that makes no sense?

@Zolko-123 mentioned 2 possible usecases : 

> well, I see 2 use-cases, but I'm not sure how relevant they are:
> 
> 1 - parts using booleans, where you design 2 shapes and subtract (or add or intersect) them together: in this case, you can use a Std_Part container for the final result, which is not an assembly but a complex yet standalone part
> 2 - for surface functions, where you design 3D surfaces and cut and intersect them, producing a Std_Part which is a general purpose 3D container

But I don't think those make enough sense to validate keeping the name Part. For case 1 I fail to see how you need a container for booleans as they already results in single tree entities. For case 2, it could be called an assembly of surfaces rather than a Part it would make more sense.

Anyone see reason why we shouldn't rename it?

I also suggest changing the icon so that the big block and small block do not share the same color. To further differentiate with PartDesign body and show in a clearer fashion the fact that it's an assembly.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Zolko-123, 2023-08-24:
> The problem is that the Std_Part is confusing for a lot of users. Many people are confused between this and part design body

in what occasions does a user see the name "_Std_Part_ " ? And what has that to do with the confusion between _Part_ and _PartDesign_ WBs ? 

> Anyone see reason why we shouldn't rename it ?

rename _what_ to _what_ ?

the `TypeId` of a _Std_Part_ is `App::Part`, so if you rename _Std_Part_ to _Std_Assembly_ and leave the `TypeId` at `App::Part`, you won't have solved **ANY** confusion but would introduce documentation nightmare. And if you plan to change the `TypeId` of _Std_Part_ then you brake many many existing tools ....

PaddleStroke, 2023-08-24:
Let me make it clearer for you as it seems you are confused here. What is confusing is the name 'Part' of Std_Part. Indeed the string 'Std_Part' does not appear in many places for the user and is not so important.

Here I did not discuss the confusion between the Part wb and the PartDesign wb. I agree the Part wb should be renamed but this is another topic of discussion.

So my suggestion is to rename the 'Part' object 'Assembly'. Changing the name 'Std_Part' to 'Std_Assembly' would also make sense. Replacing the TypeId to App::Assembly would also make sense to keep things coherent, but if this is too difficult to do then we can keep App::Part.

Zolko-123, 2023-08-24:
> Here I did not discuss the confusion between the Part wb and the PartDesign wb. I agree the Part wb should be renamed but this is another topic of discussion.

no it's **NOT** another topic: either "Part" is confusing then this confusion should be treated **EVERYWHERE** where it's confusing, or if changing a single command is enough then there is no actual confusion. You can't have it both ways

Zolko-123, 2023-08-24:
> So my suggestion is to rename the 'Part' object 'Assembly'

did you try :

```
App.ActiveDocument.addObject('App::Part','Assembly')
or
App.ActiveDocument.addObject('App::Part','TarataTsoinTsoin')
```

?

PaddleStroke, 2023-08-24:
Yes I know this mechanism. It changes the icon. However this cute mechanism adds one layer of complexity for nothing. 'Part' object doesn't makes sense, it's very clearly stated in the wiki that it's purposed for assembly.

PaddleStroke, 2023-08-24:
List of sources that shows that 'Part' name is confusing :

- https://forum.freecad.org/viewtopic.php?t=43501#:~:text=Body%20is%20a%20concept%20of,created%20with%20the%20Part%20Workbench.
- https://www.reddit.com/r/FreeCAD/comments/ohx5o3/part_vs_body/
- https://www.reddit.com/r/FreeCAD/comments/g0jocq/part_vs_body/
- https://dlamaet.dalmateysspot.nl/freecad-part-vs-body
- There are even videos about it ... https://www.youtube.com/watch?v=iWkPe8wYoZo&ab_channel=jpwillm

PaddleStroke, 2023-08-24:
Closed in favor of https://github.com/FreeCAD/FreeCAD/issues/9748 that is actually the root problem imo.

Zolko-123, 2023-08-24:
I think that the main problem is the "Structure" toolbar that allows to create various structural objects. Structural as in : these objects don't have any real geometry but are helper construction building-block objects: Part, Body, Group, Sketch, LCS ... Unfortunately, that toolbar is defined in C++ and not Python.

So the real solution would be to make the "Structure" toolbar 1) in Python and 2) make there functions that create those objects.

sasobadovinac, 2023-08-24:
If we can agree that we use the existing Part container for both Parts and Assemblies, as it is already used in Asm4 and comparing to commercial cad it is like this in NX and if I remember also in Fusion (possible also some others), then as a first step my suggestion would be to make existing functionality of the Part container to change its icon automatic, that is, if a Part is placed inside of a Part then the parent Part automatically becomes and assembly, I see no reasons to debate around that, if this is what we decide then this is simply how it has to work, a Part container with other Part containers inside of it is automatically an Assembly, and if you look at NX and Fusion you will see the same, the icon automatically changes. Then we can discuses further, probably how to solve the Body issue

sliptonic, 2023-08-24:
I'm reading a couple things that are beginning to converge on a proposed solution.
1) @sasobadovinac suggestion to automate the icon changing behavior based on whether it contains children.
2) @PaddleStroke suggestion to fix the useless tooltips to be less useless
3) @Zolko-123 suggestion to move the structure toolbar to python.  To be honest, I'm not clear what problem this solves but I'm always a fan of moving user-facing code to python.

Did I miss anything?  If not, please apply the "solution proposed" label to the issue.

grubertechnic, 2023-08-26:
> Many people are confused between this 'Part' and partDesign 'Body'.

This issue is literally a duplicate of #9748 and discussed already to the end, dozen times, open an issue again is not appropriate, write something on the exiting discussions (really easy to find) and not start one again and again.

At least in the CAD industry, a difference between Part and Body is common (also logical, a body is per se no part), but not on all platforms. But as @adrianinsaval  mentioned on this issue, the PartDesign Body can be renamed.

sasobadovinac, 2023-08-27:
> This issue is literally a duplicate of #9748...

No it is not, as you said it yourself Part and Body are two different things, this issue is about how to further develop the Part and the other issue is about how to fix the Body

Zolko-123, 2023-08-29:
> @Zolko-123 suggestion to move the structure toolbar to python. To be honest, I'm not clear what problem this solves

The current Structure toolbar has only 4 entries: 

- Part
- Group
- Link
- LinkGroup

[image]

It doesn't create datum objects, especially it doesn't allow to create datum objects in Part containers. If it was written in Python we could have the following objects:

- Assembly
- Part
- Body
- Group
- Link
- Sketch
- LCS
- Datum plane
- Datum axis
- Datum point

which are the basic building-blocks (structures) of any FreeCAD model. Coding that in Python is much easier than in C++, and there is no speed advantage for these actions.

sliptonic, 2023-08-29:
Ok.  That makes sense.  It leaves out link/linkGroup.   Where should they go?

PaddleStroke, 2024-07-02:
I'll close this since the addition of a proper tooltip to the Part tool made things clearer.

closed 2024-07-02 by PaddleStroke

krushia, 2024-07-05:
Today I learned that "Part Boolean cannot be applied to Parts".

PaddleStroke, 2024-07-05:
That's the Part workbench name the issue. It should be called something else. CSG or PartDesign2 or something else than just Part.