FreeCAD/FreeCAD #10340: TechDraw WB Section issue

supernaturaljoe, 2023-08-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.0.33668 +7 (Git)
Build type: Release
Branch: (HEAD detached at 0.21)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Bulgarian/Bulgaria (bg_BG)
Installed mods: 
  * A2plus 2021.12.0
  * ArchTextures
  * Assembly4 0.50.2
  * BIM 1.0.0
  * dodo 2021.12.0
  * ExplodedAssembly
  * Help 1.0.3
  * parts_library
[/code]
```

### Subproject(s) affected?

Techdraw

### Problem description

I'm making a section view on a kitchen assembly and during that I gives me different section view than the arrow section location (screenshot attached)
Zip file attached. File to open: ASSM_Kitchen_Section_Issue. Page "изглед А" with the section issue
[image]

### Anything else?

[kitchen.zip](https://github.com/FreeCAD/FreeCAD/files/12426947/kitchen.zip)
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

WandererFan, 2023-08-24:
Although this https://forum.freecad.org/viewtopic.php?t=80362 talks about ComplexSection, it also applies to simple sections. 

The result with the current v022 code: 
[image]

WandererFan, 2023-08-24:
Duplicate of #10212?

supernaturaljoe, 2023-08-25:
@WandererFan I installed latest v.0.21 yesterday. I have done section of another section before but never had issues till yesterday. Maybe it's a file bug or something.

supernaturaljoe, 2023-08-25:
Thanks for the response. I'll use the original view to make sections.

closed 2023-08-25 by supernaturaljoe