FreeCAD/FreeCAD #10341: [Suggested feature] Provide command shortcuts in the search bar provided in the help menu

raindropsfromsky, 2023-08-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
[code]
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.33798 (Git)
Build type: Release
Branch: master
Hash: 614c01c0b100244417644456b1f925fe665d9f11
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/India (en_IN)
Installed mods: 
  * 3DfindIT 1.2.0
  * A2plus 0.4.60n
  * ArchTextures
  * Assembly3 0.12.0
  * Assembly4 0.50.2
  * btl 0.9.5
  * CurvedShapes 1.0.4
  * Curves 0.6.13
  * ExplodedAssembly
  * fasteners 0.4.59
  * freecad.gears 1.0.0
  * Glass (Disabled)
  * Help 1.0.3
  * lattice2 1.0.0
  * Manipulator 1.5.0
  * SearchBar 1.0.2
  * sheetmetal 0.2.63
[/code]
```

### Subproject(s) affected?

None

### Problem description

There is a search bar in the **Help** menu. It shows all commands located in the menu system, which is a very useful feature.
[image]

While this listing shows the toolbar icons, it does _not_ show the keyboard shortcuts for the commands. 

As a result, the user cannot learn to launch that command with a shortcut. He remains dependent on the search bar.
So every time he has to go through the same tedious multistep process of "type some characters to find the command". 

It would be great if the listing shows the keyboard shortcuts also. That way, the user can become gradually more and more efficient, by learning new shortcuts and then developing muscle memory for them.

Secondly, please allow the user to launch the command by clicking on the entry itself, just like what the **Launcher** addon does. 

### Anything else?

Actually, my first impulse was to request for shortcuts in the Launcher addon. But I came to know from the [FC Forum](https://forum.freecad.org/viewtopic.php?p=701588&sid=6c908b626041488a9a828ab05aa7fe60#p701588) that the **Launcher** addon is no longer maintained. 

That's why I thought of putting up this request here. As such, the search bar within FC is ideally suited to provide this functionality.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2023-08-24:
Firstly you must have installed the SearchBar as it's an external Addon and not part of the core development. Therefore could you please close this and open a Feature Request with the SearchBar author at https://github.com/SuzanneSoy/SearchBar/issues

raindropsfromsky, 2023-08-24:
oops!

Sorry for the inconvenience!

closed 2023-08-24 by raindropsfromsky