FreeCAD/FreeCAD #10345: [BUG] FemMesh.cpp compile errors

BlindDuck, 2023-08-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
remote: Enumerating objects: 7518, done.
remote: Counting objects: 100% (6084/6084), done.
remote: Compressing objects: 100% (404/404), done.
remote: Total 7518 (delta 5696), reused 6051 (delta 5680), pack-reused 1434
Receiving objects: 100% (7518/7518), 41.97 MiB | 11.01 MiB/s, done.
Resolving deltas: 100% (6503/6503), completed with 1314 local objects.
From https://github.com/FreeCAD/FreeCAD
   2559d7e7b6..53a4fb14c3  master                     -> origin/master
 * [new branch]            CONTRIBUTING-PROJECT-ADMIN -> origin/CONTRIBUTING-PROJECT-ADMIN
 * [new branch]            chennes-patch-1            -> origin/chennes-patch-1
 * [new branch]            releases/FreeCAD-0-21      -> origin/releases/FreeCAD-0-21
 * [new tag]               0.21.0                     -> 0.21.0
 * [new tag]               0.21rc1                    -> 0.21rc1
 * [new tag]               0.21rc1a                   -> 0.21rc1a
Updating 2559d7e7b6..53a4fb14c3
Fast-forward
```

### Subproject(s) affected?

FEM

### Problem description

error during compilation:

[ 42%] Copying /usr/src/packages/SOURCES/FreeCAD-source/src/Mod/Fem/femtools/tokrules.py to /opt/FreeCAD/Mod/Fem/femtools/tokrules.py
[ 42%] Built target FemScriptsTarget
Scanning dependencies of target Fem_autogen
[ 42%] Automatic MOC and UIC for target Fem
[ 42%] Built target Fem_autogen
Scanning dependencies of target Fem
[ 42%] Building CXX object src/Mod/Fem/App/CMakeFiles/Fem.dir/Fem_autogen/mocs_compilation.cpp.o
[ 42%] Building CXX object src/Mod/Fem/App/CMakeFiles/Fem.dir/FemMeshObject.cpp.o
[ 42%] Building CXX object src/Mod/Fem/App/CMakeFiles/Fem.dir/FemMeshShapeObject.cpp.o
[ 42%] Building CXX object src/Mod/Fem/App/CMakeFiles/Fem.dir/FemMeshShapeNetgenObject.cpp.o
[ 42%] Building CXX object src/Mod/Fem/App/CMakeFiles/Fem.dir/FemAnalysis.cpp.o
[ 42%] Building CXX object src/Mod/Fem/App/CMakeFiles/Fem.dir/FemMesh.cpp.o
/usr/src/packages/SOURCES/FreeCAD-source/src/Mod/Fem/App/FemMesh.cpp: In member function ‘std::set<int> Fem::FemMesh::getNodesBySolid(const TopoDS_Solid&) const’:
/usr/src/packages/SOURCES/FreeCAD-source/src/Mod/Fem/App/FemMesh.cpp:893:21: error: expected ‘=’ before ‘:’ token
     for (auto aNode : nodes) {
                     ^
/usr/src/packages/SOURCES/FreeCAD-source/src/Mod/Fem/App/FemMesh.cpp:911:21: error: expected ‘;’ before end of line
 #pragma omp critical
                     ^
/usr/src/packages/SOURCES/FreeCAD-source/src/Mod/Fem/App/FemMesh.cpp:911:21: error: expected primary-expression before end of line
/usr/src/packages/SOURCES/FreeCAD-source/src/Mod/Fem/App/FemMesh.cpp:911:21: error: expected ‘;’ before end of line
/usr/src/packages/SOURCES/FreeCAD-source/src/Mod/Fem/App/FemMesh.cpp:911:21: error: expected primary-expression before end of line
/usr/src/packages/SOURCES/FreeCAD-source/src/Mod/Fem/App/FemMesh.cpp:911:21: error: expected ‘)’ before end of line
/usr/src/packages/SOURCES/FreeCAD-source/src/Mod/Fem/App/FemMesh.cpp:911:21: error: expected primary-expression before end of line
/usr/src/packages/SOURCES/FreeCAD-source/src/Mod/Fem/App/FemMesh.cpp:893:5: error: invalid type for iteration variable ‘aNode’
     for (auto aNode : nodes) {
     ^~~
/usr/src/packages/SOURCES/FreeCAD-source/src/Mod/Fem/App/FemMesh.cpp:893:15: warning: unused variable ‘aNode’ [-Wunused-variable]
     for (auto aNode : nodes) {
               ^~~~~
/usr/src/packages/SOURCES/FreeCAD-source/src/Mod/Fem/App/FemMesh.cpp:911:21: error: expected ‘}’ before end of line
 #pragma omp critical
                     ^
/usr/src/packages/SOURCES/FreeCAD-source/src/Mod/Fem/App/FemMesh.cpp:911:21: warning: no return statement in function returning non-void [-Wreturn-type]
/usr/src/packages/SOURCES/FreeCAD-source/src/Mod/Fem/App/FemMesh.cpp: At global scope:
/usr/src/packages/SOURCES/FreeCAD-source/src/Mod/Fem/App/FemMesh.cpp:911:21: error: expected declaration before end of line
gmake[2]: *** [src/Mod/Fem/App/CMakeFiles/Fem.dir/build.make:163: src/Mod/Fem/App/CMakeFiles/Fem.dir/FemMesh.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:3772: src/Mod/Fem/App/CMakeFiles/Fem.dir/all] Error 2
gmake: *** [Makefile:147: all] Error 2

### Anything else?

[FreeCAD_20230824.txt](https://github.com/FreeCAD/FreeCAD/files/12430727/FreeCAD_20230824.txt)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

grubertechnic, 2023-08-24:
That is a little bit less information for a compile issue. What OS? What cmake flags? What installed dependencies? Compiler? , .......

At least we know from the terminal output, that you seem to use the master branch.

xtemp09, 2023-08-24:
The logs applied say that you are trying to compile using _GNU 7.5.0_. FreeCAD requires _gcc 9.3_ or newer.

grubertechnic, 2023-08-24:
@xtemp09 : did you really go through the textfile? I refuse read through sloppy pasted textfile, without any effort, because it is disrespectful. Anyway, thanks.

@luzpaz : then do you consider to close it, since it is an user error?

xtemp09, 2023-08-24:
@grubertechnic, yes, I did. Actually, you can just scroll down and in the end there is summary.

xtemp09, 2023-08-24:
@BlindDuck, my tries to compile FreeCAD for openSUSE were futile. The problem, alas, came from python/pyside/shiboken. Their combination just cannot be used. I think, their compilation from source may shed light onto this. The compilation guide for openSUSErs, published on the FreeCAD forum, is outdated and cannot be used for newer openSUSEs.

I concluded that the best way is to use FreeCAD supplied by repository or to download _AppImage_ or _Flatpak_.

BlindDuck, 2023-08-24:
@xtemp09 
I have compieled FreeCAD several times before on openSUSE Leap 15.3 with
-- Compiler:                    /usr/bin/c++ (7.5.0)

cmake --prefix=/opt/FreeCAD/ /usr/src/packages/SOURCES/FreeCAD-source -DBUILD_QT5=ON -DPYTHON_EXECUTABLE=/usr/bin/python3 -DCMAKE_BUILD_TYPE=Release

However with the newer versions from git it got difficult to compile without aborting.

Due to the fact that I am only a user the supplied information may not be the ones disiered.

Thank you for your answer.

BlindDuck, 2023-08-24:
> The logs applied say that you are trying to compile using _GNU 7.5.0_. FreeCAD requires _gcc 9.3_ or newer.

As long as the configuration checks are OK I also did not bother:
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc - works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ - works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Compiler: GNU, version: 7.5.0

luzpaz, 2023-08-25:
For convenience, please always try to the dependency report: 

```
   ==============
     System
   ==============

-- bindir:                      bin
-- BLAS:                        -undefined-
-- CMAKE_VERSION:               3.17.0
-- Compiler:                    /usr/bin/c++ (7.5.0)
-- datadir:                     share
-- docdir:                      share/doc/FreeCAD
-- includedir:                  include
-- libdir:                      lib64
-- prefix:                      /usr/local
-- Python:                      3.9.15 [/usr/bin/python3] Suffix: [.cpython-36m-x86_64-linux-gnu]

   ==============
     Config
   ==============

-- BUILD_FEM:                   ON
-- BUILD_GUI:                   ON
-- BUILD_SMESH:                 ON
-- BUILD_START:                 ON
-- BUILD_TECHDRAW:              ON
-- BUILD_VR:                    OFF 
-- BUILD_WEB:                   ON
-- CMAKE_BUILD_TYPE:            Release
-- CMAKE_CXX_FLAGS:             -Wall -Wextra -Wno-write-strings 
-- CMAKE_CXX_STANDARD:          17
-- CMAKE_INSTALL_PREFIX:        /usr/local
-- CMAKE_PREFIX_PATH:           -undefined-
-- ENABLE_DEVELOPER_TESTS:      ON
-- FREECAD_CREATE_MAC_APP:      -undefined-
-- FREECAD_LIBPACK_USE:         OFF 
-- FREECAD_QT_VERSION:          Auto
-- FREECAD_USE_EXTERNAL_KDL:    OFF 
-- FREECAD_USE_EXTERNAL_SMESH:  OFF 
-- FREECAD_USE_FREETYPE:        ON
-- FREECAD_USE_PYBIND11:        OFF 
-- PYTHON_EXECUTABLE:           /usr/bin/python3
-- PYTHON_LIBRARY:              /usr/lib64/libpython3.6m.so
-- USE_CUDA:                    -undefined-
-- USE_OPENCV:                  -undefined-

   ==============
     Libraries
   ==============

-- Boost:                       1.75.0
-- Coin3D:                      3.1.3 [/usr/lib64/libCoin.so] [/usr/include]
-- Coin3D_DOC:                  not found
-- DesignerPlugin:              not built (BUILD_DESIGNER_PLUGIN is OFF)
-- Doxygen:                     1.8.14 Language: English
-- Eigen3:                      3.3.4
-- fmt:                         Sources downloaded to /opt/FreeCAD/_deps/fmt-src
-- Freetype:                    2.10.4
-- HDF5:                        1.10.7
-- Matplotlib:                  2.2.5 PathDirs: /usr/lib64/python3.6/site-packages/matplotlib
-- MEDFile:                     3.2.0 [/usr/lib64/libmedC.so;/usr/lib64/libmed.so] [/usr/include/med]
-- NETGEN:                      not enabled
-- OCC:                         7.3.0 [/usr/lib] [/usr/include/opencascade]
-- OCC_Libs:                    [TKFillet;TKMesh;TKernel;TKG2d;TKG3d;TKMath;TKIGES;TKSTL;TKShHealing;TKXSBase;TKBool;TKBO;TKBRep;TKTopAlgo;TKGeomAlgo;TKGeomBase;TKOffset;TKPrim;TKSTEPBase;TKSTEPAttr;TKSTEP209;TKSTEP;TKHLR;TKFeat]
-- OpenGL_Lib:                  [/usr/lib64/libGL.so]
-- OpenGLU_Incl:                [/usr/include]
-- OpenGLU_Lib:                 [/usr/lib64/libGLU.so]
-- pivy:                        -undefined-
-- pybind11:                    not enabled
-- PYCXX:                       7.1.7 Incl: /usr/src/packages/SOURCES/FreeCAD-source/src Src:/usr/src/packages/SOURCES/FreeCAD-source/src/CXX
-- PySide:                      5.12.3 [/usr/include/PySide2]
-- PySideTools:                 v: 2  uic: [/usr/bin/pyside2-uic]  rcc: [/usr/bin/pyside2-rcc]
-- QtConcurrent:                5.12.7
-- QtCore:                      5.12.7
-- QtNetwork:                   5.12.7
-- QtOpenGL:                    5.12.7
-- QtPrintSupport:              5.12.7
-- QtSvg:                       5.12.7
-- QtTest:                      5.12.7
-- QtUiTools:                   5.12.7
-- QtWebEngineWidgets:          5.15.3
-- QtWidgets:                   5.12.7
-- QtXml:                       5.12.7
-- Rift:                        not enabled (BUILD_VR)
-- Shiboken:                    5.12.3 [/usr/include/shiboken2;/usr/include/python3.6m]
-- SMESH:                       7.7.1.0 build internal
-- SPNAV:                       [/usr/lib64/libspnav.so] [/usr/include]
-- SWIG:                        3.0.12
-- VTK:                         8.2.0
-- XercesC:                     3.2.3 [/usr/lib64/libxerces-c.so] [/usr/include]
-- ZLIB:                        1.2.11

=================================================
Now run 'cmake --build /opt/FreeCAD' to build FreeCAD
=================================================

-- Configuring done
-- Generating done
-- Build files have been written to: /opt/FreeCAD
```

xtemp09, 2023-08-25:
@BlindDuck, yes, you have compiled older versions. Support for Ubuntu 18.04 LTS ended this year, therefore, the requirements have been raised.

@luzpaz, _cmake_ is supposed to report that the compiler version is insufficient. What developer/maintainer fixes _cmake_?

BlindDuck, 2023-08-25:
@luzpaz:
Aye, I'll skip the git +/- lines, then the dependency report is not in line 3110.

I've installed gcc12 in addtion to gcc7, however it is not detected.
Just use 12 instead of 7 does not seem to bee so easy.

luzpaz, 2023-08-25:
@xtemp09 i'm confused though, why is the cmake also allowing OCC 7.3.0 version to pass IIRC the minimum was raised. I'm wondering if @BlindDuck is pulling current FreeCAD master HEAD ? 

> What developer/maintainer fixes cmake?

To answer your question, there are a few folk but lets not bug them just yet. Lets see if we can solve this ourselves, if not, we can ping them. 

https://github.com/FreeCAD/FreeCAD/blob/602841c5f0f2c9ce332f9d7c8df7153ac73454ba/cMake/FindOCC.cmake#L164-L172

Weird, so cmake doesn't fail when finding an OCC version less than 7.5.0  ?

BlindDuck, 2023-08-25:
@luzpaz I have the same lines of code at the end of the file:
`
  if(NOT OCC_VERSION_STRING VERSION_LESS 7.5.0)
    list(APPEND OCC_OCAF_LIBRARIES TKRWMesh)
  endif(NOT OCC_VERSION_STRING VERSION_LESS 7.5.0)
  message(STATUS "-- Found OCE/OpenCASCADE version: ${OCC_VERSION_STRING}")
  message(STATUS "-- OCE/OpenCASCADE include directory: ${OCC_INCLUDE_DIR}")
  message(STATUS "-- OCE/OpenCASCADE shared libraries directory: ${OCC_LIBRARY_DIR}")
else(OCC_FOUND)
  #message(SEND_ERROR "Neither OpenCASCADE Community Edition nor OpenCasCade were found: will not build CAD modules!")
endif(OCC_FOUND)
`

The content of ".git/FETCH_HEAD" is:
`602841c5f0f2c9ce332f9d7c8df7153ac73454ba                branch 'master' of https://github.com/FreeCAD/FreeCAD
8dc6c67fc02a215ef839bb27ddbee757d7e3b80f        not-for-merge   branch 'CONTRIBUTING-PROJECT-ADMIN' of https://github.com/FreeCAD/FreeCAD
f631be100e01b80878ccad17d7ff42d0f1b08afb        not-for-merge   branch 'chennes-patch-1' of https://github.com/FreeCAD/FreeCAD
140236c7747aecaedd7e199a05ee81c1081378d9        not-for-merge   branch 'development/toponaming' of https://github.com/FreeCAD/FreeCAD
5d0ff3556a2baf6312b2e55b13399021a6551cc2        not-for-merge   branch 'releases/FreeCAD-0-13' of https://github.com/FreeCAD/FreeCAD
f25e6e4716fb63ef3ac618ce9e552761bbc1b4b1        not-for-merge   branch 'releases/FreeCAD-0-14' of https://github.com/FreeCAD/FreeCAD
244b3aef360841646cbfe80a1b225c8b39c8380c        not-for-merge   branch 'releases/FreeCAD-0-15' of https://github.com/FreeCAD/FreeCAD
da2d364457257a7a8c6fb2137cea12c45becd71a        not-for-merge   branch 'releases/FreeCAD-0-16' of https://github.com/FreeCAD/FreeCAD
e1db7bff8ff36439373eb9a426567d8afe289c4d        not-for-merge   branch 'releases/FreeCAD-0-17' of https://github.com/FreeCAD/FreeCAD
b7b014ce43da53d0405177e3ea6b31e4a94ab392        not-for-merge   branch 'releases/FreeCAD-0-18' of https://github.com/FreeCAD/FreeCAD
0a49fd05a46447aba70b6f18b5bb8210175dd471        not-for-merge   branch 'releases/FreeCAD-0-19' of https://github.com/FreeCAD/FreeCAD
fcad80494f8937a98be6b2d8fc45c54c930b7817        not-for-merge   branch 'releases/FreeCAD-0-20' of https://github.com/FreeCAD/FreeCAD
a7bf4f33f864abec29463f233d090cc63d5a47bd        not-for-merge   branch 'releases/FreeCAD-0-21' of https://github.com/FreeCAD/FreeCAD
`

luzpaz, 2023-08-25:
OK, @BlindDuck is using the latest HEAD. But so many 3rd party dependencies on their system are  inferior versions.  The issue here is why CMake is not complaining about such things.

BlindDuck, 2023-08-28:
@luzpaz
gcc 7.5.0 was used, not 7.3.0!
Everthig fine, expect the raised requirements for gcc 9.3 xtemp09 mentioned while only 7.5.0 is in the file. 

OS: openSUSE Leap 15.3
My installed gcc versions are now:
  7.5.0
  9.3.1
10.4.0
11.3.0
12.2.1

and changed via:
$ export CC=/usr/bin/gcc-12
$ export CXX=/usr/bin/g++-12
$ update-alternatives --install /usr/bin/gcc gcc /usr/bin/gcc-12 80

7 (of course) and 12 (supprisingly) did not compile either:
[FreeCAD_20230827_7.txt](https://github.com/FreeCAD/FreeCAD/files/12454926/FreeCAD_20230827_7.txt)
[FreeCAD_20230827_9.txt](https://github.com/FreeCAD/FreeCAD/files/12454927/FreeCAD_20230827_9.txt)
[FreeCAD_20230827_10.txt](https://github.com/FreeCAD/FreeCAD/files/12454928/FreeCAD_20230827_10.txt)
[FreeCAD_20230827_11.txt](https://github.com/FreeCAD/FreeCAD/files/12454929/FreeCAD_20230827_11.txt)
[FreeCAD_20230827_12.txt](https://github.com/FreeCAD/FreeCAD/files/12454930/FreeCAD_20230827_12.txt)

luzpaz, 2023-11-04:
@marioalexis84 maybe you can take a look at this at some point ?

marioalexis84, 2023-11-04:
The error reported in file FreeCAD_20230827_12.txt says that the error occurs in `src/Mod/TechDrawXMLQuery.h`
Can you try including the `<functional>` header in that file?
Change:
```
#include <Mod/TechDraw/TechDrawGlobal.h>
#include <QList>
```
To:
```
#include <functional>
#include <Mod/TechDraw/TechDrawGlobal.h>
#include <QList>
```

luzpaz, 2023-11-05:
@BlindDuck :point_up_2:

BlindDuck, 2023-11-13:
Sorry for the delay, did not have access to this PC.
> The file FreeCAD_20230827_12.txt says that the error occurs in `src/Mod/TechDrawXMLQuery.h` Can you try including the `<functional>` header in that file? Change:

@marioalexis84 
I do not have this file in my git pull!
[image]

How shall I proceed?

luzpaz, 2023-11-13:
@BlindDuck are you on the master branch or main ?

BlindDuck, 2023-11-13:
master, from my first post:
`remote: Enumerating objects: 7518, done.
remote: Counting objects: 100% (6084/6084), done.
remote: Compressing objects: 100% (404/404), done.
remote: Total 7518 (delta 5696), reused 6051 (delta 5680), pack-reused 1434
Receiving objects: 100% (7518/7518), 41.97 MiB | 11.01 MiB/s, done.
Resolving deltas: 100% (6503/6503), completed with 1314 local objects.
From https://github.com/FreeCAD/FreeCAD
   2559d7e7b6..53a4fb14c3  master                     -> origin/master
 * [new branch]            CONTRIBUTING-PROJECT-ADMIN -> origin/CONTRIBUTING-PROJECT-ADMIN
 * [new branch]            chennes-patch-1            -> origin/chennes-patch-1
 * [new branch]            releases/FreeCAD-0-21      -> origin/releases/FreeCAD-0-21
 * [new tag]               0.21.0                     -> 0.21.0
 * [new tag]               0.21rc1                    -> 0.21rc1
 * [new tag]               0.21rc1a                   -> 0.21rc1a
Updating 2559d7e7b6..53a4fb14c3
Fast-forward`

luzpaz, 2023-11-13:
We switched to 'main' from 'master' just recently. Pull from main instead

BlindDuck, 2023-11-14:
$ git switch main
branch 'main' set up to track 'origin/main'.
Switched to a new branch 'main'
/usr/src/packages/SOURCES/FreeCAD-source # date; git pull
Tue Nov 14 10:29:59 CET 2023
remote: Enumerating objects: 9, done.
remote: Counting objects: 100% (4/4), done.
remote: Compressing objects: 100% (4/4), done.
remote: Total 9 (delta 0), reused 2 (delta 0), pack-reused 5
Unpacking objects: 100% (9/9), 3.13 KiB | 118.00 KiB/s, done.
From https://github.com/FreeCAD/FreeCAD
   6373760845..352fdc2d22  main       -> origin/main
Updating 6373760845..352fdc2d22
Fast-forward
 conda/cmake.cmd              |  1 +
 conda/cmake.sh               |  3 +++
 conda/conda-env.yaml         |  3 +++
 conda/environment.devenv.yml | 11 +++++------
 conda/setup-environment.cmd  | 10 ++++++++++
 conda/setup-environment.sh   | 12 ++++++++++++
 6 files changed, 34 insertions(+), 6 deletions(-)
 create mode 100644 conda/cmake.cmd
 create mode 100755 conda/cmake.sh
 create mode 100644 conda/setup-environment.cmd
 create mode 100755 conda/setup-environment.sh

However I did not see that src/Mod/TechDrawXMLQuery.h is pulled!?

luzpaz, 2023-11-14:
Do you mean https://github.com/FreeCAD/FreeCAD/blob/main/src/Mod/TechDraw/App/XMLQuery.h ?

FEA-eng, 2024-03-17:
@maxwxyz It's a rather old compilation error. I think that it could be closed due to a lack of feedback and no confirmation,

closed 2024-03-17 by maxwxyz