FreeCAD/FreeCAD #10346: [Problem] [Sketcher] [Reference Constraint] Multiple reference constraint allowed

V5Studio, 2023-08-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
[code]
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.21.0.33675 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.0)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * 3DfindIT 1.2.0
  * parts_library
[/code]
```

### Subproject(s) affected?

Sketcher

### Problem description

Can add the same (ref) constraint many times to the same element. Adding more than one will result in constraint being not selectable with mouse click

1. Add an element to sketch (ex: line)
2. Switch to reference constraint
3. Pick a tool (ex: horizontal distance)
4. Click on (2) to add constraint
5. Click again on (2) to add same constraint

Expected behavior: New constraint shouldn't be added (or should it replace existing one?)

Also, if you try adding with a driving constraint (skip step 2), it will result in below error. It does not add a duplicate constraint
11:32:38  Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'ValueError'>: Datum 39.86 mm for the constraint with index 1 is invalid
11:32:38  App.getDocument('Unnamed').getObject('Sketch').setDatum(1,App.Units.Quantity('39.860000 mm'))

### Anything else?

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2023-08-25:
Reference constraints are not used by the constraints solver and therefore not checked for redundancy. It would not make sense to do that. It would unnecessarily limit their use.

The 5th issue in the image is probably due to wrongly selecting the dimension by the dimension line instead of the dimension text.

V5Studio, 2023-08-25:
@Roy-043 The issue I have is that by mistake I have added the same reference constraint twice, and by doing so I am not able to select it anymore. Is there any use case where the same need to be added more than once?

Roy-043, 2023-08-25:
IMO the selection problem should not be solved by limiting reference constraints.

Please note that when you click the dimension text in the 3D view you are probably selecting the first of the dimensions. It is just that you do not see it being highlighted in the 3D view because the other dimensions are in the way. You can verify that a selection in fact occurs by looking at the constraints section in the task panel. You can also drag the selected dimension to a new location.

There are other situations where constraints are hard to select because they are close together or overlap. Such problems can often be overcome by zooming in or out a little.

maxwxyz, 2024-03-24:
Closing this issue as intended behavior.

closed 2024-03-24 by maxwxyz