FreeCAD/FreeCAD #10353: [Issue] TopoNaming Phase 3 (card 1 of 35) - Add History Support/Part Workbench/PropertyTopoShape

JohnAD, 2023-08-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
[code]
OS: Pop!_OS 22.04 LTS (pop:GNOME/pop)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33675 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.0)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Ondsel-Lens 2023.8.8.02
[/code]
```

### Subproject(s) affected?

Part

### Problem description

This is a card for Toponaming fix Phase 3 (add naming) Part 1 (history expansion).

Please see TN issue https://github.com/FreeCAD/FreeCAD/issues/8432 for overall discussion.

Also see project https://github.com/orgs/FreeCAD/projects/2/views/1 for tracking.

Phase 3 does not USE the naming yet, that is Phase 4.

## Specifics

Naming is based on history. Add the RealThunder changes to history to expand it's support.

RealThunder reference branch: https://github.com/FreeCAD/FreeCAD/tree/development/toponaming

Do this for the files mentioned in the issue title. The few cards are generic and the remaining ones are for specific `Feature` elements in the related workbench.

Please write one or more unit tests.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

chennes, 2023-09-02:
I guess it's time to get the ball rolling... I'll grab this one and get started.

Pesc0, 2023-09-18:
Hi Chris, you working on this?

chennes, 2023-09-18:
We'll get that PR merged once we're confident it doesn't break anyone's documents... changing the Document writing is something we want to be careful about.

Pesc0, 2023-09-18:
Sure! I didn't want to create pressure, I asked simply because I might need to "get inspired" for card 2 

_ergo copy what you're doing_

chennes, 2024-03-18:
Closed by #12683

closed 2024-03-18 by chennes