FreeCAD/FreeCAD #10354: [Issue] TopoNaming Phase 3 (card 2 of 35) - Add History Support/Part Workbench/PartFeature

JohnAD, 2023-08-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
[code]
OS: Pop!_OS 22.04 LTS (pop:GNOME/pop)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33675 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.0)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

Part

### Problem description

This is a card for Toponaming fix Phase 3 (add naming) Part 1 (history expansion).

Please see TN issue https://github.com/FreeCAD/FreeCAD/issues/8432 for overall discussion.

Also see project https://github.com/orgs/FreeCAD/projects/2/views/1 for tracking.

Phase 3 does not USE the naming yet, that is Phase 4.

# Specifics

Naming is based on history. Add the RealThunder changes to history to expand it's support.

RealThunder reference branch: https://github.com/FreeCAD/FreeCAD/tree/development/toponaming

Do this for the files mentioned in the issue title. The few cards are generic and the remaining ones are for specific `Feature` elements in the related workbench.

Please add one or more unit tests.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Pesc0, 2023-09-14:
https://github.com/FreeCAD/FreeCAD/blob/development/toponaming/src/Mod/Part/App/PartFeatures.h
This one? 

also is the branch correct? dont we want to use https://github.com/realthunder/FreeCAD/tree/LinkDaily ?

chennes, 2023-09-14:
The branch is correct -- realthunder has done quite a bit of work rebasing to master in that branch. It's still out of date of course, that work was quite a few months ago now, but it will be easier to start from there than from LinkDaily.

Pesc0, 2023-09-14:
Alright thanks! Will get to work

Pesc0, 2023-10-06:
@JohnAD I think we are missing some stuff in `App::GeoFeature`?

Pesc0, 2023-10-06:
Also, should the changes in this issue be able to compile by themselves? or this is just a piece of the puzzle that wont necessarily compile without the other issue being solved?

chennes, 2024-03-18:
Closed by #12804

closed 2024-03-18 by chennes