FreeCAD/FreeCAD #10356: [Issue] TopoNaming Phase 3 (card 4 of 35) - Add History Support/Part Workbench/FeaturePartExtension

JohnAD, 2023-08-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
[code]
OS: Pop!_OS 22.04 LTS (pop:GNOME/pop)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33675 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.0)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

Part

### Problem description

This is a card for Toponaming fix Phase 3 (add naming) Part 1 (history expansion).

Please see TN issue https://github.com/FreeCAD/FreeCAD/issues/8432 for overall discussion.

Also see project https://github.com/orgs/FreeCAD/projects/2/views/1 for tracking.

Phase 3 does not USE the naming yet, that is Phase 4.

# Specifics

Naming is based on history. Add the RealThunder changes to history to expand it's support.

RealThunder reference branch: https://github.com/FreeCAD/FreeCAD/tree/development/toponaming

Do this for the files mentioned in the issue title. The few cards are generic and the remaining ones are for specific `Feature` elements in the related workbench.

Please add one or more unit tests.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

chennes, 2024-03-18:
@JohnAD what is FeaturePartExtension?

bgbsww, 2024-03-19:
Hmm, maybe this was meant to be FeatureExtrusion, which notably doesn't seem to have an issue card to link it to.

chennes, 2024-03-20:
I'm going to decide you're right here and rename the card.

PR #13030 (2024-03-22): Toponaming/Part Move in feature extrusion
Fix #10356
closed 2024-03-22 by chennes